RfidResearchGroup/proxmark3 #663: ERROR: invalid serial port /dev/ttyACM0

2rsharipov, 2020-04-09:
Hello!

I have Proxmark3 Easy, successfully and without errors  instructions:
https://github.com/Chrissy-Morgan/proxmark3/blob/master/Installation_Instructions/Ubuntu-Installation-Instructions.md
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md

**An instruction I made based on your wiki knowledge (Ubuntu 18.04)**
**1.Update**
sudo apt-get update

**2.Requirements**
sudo apt-get install p7zip git build-essential libreadline5 libreadline-dev libusb-0.1-4 libusb-dev libqt4-dev perl pkg-config wget libncurses5-dev gcc-arm-none-eabi

**3.Clone Fork**
git clone https://github.com/RfidResearchGroup/proxmark3.git

**4.Change Directory**
cd proxmark3

**5.Get the latest commits**
git pull

**6.Install the blacklist rules and add user to dialout group**
sudo make udev

**7.Restart Ubuntu**

**8.Enter proxmark folder**
cd proxmark3

**9.Clean and Compile**
make clean && make all

**10.Install**
sudo make install

**11.Flash the BOOTROM & FULLIMAGE**
I tried different options for firmware.
_- I flashed with separate commands:_
pm3-flash-bootrom /dev/ttyACM0 -b bootrom/obj/bootrom.elf
pm3-flash-fullimage /dev/ttyACM0 -b armsrc/obj/fullimage.elf
_- I flashed with a single command:_
pm3-flash-all
And I always had a successful firmware result:
`administrator@adminpc:~/proxmark3$ pm3-flash-all
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/administrator/.proxmark3/log_20200409.txt 
[+] About to use the following files:          
[+]     /usr/local/bin/../share/proxmark3/firmware/bootrom.elf          
[+]     /usr/local/bin/../share/proxmark3/firmware/fullimage.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
 🕐 .Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00100000-0x00180000          
[+] Loading ELF file /usr/local/bin/../share/proxmark3/firmware/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000e30->0x00000e30) [R X] @0x298          
          
[+] Loading ELF file /usr/local/bin/../share/proxmark3/firmware/fullimage.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x00041bb8->0x00041bb8) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00143bb8 (0x0000141c->0x0000141c) [RW ] @0x41c4c          
[=] Note: Extending previous segment from 0x41bb8 to 0x42fd4 bytes          
          
          
[+] Flashing... 
          
[... 533 lines cut ...]
In this case, after flashing to the official firmware, this error disappears
And it appears again after the firmware on the firmware above

**13.Remove Modem Manager**
sudo apt remove modemmanager

**14.Restart Ubuntu**

**15.Change into the client folder**
cd proxmark3/client

**16.Run the client**
./proxmark3 /dev/ttyACM0

_But I have an error:_
`administrator@adminpc:~/proxmark3/client$ ./proxmark3 /dev/ttyACM0
[=] Session log /home/administrator/.proxmark3/log_20200409.txt 
[=] Using UART port /dev/ttyACM0           
          
[!]  ⚠️  ERROR: invalid serial port /dev/ttyACM0 `

**When re-flashing the original firmware, the device is correctly detected and works.
But I want to use the full capabilities of the device on your firmware!**

**Please help me solve the problem**

iceman1001, 2020-04-10:
... did you compile the project for the correct platform since you are using a non-rdv4 device?

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md

2rsharipov, 2020-04-10:
Dear **iceman1001**!

Thank for your prompt reply!

1. I created a file Makefile.platform in the root directory "proxmark3" of the repository:
```
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

PLATFORM=PM3OTHER
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
#STANDALONE=LF_SAMYRUN
```

2. I run command **make clean**
```
administrator@adminpc:~/proxmark3$ make clean

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
[!] Platform definitions changed, cleaning bootrom/armsrc/recovery first...
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
```
3. I run command **make SKIPQT=1**
```
administrator@adminpc:~/proxmark3$ make SKIPQT=1

===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[-] CC proxmark3.c
[-] CC uart/uart_posix.c
[-] CC uart/uart_win32.c
[-] CC ui.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC ../common/util_posix.c
[-] CC scandir.c
[-] CC ../common/crc16.c
[-] CC comms.c
[-] CC ../common/crapto1/crapto1.c
[-] CC ../common/crapto1/crypto1.c
[-] CC mifare/mifaredefault.c
[-] CC mifare/mfkey.c
[-] CC tea.c
[-] CC fido/additional_ca.c
[-] CC fido/cose.c
[-] CC fido/cbortools.c
[-] CC fido/fidocore.c
[-] CC crypto/asn1dump.c
[-] CC crypto/libpcrypto.c
[-] CC crypto/asn1utils.c
[-] CC cliparser/argtable3.c
[... 425 lines cut ...]
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    : 
        .    .. .. . . .. ...  .      .    .   . . ..    .  
...................................................................
.............................. OK           

          
[+] All done.           

Have a nice day!   
```
6.Change into the client folder
cd proxmark3/client

7. I run command **./proxmark3 /dev/ttyACM0**

**But I have an error again:**
```
administrator@adminpc:~/proxmark3/client$ ./proxmark3 /dev/ttyACM0
[=] Session log /home/administrator/.proxmark3/log_20200410.txt 
[=] Using UART port /dev/ttyACM0           
          
[!]  ⚠️  ERROR: invalid serial port /dev/ttyACM0 
```

2rsharipov, 2020-04-10:
I tried to connect Proxmark3 Easy to Windows 10, where everything is configured for working with the device and the driver. But it is also not defined in Windows 10.

I have information in **"Manage devices"**:
**Tab "General"**
_**Location**_ - Port_#0004.Hub_#0003

_**Device status:**_
Windows has stopped this device because it has reported problems. (Code 43)

A request for the USB device descriptor failed.

**Tab "Events"**
_**1. Timestamp - 4/5/2020 11:24:45 AM (Device not started):**_
Device USB\VID_0000&PID_0002\5&262ed807&0&4 had a problem starting.

Driver Name: usb.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Service: 
Lower Filters: 
Upper Filters: 
Problem: 0x2B
Problem Status: 0x0

_**2. Timestamp - 4/5/2020 11:24:45 AM (Device configured usb.inf):**_
Device USB\VID_0000&PID_0002\5&262ed807&0&4 was configured.

Driver Name: usb.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 06/21/2006
Driver Version: 10.0.18362.693
Driver Provider: Microsoft
Driver Section: BADDEVICE.Dev.NT
Driver Rank: 0xFF0000
Matching Device Id: USB\DEVICE_DESCRIPTOR_FAILURE
Outranked Drivers: usb.inf:USB\DEVICE_DESCRIPTOR_FAILURE:00FF2000
Device Updated: false
Parent Device: USB\ROOT_HUB30\4&d858888&0&0

**I really need your help in solving the problem!**

iceman1001, 2020-04-14:
...did you kill the modem manager?
... use the  shell scripts,
- ./pm3-flash-bootrom
- ./pm3-flash-fullimage
-./pm3

Which OS are you on?
Do you still have the issue?

closed 2020-04-15 by iceman1001

2rsharipov, 2020-04-21:
Dear **iceman**!

**Great, you are a professional and master! My device is working! I sincerely thank you!**

Since Proxmark3 Easy is not detected after your firmware, I always reflash Proxmark3 Easy in service mode:
I hold down the device button, run the command, wait for the firmware to finish, and release the device button!

1.I run command cd proxmark3
2. I run command ./pm3-flash-bootrom (in service mode)
3. I run command ./pm3-flash-fullimage (in service mode)
4. I run command ./pm3

cascha42, 2020-05-10:
Hi,
I seem to have the same problem. Only the "original" firmware is running. As soon as I use the current version from this repo, the fullimage seems to have a problem. The boot loader seems to bei fine, because the serial device is recognized when the push-button is pressed while inserting.

Flashing about JTAG and help on the IRC was also unsuccessful.

Also tried the Advanced Compilation for other Platform, but no success

PM3 Easy 512MB, Ubuntu 19.04, im pretty sure the problem is not in the PC site.

Thanks a lot,
cascha42