RfidResearchGroup/proxmark3 #675: staticnested found no keys

l3ob, 2020-04-13:
is it normal or bug?
let me know if need more information.

```
[usb] pm3 --> hf 14a info

[+]  UID: EB 4D 66 18
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] POSSIBLE TYPE:    MIFARE Classic 1K / Classic 1K CL2
[+] POSSIBLE TYPE:    MIFARE Plus 2K / Plus EV1 2K
[+] POSSIBLE TYPE:    MIFARE Plus CL2 2K / Plus CL2 EV1 2K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: weak
[+] Static nonce: yes
```

```
[usb] pm3 --> hf mf chk *1 ?
[=] No key specified, trying default keys
[ 0] ffffffffffff
[ 1] 000000000000
[ 2] a0a1a2a3a4a5
[ 3] b0b1b2b3b4b5
[ 4] c0c1c2c3c4c5
[ 5] d0d1d2d3d4d5
[ 6] aabbccddeeff
[ 7] 1a2b3c4d5e6f
[ 8] 123456789abc
[ 9] 010203040506
[10] 123456abcdef
[11] abcdef123456
[12] 4d3a99c351dd
[13] 1a982c7e459a
[14] d3f7d3f7d3f7
[15] 714c5c886e97
[16] 587ee5f9350f
[17] a0478cc39091
[18] 533cb6c723f6
[19] 8fd0a4f256e9
[20] 0000014b5c31
[21] b578f38a5c61
[22] 96a301bce267
................................
[=] Time in checkkeys: 7 seconds

[=] testing to read key B...

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 001 | ------------   | 0 | ------------   | 0 |
[+] | 002 | ------------   | 0 | ------------   | 0 |
[+] | 003 | ------------   | 0 | ------------   | 0 |
[+] | 004 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 005 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 006 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 007 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 008 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 009 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 010 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 011 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 012 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 013 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 014 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 015 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0 :Failed / 1 :Success)
```

```
[usb] pm3 --> hf mf staticnested 1 4 A 37535ba34aab d

..........

[+] time in static nested: 2718 seconds

[=] trying to read key B...

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 001 | 37535ba34aab   | 1 | 37535ba34aab   | 1 |
[+] | 002 | ------------   | 0 | ------------   | 0 |
[+] | 003 | ------------   | 0 | ------------   | 0 |
[+] | 004 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 005 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 006 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 007 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 008 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 009 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 010 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 011 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 012 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 013 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 014 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 015 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0 :Failed / 1 :Success)
[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-EB4D6618-key-2.bin --> 0xffffffffffff has been inserted for unknown keys.
```

iceman1001, 2020-04-13:
bummer,  try running it with  ff ff ff ff ff ff as input?

l3ob, 2020-04-13:
@iceman1001 same with ffffffffffff

found this topic: http://www.proxmark.org/forum/viewtopic.php?id=7668

works with offical repo
don't know what's wrong with this build

iceman1001, 2020-04-13:
Seems Proxmark3 Easy has some issues running on this repo.

closed 2020-04-14 by iceman1001