RfidResearchGroup/proxmark3 #681: macos making error

l3ob, 2020-04-13:
> error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]

remove -Werror can work

```
===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
/Applications/Xcode.app/Contents/Developer/usr/bin/make --no-print-directory -C client all DESTDIR=
[-] CC proxmark3.c
mkdir -p obj/
gcc -MT obj/proxmark3.o -MMD -MP -MF obj/proxmark3.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/proxmark3.o proxmark3.c
mv -f obj/proxmark3.Td obj/proxmark3.d && touch obj/proxmark3.o
[-] CC uart/uart_posix.c
mkdir -p obj/
gcc -MT obj/uart_posix.o -MMD -MP -MF obj/uart_posix.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/uart_posix.o uart/uart_posix.c
mv -f obj/uart_posix.Td obj/uart_posix.d && touch obj/uart_posix.o
[-] CC uart/uart_win32.c
mkdir -p obj/
gcc -MT obj/uart_win32.o -MMD -MP -MF obj/uart_win32.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/uart_win32.o uart/uart_win32.c
mv -f obj/uart_win32.Td obj/uart_win32.d && touch obj/uart_win32.o
[-] CC ui.c
mkdir -p obj/
gcc -MT obj/ui.o -MMD -MP -MF obj/ui.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/ui.o ui.c
mv -f obj/ui.Td obj/ui.d && touch obj/ui.o
[-] CC ../common/commonutil.c
mkdir -p obj/
gcc -MT obj/commonutil.o -MMD -MP -MF obj/commonutil.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/commonutil.o ../common/commonutil.c
mv -f obj/commonutil.Td obj/commonutil.d && touch obj/commonutil.o
[-] CC util.c
mkdir -p obj/
gcc -MT obj/util.o -MMD -MP -MF obj/util.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/util.o util.c
mv -f obj/util.Td obj/util.d && touch obj/util.o
[-] CC ../common/util_posix.c
mkdir -p obj/
gcc -MT obj/util_posix.o -MMD -MP -MF obj/util_posix.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/util_posix.o ../common/util_posix.c
mv -f obj/util_posix.Td obj/util_posix.d && touch obj/util_posix.o
[-] CC scandir.c
mkdir -p obj/
gcc -MT obj/scandir.o -MMD -MP -MF obj/scandir.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/scandir.o scandir.c
mv -f obj/scandir.Td obj/scandir.d && touch obj/scandir.o
[-] CC ../common/crc16.c
mkdir -p obj/
gcc -MT obj/crc16.o -MMD -MP -MF obj/crc16.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/crc16.o ../common/crc16.c
mv -f obj/crc16.Td obj/crc16.d && touch obj/crc16.o
[-] CC comms.c
mkdir -p obj/
gcc -MT obj/comms.o -MMD -MP -MF obj/comms.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/comms.o comms.c
mv -f obj/comms.Td obj/comms.d && touch obj/comms.o
[-] CC ../common/crapto1/crapto1.c
mkdir -p obj/crapto1/
gcc -MT obj/crapto1/crapto1.o -MMD -MP -MF obj/crapto1/crapto1.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/crapto1/crapto1.o ../common/crapto1/crapto1.c
mv -f obj/crapto1/crapto1.Td obj/crapto1/crapto1.d && touch obj/crapto1/crapto1.o
[-] CC ../common/crapto1/crypto1.c
mkdir -p obj/crapto1/
gcc -MT obj/crapto1/crypto1.o -MMD -MP -MF obj/crapto1/crypto1.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/crapto1/crypto1.o ../common/crapto1/crypto1.c
mv -f obj/crapto1/crypto1.Td obj/crapto1/crypto1.d && touch obj/crapto1/crypto1.o
[-] CC mifare/mifaredefault.c
mkdir -p obj/mifare/
gcc -MT obj/mifare/mifaredefault.o -MMD -MP -MF obj/mifare/mifaredefault.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/mifare/mifaredefault.o mifare/mifaredefault.c
mv -f obj/mifare/mifaredefault.Td obj/mifare/mifaredefault.d && touch obj/mifare/mifaredefault.o
[-] CC mifare/mfkey.c
mkdir -p obj/mifare/
gcc -MT obj/mifare/mfkey.o -MMD -MP -MF obj/mifare/mfkey.Td -Wall -Werror -g -O3 -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -Iuart -I/usr/local/opt/readline/include -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -I../common/zlib -I./reveng -I./amiitool -c -o obj/mifare/mfkey.o mifare/mfkey.c
mv -f obj/mifare/mfkey.Td obj/mifare/mfkey.d && touch obj/mifare/mfkey.o
[... 35 lines cut ...]
cliparser/argtable3.c:417:17: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                warnx(ambig, (int)current_argv_len,
                ^
cliparser/argtable3.c:427:17: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                warnx(noarg, (int)current_argv_len,
                ^
cliparser/argtable3.c:457:17: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                warnx(recargstring,
                ^
cliparser/argtable3.c:475:13: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
            warnx(illoptstring, current_argv);
            ^
cliparser/argtable3.c:651:13: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
            warnx(illoptchar, optchar);
            ^
cliparser/argtable3.c:662:17: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                warnx(recargchar, optchar);
                ^
cliparser/argtable3.c:683:21: error: implicit declaration of function 'warnx' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                    warnx(recargchar, optchar);
                    ^
7 errors generated.
make[1]: *** [obj/cliparser/argtable3.o] Error 1
make: *** [client/all] Error 2
```

uhei, 2020-04-14:
Although I'm not able to fix this, just for information:

Latest HEAD (054f28e7830bee2a72a31ccc3a87581aed704e5b) compiles just fine with clang 11.0.3 on macOS 10.15.4 (Catalina)

```
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.29)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

iceman1001, 2020-04-14:
So on a old clang v10,  and Darvin v17.7 it fails,  vs on a newer OS and clang it works.

@l3ob   How about you upgrade your CLANG ?

Otherwise the define for WIN32 in  **client/cliparser/argtable3.c**  might be needed for OSX. (_APPLE)
```
#ifdef _WIN32

/* Windows needs warnx().  We change the definition though:
 *  1. (another) global is defined, opterrmsg, which holds the error message
 *  2. errors are always printed out on stderr w/o the program name
 * Note that opterrmsg always gets set no matter what opterr is set to.  The
 * error message will not be printed if opterr is 0 as usual.
 */

#include <stdio.h>
#include <stdarg.h>

#define MAX_OPTER_MSG_SIZE 128

extern char opterrmsg[MAX_OPTER_MSG_SIZE];
char opterrmsg[MAX_OPTER_MSG_SIZE]; /* buffer for the last error message */

static void warnx(const char *fmt, ...) {
........ --truncated ---
#else
#include <err.h>
#endif /*_WIN32*/
```

l3ob, 2020-04-15:
@iceman1001 problem found

my /usr/local/include has another `err.h` created by openssl
so the compiler use this instead of /usr/inlucde/
everything works fine now when I remove it

```
$ gcc -v -E -
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang" -cc1 -triple x86_64-apple-macosx10.13.0 -Wdeprecated-objc-isa-usage -Werror=deprecated-objc-isa-usage -E -disable-free -disable-llvm-verifier -discard-value-names -main-file-name - -mrelocation-model pic -pic-level 2 -mthread-model posix -mdisable-fp-elim -fno-strict-return -masm-verbose -munwind-tables -target-cpu penryn -dwarf-column-info -debugger-tuning=lldb -target-linker-version 409.12 -v -resource-dir /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/10.0.0 -fdebug-compilation-dir /usr/local/include -ferror-limit 19 -fmessage-length 80 -stack-protector 1 -fblocks -fencode-extended-block-signature -fobjc-runtime=macosx-10.13.0 -fmax-type-align=16 -fdiagnostics-show-option -fcolor-diagnostics -o - -x c -
clang -cc1 version 10.0.0 (clang-1000.11.45.5) default target x86_64-apple-darwin17.7.0
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/10.0.0/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include
 /usr/include
 /System/Library/Frameworks (framework directory)
 /Library/Frameworks (framework directory)
End of search list.
```

closed 2020-04-15 by l3ob