RfidResearchGroup/proxmark3 #682: The EML simulation

Windslab, 2020-04-13:
Hi,

When an EML 14a in the simulation mode then it looks okay. The MCT tool has read UID aswell, soft for the card too, but when you trying to read blocks/sectors it's looking like break>repeat to infinity.

Me feel like a an emulated data not permanent for reader or it's permanent for writing. By the way it's not good.

Many Thanks

iceman1001, 2020-04-14:
The google translation makes it hard to understand what you are trying to say.

Is there an issue with a command? In such case which command?  Explain what you were trying to do.

Windslab, 2020-04-14:
I talking about this:
1. hf mf eload 1 hf-mf-11223344556677-dump
2. hf mf sim

When MIFARE Classic Tool trying to read blocks/sectors it's can't and for me it looks like you remove a card and put it again to the phone. The situation is similar for the another soft.

To put it simply: You cannot read that you trying to simulate because something drop a connection between Proxmark and reader.

iceman1001, 2020-04-14:
you add a 7byte uid dump and you run sim without params.   The nfc antenna on smart phones aren't very good.  Just see how you need put a normal tag there to make a good coupling.

I still see no issue here.

Windslab, 2020-04-14:
Haha...

iceman1001, 2020-04-14:
The MCT app / Android seems to do a full anticollision first,  then it does the short cut version 0x9370 uid..
and for each auth command it sends,  the simlation answers with 0x04.  I wonder why it sends a NACK all the time.

```
[usb] pm3 --> hf mf sim t 1
[=] Mifare 1K  |  UID  N/A
[=] Options [ numreads: 0, flags: 272 (0x110) ]
#db# Enforcing Mifare 1K ATQA/SAK
#db# 4B UID: bde22b43
#db# ATQA  : 00 04
#db# SAK   : 08
#db# Emulator stopped. Tracing: 1  trace length: 13396
[usb] pm3 --> hf mf list
[+] Recorded activity (trace len = 13396 bytes)
[=] Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
[=] ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       1056 | Rdr |26                                                                       |     | REQA
       2100 |       4468 | Tag |04  00                                                                   |     |
      11656 |      16424 | Rdr |50  00  57  cd                                                           |  ok | HALT
      93750 |      94742 | Rdr |52                                                                       |     | WUPA
      95850 |      98218 | Tag |04  00                                                                   |     |
     105416 |     107880 | Rdr |93  20                                                                   |     | ANTICOLL
     108924 |     114812 | Tag |bd  e2  2b  43  37                                                       |     |
     121928 |     132456 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
     133500 |     137020 | Tag |08  b6  dd                                                               |     |
     479112 |     483880 | Rdr |50  00  57  cd                                                           |  ok | HALT
     644280 |     645272 | Rdr |52                                                                       |     | WUPA
     646380 |     648748 | Tag |04  00                                                                   |     |
     655944 |     666472 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
     667516 |     671036 | Tag |08  b6  dd                                                               |     |
     786440 |     791208 | Rdr |50  00  57  cd                                                           |  ok | HALT
     921214 |     922206 | Rdr |52                                                                       |     | WUPA
     923314 |     925682 | Tag |04  00                                                                   |     |
     932862 |     943390 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
     944434 |     947954 | Tag |08  b6  dd                                                               |     |
    1086042 |    1090746 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
    1092494 |    1093134 | Tag |04                                                                       |     |
    1099960 |    1109272 | Rdr |d1  6f  f6  81  39  24  2b  c1                                           | !crc|
    1110764 |    1111404 | Tag |04                                                                       |     |
    1229368 |    1234136 | Rdr |50  00  57  cd                                                           |  ok | HALT
    1363418 |    1364410 | Rdr |52                                                                       |     | WUPA
    1365518 |    1367886 | Tag |04  00                                                                   |     |
    1375084 |    1385612 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
    1386656 |    1390176 | Tag |08  b6  dd                                                               |     |
    1512984 |    1517752 | Rdr |50  00  57  cd                                                           |  ok | HALT
    1633720 |    1634712 | Rdr |52                                                                       |     | WUPA
    1635820 |    1638188 | Tag |04  00                                                                   |     |
    1645354 |    1655882 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
    1656990 |    1660510 | Tag |08  b6  dd                                                               |     |
    1785052 |    1789756 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
    1791504 |    1792144 | Tag |04                                                                       |     |
    1799002 |    1808378 | Rdr |dd  4d  23  6c  8a  d5  ee  63                                           | !crc|
    1809678 |    1810318 | Tag |04                                                                       |     |
    1900888 |    1905656 | Rdr |50  00  57  cd                                                           |  ok | HALT
[... cut ...]
    2298710 |    2301078 | Tag |04  00                                                                   |     |
    2308258 |    2318786 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
    2319830 |    2323350 | Tag |08  b6  dd                                                               |     |
    4179704 |    4184472 | Rdr |50  00  57  cd                                                           |  ok | HALT
    4284984 |    4285976 | Rdr |52                                                                       |     | WUPA
```

iceman1001, 2020-04-14:
Ok, I think I found the bug.  Pull latest / compile / flash and test.  

You were correct,  there was a bug and it was mine to start with.

Windslab, 2020-04-14:
Now MCT has read card till to end and then drop at 100%(that's looks like at last moment you remove a tag and put it again, android tell me that's a new tag has been founded). Also it's drop when you trying to brute keys from MCT.

iceman1001, 2020-04-14:
On my MCT tool it does 100%,  shows the dump.   Did you have right accessrights/got all keys?

Brute keys from MCT?   The check keys from default key list works fine when I run it.

iceman1001, 2020-04-14:
Which device do you have? En Easy?

Windslab, 2020-04-14:
Yes an China Easy... After you told me that my antena are not good I have had remove middle textolit. My phone is okay...

Windslab, 2020-04-14:
Sorry I didn't see your pre last messege. Yes the accessrights/got all keys is okay. Yes check keys from default key list.

MCT at 100%:
1. Lost connection to the tag
2. New tag found

Windslab, 2020-04-14:
PS in moment between 1 and 2 point all Green and Red led are off and then has on again

Windslab, 2020-04-14:
@iceman1001 please find attached log: https://www.dropbox.com/s/t53r4dsz1ss4bkx/hf_mf_sim.log?dl=0

iceman1001, 2020-04-14:
Did you pull latest / compile / flash?  
Your log is full of the bug that I fixed..

```
  131461944 |  131472472 | Rdr |93  70  5d  a2  5d  1f  bd  6a  4f                                       |  ok | SELECT_UID
  131473516 |  131477036 | Tag |88  be  59                                                               |     |
  131559408 |  131564176 | Rdr |61  20  2f  43                                                           |  ok | AUTH-B(32)
  131565732 |  131566372 | Tag |04                                                                       |     |
  131576176 |  131585488 | Rdr |c2! 5a! e4  49! b6! d0! d8! 27                                           | !crc| RESTORE(90)
  131586980 |  131587620 | Tag |04                                                                       |     |
  131623922 |  131628690 | Rdr |50  00  57  cd                                                           |  ok | HALT
  131715256 |  131716248 | Rdr |52                                                                       |     | WUPA
  131717356 |  131719724 | Tag |04  00                                                                   |     |
  131726942 |  131737470 | Rdr |93  70  5d  a2  5d  1f  bd  6a  4f                                       |  ok | SELECT_UID
  131738514 |  131742034 | Tag |88  be  59                                                               |     |
  131803066 |  131807834 | Rdr |61  20  2f  43                                                           |  ok | AUTH-B(32)
  131809390 |  131810030 | Tag |04                                                                       |     |
  131819406 |  131828782 | Rdr |0b  a4! 4d! bd  b9! 0e  1f  b5                                           | !crc|
  131830018 |  131830658 | Tag |04                                                                       |     |
  131877090 |  131881858 | Rdr |50  00  57  cd                                                           |  ok | HALT
  131962808 |  131963800 | Rdr |52                                                                       |     | WUPA
  131964908 |  131967276 | Tag |04  00                                                                   |     |
  131975482 |  131986010 | Rdr |93  70  5d  a2  5d  1f  bd  6a  4f                                       |  ok | SELECT_UID
  131987054 |  131990574 | Tag |88  be  59                                                               |     |
  132081044 |  132085812 | Rdr |61  20  2f  43                                                           |  ok | AUTH-B(32)
  132087368 |  132088008 | Tag |04     
```

Windslab, 2020-04-14:
Yes.

[image]

iceman1001, 2020-04-14:
doesn't look like your device doesn't have it because your log.
My log looks like this after   MCT tool against a hf mf sim.

```
  55396772 |   55397764 | Rdr |52                                                                       |     | WUPA
   55398872 |   55401240 | Tag |04  00                                                                   |     |
   55408454 |   55418982 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
   55420026 |   55423546 | Tag |08  b6  dd                                                               |     |
   55493666 |   55498370 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
   55503062 |   55507798 | Tag |3d  e7  2c  25                                                           |     | AUTH: nt
   55514652 |   55524028 | Rdr |96  ca  23  68  7d  cd  51  af                                           |     | AUTH: nr ar (enc)
   55590188 |   55594956 | Rdr |50  00  57  cd                                                           |     | HALT
   55677624 |   55678616 | Rdr |52                                                                       |     | WUPA
   55679724 |   55682092 | Tag |04  00                                                                   |     |
   55689322 |   55699850 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
   55700894 |   55704414 | Tag |08  b6  dd                                                               |     |
   55804672 |   55809440 | Rdr |61  00  2d  62                                                           |  ok | AUTH-B(0)
   55813940 |   55818612 | Tag |1d  f4  39  78                                                           |     | AUTH: nt
   55825548 |   55834860 | Rdr |14  95  8e  1f  7d  cc  3d  72                                           |     | AUTH: nr ar (enc)
   55901260 |   55906028 | Rdr |50  00  57  cd                                                           |     | HALT
   55996542 |   55997534 | Rdr |52                                                                       |     | WUPA
   55998642 |   56001010 | Tag |04  00                                                                   |     |
   56008206 |   56018734 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
   56019778 |   56023298 | Tag |08  b6  dd                                                               |     |
   56113626 |   56118330 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
   56123022 |   56127758 | Tag |dd  c4  57  73                                                           |     | AUTH: nt
   56134648 |   56143960 | Rdr |a7  d9  b3  78  bd  59  1f  a8                                           |     | AUTH: nr ar (enc)
   56152684 |   56157420 | Tag |8d  ac  69  b2!                                                          |     | AUTH: at (enc)
   56254848 |   56259552 | Rdr |e8  51  3b  64                                                           |     |
            |            |  *  |                                             key ffffffffffff prng WEAK  |     |
            |            |  *  |61  00  2D  62                                                           |  ok | AUTH-B(0)
   56266164 |   56270836 | Tag |22! bb! b1  ee                                                           |     | AUTH: nt (enc)
   56277724 |   56287100 | Rdr |14  81  f6  33  98  4f  88  6f                                           |     | AUTH: nr ar (enc)
   56295696 |   56300432 | Tag |ee  ae  24! ab                                                           |     | AUTH: at (enc)
   56374106 |   56378810 | Rdr |19  56  21  28                                                           |     |
            |            |  *  |                                               last used key ffffffffffff|     |
            |            |  *  |60  04  D1  3D                                                           |  ok | AUTH-A(4)
   56385422 |   56390094 | Tag |22! bb! b1  ee                                                           |     | AUTH: nt (enc)
   56464828 |   56469596 | Rdr |50  00  57  cd                                                           |     | HALT
   56553306 |   56554298 | Rdr |52                                                                       |     | WUPA
   56555406 |   56557774 | Tag |04  00                                                                   |     |
   56564970 |   56575498 | Rdr |93  70  bd  e2  2b  43  37  3f  75                                       |  ok | SELECT_UID
   56576542 |   56580062 | Tag |08  b6  dd                                                               |     |
```

Windslab, 2020-04-14:
Try to simulate this one: https://www.dropbox.com/s/rzbl9yk70f430kl/hf-mf-5DA25D1F-dump.eml?dl=0

I have had checked all. My device has reflashed...

Windslab, 2020-04-14:
Yeah `50 00 57 cd` everywhere...

Windslab, 2020-04-14:
I found case. Without two blocks for me all working good:

https://github.com/RfidResearchGroup/proxmark3/blob/f1f0e277b8b89ff0c729258842b5f9d7da6ff8d6/armsrc/mifaresim.c#L790-L804

Windslab, 2020-04-15:
Yeah me was wrong... Your fix working as well. Seems it's called mistake by China Proxmark. Antenna really not good. Team work.

Please close after read.

commit 054f28e7 (2020-04-14): fix #682 - hf mf sim -  not working good with MCT Tool app.   Actually it didnt work at all
closed 2020-04-15 by iceman1001