RfidResearchGroup/proxmark3 #683: "hf 14a list" could not sniff traffic from card

Ghost-Assassin, 2020-04-14:
**Describe the bug**
hf 14a sniff is no longer able to sniff the conversation between reader and card for my Proxmark3 Easy (256K Version). 

**To Reproduce**
Steps to reproduce the behavior:
I used this version, was the latest on April 14th: https://github.com/RfidResearchGroup/proxmark3/tree/0eb303c1523a4249cb046a92f2ec63c90d941f48

1. Compile the latest version of firmware both bootrom and fullimage with command "make PLATFORM=PM3OTHER", and installed successfully with built-in flasher. 
2. Restart PM3 by disconnect from the computer, and connect back to the computer.
3. Launch the client which was built by step 1.  
4. run "hf 14a sniff", place the reader under PM3, the led C is lit. 
5. place card above PM3, the led B is no lit.
6. "hf list" shows the traffic only from the reader, no traffic was captured for card. 

**Expected behavior**
"hf 14a sniff" should be able to capture both direction traffic. 

**Desktop (please complete the following information):**
 - OS: Tried on Windows 7 and Ubuntu 18.04
 - hw version and hf list 
[image]
[image]
[image]

**Additional context**
I have tried to flash to stock firmware 2.5.0 which worked as expected, so the PM3 itself should not have a hardware issue. Firmwares I tried is added in the attachment. 
[firmware_win.zip](https://github.com/RfidResearchGroup/proxmark3/files/4473304/firmware_win.zip)

Ghost-Assassin, 2020-04-14:
PS. the stock 2.5.0 firmware would constantly capture all the traffic with or without card interaction. The latest iceman firmware seems only captures traffic if the card is already on the card reader. 

I have tested 3 scenarios: 
- If I place the card on the card reader, then start sniffing on the console, then move the PM3 near the card, I will be able to capture the traffic from both directions. 
- If I start sniffing first and place the card on top of PM3 then move the PM3 to the reader, I won't be able to capture traffic from both directions.  
- If I start sniffing first, then place the PM3 on the top of card reader, then move the card on top of PM3, I will only be able to sniff from card reader, but not from card.

iceman1001, 2020-04-14:
Maybe if you read the helptext you find how to work with it better.   The problem with sniff is that lack of space to store the trace data.  It eats it up fast.

I have no problem with this command.

```
[usb] pm3 --> hf 14a sniff h
It get data from the field and saves it into command buffer.
Buffer accessible from command 'hf list 14a'
Usage:  hf 14a sniff [c][r]
c - triggered by first data from card
r - triggered by first 7-bit request from reader (REQ,WUP,...)
Examples:
        hf 14a sniff c r
```

Ghost-Assassin, 2020-04-14:
thanks for your reply iceman : )
I tried the method you mentioned. Yes, it would pick up the WUPA from the reader just like the stock firmware does. But still, my PM3 did not pick up any conversation from the card. 

Just like this screenshot shows, it captured messages from the reader but missed everything from the card. 
[image]

If I set it to trigger by first data from card, it seems will never be able to start capturing. 
[image]
 
I also tried to **start sniffing after the card is already on the reader**.
I will be able to capture conversations from both card and reader, and it seems I could capture at most 39153 bytes. 
[image]

If this function worked with no problem on your PM3, I will try to find another 512K version to cross-compare if mine has any hardware issues (I hope not).

Ghost-Assassin, 2020-04-14:
nvm, I tried on another reader. I placed the same card on my phone with the exact method, it worked no problem. 

Seems the reader has some problem caused this issue. I will try to find out what caused this problem and see if it could help anyone has this issue in the future.

closed 2020-04-14 by Ghost-Assassin

Ghost-Assassin, 2020-04-14:
wait a sec, 
you are not gonna believe it, I have to place my card around 45° to capture its traffic. 
It has to be this weird angle to make it work LOL ...
[image]

iceman1001, 2020-04-14:
Good that it resolved itself.

sniffing is picky about positions and antenna couplings vs reader field. 
the bigbuffer is 40kb,  so 39kb log data sounds like you filled it up.  Don't know what more you expected from it.