RfidResearchGroup/proxmark3 #686: [idea]To be a tester

Windslab, 2020-04-15:
I have to think that I like what me doing for community last time. I wishing to continue, maybe this way can turn me on into team as member in future.

I have feeling that's I need to change my chinese hardware. The main reason why me bought it that's money. To be a tester I need to use a last version of the boards.

The price at lab401 of RDV4.1 for me is directly too much. Also in my country I should be pay 20% of taxes at customs. Do anybody have a discount cupons, if it's exists might for community members? Maybe you have no needed a Proxmark board and you can send it to me in Ukraine?

Many Thanks

iceman1001, 2020-04-15:
Not a topic for this repo.

closed 2020-04-15 by iceman1001

Windslab, 2020-04-15:
That's topic for you...

Hello from FluxBB:
Unable to connect to MySQL and select database. MySQL reported: No such file or directory.