RfidResearchGroup/proxmark3 #693: Discussion on cmake

doegox, 2020-04-16:
About issues in the https://github.com/RfidResearchGroup/proxmark3/tree/newcmake branch

doegox, 2020-04-16:
@bkerler I get a "Qt5 library not found, not building gui" with cmake
What should I do to help you debugging the situation ?

bkerler, 2020-04-16:
I haven't yet tested the qt5 gui. That's next on my todo to verify it works.

bkerler, 2020-04-16:
Question : Do we still need qt4 support as well ? Right now, cmake only is modified to support qt5.

doegox, 2020-04-16:
for me we can drop Qt4, @iceman1001 ?

mwalker33, 2020-04-16:
I have my original proxspace installed on windows 10 and the qt ploat/slider windows work ok.
Also testing on my kali Linux and again both seem to work as expected.
from kali
pkg-config --modversion QtCore
4.8.7

iceman1001, 2020-04-16:
Hm,  dropping support QT4,  lets see,  it comes down to *nix distros if some doesn't have QT5.
We already dropped support for Ubuntu 12,14,15,16,  because of GCC.    
We should at least support a few generations.  Proxspace v3.3 has QT5? 
WSL uses bonic ubuntu 18.
OSX ??

iceman1001, 2020-04-16:
hm,,   .gitignore  needs to reflect the directory changes :)

doegox, 2020-04-16:
> hm,, .gitignore needs to reflect the directory changes :)

that's not a discussion for here, you can push the fix directly on master ;)

doegox, 2020-04-16:
> I have my original proxspace installed on windows 10 and the qt ploat/slider windows work ok.
> Also testing on my kali Linux and again both seem to work as expected.
> from kali
> pkg-config --modversion QtCore
> 4.8.7

Well on my machine too because I've Qt4 and Qt5, and proxmark takes the Qt5.

Do you have Qt5 as well? Look for Qt5Core

```
 pkg-config --modversion Qt5Core
5.12.5
```

iceman1001, 2020-04-16:
WSL has:
```
pkg-config --modversion Qt5Core
5.9.5
```

bkerler, 2020-04-16:
Ok, fixed QT5 support on cmake. Pull request #694

iceman1001, 2020-04-16:
pull request fix?   Did we already add CMAKE??!?

doegox, 2020-04-16:
it's a pull req on the newcmake branch @iceman1001 , relax ;)

doegox, 2020-04-16:
next step : getting ``make install / uninstall`` working ? :)

mwalker33, 2020-04-16:
On my setup (win10 proxspace), when I compile the client, it adds "ui_overlays.h" to client/src/ui folder.
Everything works fine (no compile issues and plot etc all works), but it means I have a new file in my git compare.  Note: At the moment, I have only tested from my dev compare to my branch.

doegox, 2020-04-16:
I added Travis jobs on the branch.
I had to lower the minimal cmake version to the one available in Travis env:
``cmake_minimum_required(VERSION 3.12.4)`` 
is it ok ?

The build succeeds under linux:
https://travis-ci.org/github/RfidResearchGroup/proxmark3/jobs/675980327

But it fails under OSX:
https://travis-ci.org/github/RfidResearchGroup/proxmark3/jobs/675980330#L194

doegox, 2020-04-16:
> On my setup (win10 proxspace), when I compile the client, it adds "ui_overlays.h" to client/src/ui folder.

yes same under Linux :(

doegox, 2020-04-16:
@mwalker33 I pushed a fix for ui_overlays.h it's now in build

mwalker33, 2020-04-17:
@doegox
QT in my Proxspace : 5.10.1

ui_overlays.h looks now :)

iceman1001, 2020-04-17:
ADDITIONAL_SRC   for APPLE isn't used together with TARGET SOURCES.
Not sure why bother with a tmp var for it, when it adds direct to TARGET_SOURCES for "QT5_FOUND" tests.   Should have same style.

iceman1001, 2020-04-17:
pushed a fix.  Looks like it compiles.

iceman1001, 2020-04-17:
hm,  well the makefile builds on osx,  not cmake.   2 down 1 to go.
the pm3test.sh file,  does its test find the stuff with all directory changes?  I mean in the individual tests like for hardnested tables etc?

iceman1001, 2020-04-17:
env variable PM3BIN isn't used inside pm3test.sh,   

All tests seems to be looking for a hardcode path to the pm3 client. CMAKE would have put it inside /client/build/  right?
```
 if ! CheckExecute "proxmark help" "./client/proxmark3 -h" "wait"; then break; fi
  if ! CheckExecute "proxmark help text ISO7816" "./client/proxmark3 -t 2>&1" "ISO7816"; then break; fi
  if ! CheckExecute "proxmark help text hardnested" "./client/proxmark3 -t 2>&1" "hardnested"; then break; fi
```

iceman1001, 2020-04-17:
WSL cmake version is  v3.10.2   ,  I had to lower it even more @doegox  :)

doegox, 2020-04-17:
> hm, well the makefile builds on osx, not cmake. 2 down 1 to go.
> the pm3test.sh file, does its test find the stuff with all directory changes? I mean in the individual tests like for hardnested tables etc?

Yes it does, see cmake on linux compiles and tests fine

doegox, 2020-04-17:
> env variable PM3BIN isn't used inside pm3test.sh,
>
> All tests seems to be looking for a hardcode path to the pm3 client. CMAKE would have put it inside /client/build/ right?
>
> ```
>  if ! CheckExecute [...]

Where are you looking ?
on the newcmake branch (the only one wehere there is cmake at the moment) I have
```
  if ! CheckExecute "proxmark help"                    "$PM3BIN -h" "wait"; then break; fi

```

iceman1001, 2020-04-17:
Hm, I must have been in the wrong local repo.  My bad.  PM3BIN is used in pm3tests.sh.

doegox, 2020-04-17:
Ok I'll try to recap the list of things to still look at, and keep the list updated later in the thread.

- [ ] cmake version 3.16 : is it ok to lower it ? it's already lowered to 3.12.4 in the branch for Travis. For WSL it has to be lowered to 3.10.2
- [ ] fails on OSX, cf Travis builds
- [ ] need to add make install / uninstall 
- [ ] do we need headers in CMakeLists.txt ? everything seems to work fine without them explicitly in TARGET_SOURCES, ok to remove them ?

iceman1001, 2020-04-17:
Ok, testing on OSX now. 
cmake doesn't find Qt5 despite its installed on system.  There is a PR,  I will merge it and test again

iceman1001, 2020-04-17:
hm,  got a fix for cmake file,  so it doesn't break.  
However I get a readline problem now.

doegox, 2020-04-17:
@iceman1001 now even the build on linux fails :(
It happened when Merge pull request #696 from uhei/newcmake

uhei, 2020-04-17:
> hm, got a fix for cmake file, so it doesn't break.
> However I get a readline problem now.

There was commit 94b782ee107c9981e0eb2172bdada78bad462f30 to fix this readline thing for macOS

doegox, 2020-04-17:
@uhei it seems your commit, when merged, broke the compilation on Linux :

https://travis-ci.org/github/RfidResearchGroup/proxmark3/jobs/676105617

iceman1001, 2020-04-17:
it works locally on  my Ubuntu 19.04 /QT5 aswell...

doegox, 2020-04-17:
to be compared with build before merge : https://travis-ci.org/github/RfidResearchGroup/proxmark3/jobs/676050472

uhei, 2020-04-17:
My fault. Yes it is REQUIRED

doegox, 2020-04-17:
ok I pushed another branch to test without REQUIRED, I'll merge if travis is happy

doegox, 2020-04-17:
ok mine compiled fine, I'll merge your PR with is identical

doegox, 2020-04-17:
Updated status 

- [ ] cmake version 3.16 : is it ok to lower it ? it's already lowered to 3.12.4 in the branch for Travis. For WSL it has to be lowered to 3.10.2
- [ ] fails on OSX, cf Travis builds => readline? https://travis-ci.org/github/RfidResearchGroup/proxmark3/jobs/676109622
- [ ] need to add make install / uninstall 
- [ ] do we need headers in CMakeLists.txt ? everything seems to work fine without them explicitly in TARGET_SOURCES, ok to remove them ?

uhei, 2020-04-17:
Files PR #698 to fix compilation on macOS

bkerler, 2020-04-17:
some headers are needed otherwise it won't compile (especially the deps), at least that was my experience. make install should already work. didn't test make uninstall.

doegox, 2020-04-17:
> some headers are needed otherwise it won't compile (especially the deps), at least that was my experience. make install should already work. didn't test make uninstall.

@bkerler well without headers it compiles fine on linux and OSX, see https://travis-ci.org/github/RfidResearchGroup/proxmark3/builds/676157357 so...

doegox, 2020-04-17:
>  make install should already work. didn't test make uninstall.
Good! Apparently uninstall & cmake is a mess, let's forget about it

doegox, 2020-04-17:
Lowering min cmake version works too https://travis-ci.org/github/RfidResearchGroup/proxmark3 so I'll merge both for now.

doegox, 2020-04-17:
Updated status 

- [x] cmake version 3.16 : lowered to 3.10.2 for WSL
- [x] fails on OSX : fixed 
- [x] need to add make install / uninstall : already ok
- [x] removed headers in CMakeLists.txt
- [ ] add a cmake job on AppVeyor

doegox, 2020-04-17:
@merlokk would it be possible to add a job on appveyor to test the cmake build ? on branch https://github.com/RfidResearchGroup/proxmark3/tree/newcmake

iceman1001, 2020-04-17:
nice nice,
I am not sure I follow with the **make install / make uninstall** is a mess comment and OK status.

doegox, 2020-04-17:
> nice nice,
> I am not sure I follow with the **make install / make uninstall** is a mess comment and OK status.

* make install works already
* make uninstall is a mess to add, cf https://stackoverflow.com/questions/41471620/cmake-support-make-uninstall

iceman1001, 2020-04-17:
we still have **make uninstall** that will work, right?

doegox, 2020-04-17:
the regular one yes of course, see the title of this thread ;)

iceman1001, 2020-04-17:
well,  if hinting to title of this thread mean you can talk about **make install** but i can't talk about ** make uninstall** is a bit unfare

doegox, 2020-04-17:
We talk about make install/uninstall after execution of cmake ;)
The other make install/uninstall is still there untouched.
So today (in this branch) you can do:
```
cd client
make
make install
make uninstall
```
or
```
cd client
mkdir build
cd build
cmake ..
make
make install
```

iceman1001, 2020-04-17:
_dark arts of cmake woodoo_     Its evil.

The question for me is ,  can we run the normal make uninstall  (client/)  after  we did a  (cmake make install)  and get everything removed?    Or are we screwed?

doegox, 2020-04-17:
yes you can as both methods install the same things.

iceman1001, 2020-04-17:
ok, thats good. Now I follow better :)
Lets see when @merlokk  has time to look at appveyor!

doegox, 2020-04-19:
ok, so far, you might have noticed Makefiles and CMakeLists wnet to some rework.
Indeed I tried to make them doing more similar compilation operations for easier future maintenance.
The cmake way of thinking made me also improve stuffs in the historical Makefile.

doegox, 2020-04-19:
ok, so far, you might have noticed Makefiles and CMakeLists wnet to some rework.
Indeed I tried to make them doing more similar compilation operations for easier future maintenance.
The cmake way of thinking made me also improve stuffs in the historical Makefile.

What's left as todo:
- [ ] add a cmake job on AppVeyor
- [ ] add ``-lm -Wl,--as-needed -latomic -Wl,--no-as-needed`` for compilation on Raspberry Pi Zero (it can be added for all gcc platforms as in historical Makefile except for ``APPLE`` or more exactly the clang pretending to be gcc on OSX)
- [ ] think about C standards... historical pm3 Makefile imposes c99/c++11 on most of the code while cmake doesn't impose c99 much (only on deps except lua) and defaults to c11/c++14. I'm not sure we need the c99, unless for a weird platform? I'd be in favor of removing the c99 and wait for someone shouting about broken compilation somewhere... @iceman1001 ?

iceman1001, 2020-04-19:
OSX/*nix/Android/RPi/Windows WSL/Proxspace MingW...  
yes, we support a whole bunch of OS's.  Something to remember,   but hey,  lets remove c99 if the compiler now enforces c11....

iceman1001, 2020-04-22:
With the magic hands of @doegox  the project repo / client builds on RPI Zero with guiless (no QT) and cmake.

iceman1001, 2020-04-22:
And some little fix for WSL aswell,  made it work there nicely.

[1 more reply]

closed 2020-04-22 by doegox