RfidResearchGroup/proxmark3 #705: make clear && make all: outputs *** No rule to make target 'clear'. Stop

Omie2468, 2020-04-20:
I installed Proxspace on Windows as I have been working with the Windows to use the PM3 RDV4 install because on Kali when I run;

make clear && make all

It outputs *** No rule to make target 'clear'. Stop

https://hastebin.com/ibujeduket.makefile
https://hastebin.com/unojoriyeh.makefile
https://hastebin.com/aforiseben.makefile
https://hastebin.com/upubazibix.makefile

These are the 4 makefiles in my directory, from what I can see 'clear' only appears in the host file.

Did I install something wrong? I did use the video that is on the installation page: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

The only difference between the video it seems is that the actual written text doesn't say to do; make clear && make all, but I assume it's needed.

mwalker33, 2020-04-20:
I use both windows (proxspace) and kali and have it working on both.
In kali, can you try this.
1. If you have an existing proxmark3 folder, rename it.
mv proxmark3 proxmark3-old
2. git clone the latest files.
git clone https://github.com/RfidResearchGroup/proxmark3.git
3. change directory into the proxmark3 folder
cd proxmark3
4. build 
make clean && make all

Omie2468, 2020-04-20:
Thanks, got it working, thank God!

closed 2020-04-20 by Omie2468