RfidResearchGroup/proxmark3 #708: CMD_HF_MIFARE_READBL and 14a RAW in appmain.c

Windslab, 2020-04-21:
1. Error after 40 block 'cmds.CMD_HF_MIFARE_READBL'
2. Have no RAW command to send 14a as function in `appmain.c`

Please find code issues at the: https://github.com/RfidResearchGroup/proxmark3/pull/707 How much time you need to apply a Pull? Delete a script if you don't wish to apply.

Ice, you cannot say to me `Not enough of information in your topics`. We are working in Russian style to my posted Issues. I have find bugs and you check the all code where it is. You can't skip my post, that's open source, no admin games.

That's why you don't understend what I writing to you. That's not Google Translite, that's my English and mentalitet of EU people. You never say directly and trying to mute to not convenient questions.

FYI: I not programist, I an Engineer. Programers do code/text of an Engineer told. Hope you understood now.

Regards

doegox, 2020-04-21:
Please calm down @Windslab 

Yes it's open source, which means we, "admins", are just volunteers trying to free some time to work on the project, and as you can see there are 39 open issues and what you don't see there are 98 more open issues in an internal tracker.
So maybe your issues and PRs are not treated as fast as you wish but please realize a lot of people could complain for much older pending issues.

And it's true that it's very hard to understand your communication, which means it takes much more time to investigate your issues and PRs because we never blindly merge PRs without understanding properly what's proposed and the consequences of the integration.

doegox, 2020-04-21:
No I'm trying to understand the ticket you opened here.
Is it to explain a new issue is it just a rehearsal of #707 ?

Windslab, 2020-04-21:
@doegox That's all okay, nothing personal at all! I've seen that @iceman1001 for the first time been very quickly response and friendly  and now hi ignoring me. Thanks for your answer.

Windslab, 2020-04-21:
@doegox please take a look at:

> local function checkkey()
    local status = 0
    for i = 1, #eml do
        cmd = Command:newNG{cmd = cmds.CMD_HF_MIFARE_READBL, data = ('%02s%02s%s'):format((i-1), default_key_type, default_key)}
        if (getblockdata(cmd:sendNG(false)) == true) then
            status = status + 1
            if default_key_type == '00' then
                print(('%s %02s %s %s %s'):format('Block:', (i-1), 'KeyA', default_key, 'OK'))
            else
                print(('%s %02s %s %s %s'):format('Block:', (i-1), 'KeyB', default_key, 'OK'))
            end
        else
            break
        end
    end
    if status == #eml then
        return true
    end
end

This function works only to 39 block and I don't know why. Yes it's by me... mb that's why...

doegox, 2020-04-21:
I'm sorry but I don't know Lua :)
Try to reproduce the same series of commands with pm3 CLI and see if the issue is in your script or in pm3 code.

doegox, 2020-04-21:
```
example = [[
    1. script run mfc_gen3_writer
```
This sounds already as a better filename, why your script is called ``hf_writer`` ? It's a far too generic name

Windslab, 2020-04-21:
> I'm sorry but I don't know Lua :)
> Try to reproduce the same series of commands with pm3 CLI and see if the issue is in your script or in pm3 code.

I can't this to do. I need to know it is a card all 63 sectors has B keys FFFFFFFFFFFF or only 62. In case 63 I honestly know that's I can use B keys to write all sectors with my chosen dump file.

Yeah I don't know any programming language at all. This all code from Google...

Windslab, 2020-04-21:
> ```
> example = [[
>     1. script run mfc_gen3_writer
> ```
> 
> This sounds already as a better filename, why your script is called `hf_writer` ? It's a far too generic name

Yes you right will free to rename, but please after apply my Pull.
But you need include one think if I or somebody will add a suport of MFP or Desfire in future, the script name `mfc_gen3_writer` will sounds not good.

To put it simply, the idea is to make a universal dump manager. Write/Save data to cards.

doegox, 2020-04-21:
it will be time to change to mf_writer or sth like that if indeed one day it supports mfp and df, but
``hf_ > hf_14a_ > mf_ > mfc_`` and for now it only covers some mfc

doegox, 2020-04-21:
(I agree current scripts names are a mess and should be changed too)

Windslab, 2020-04-21:
> it will be time to change to mf_writer or sth like that if indeed one day it supports mfp and df, but
> `hf_ > hf_14a_ > mf_ > mfc_` and for now it only covers some mfc

> (I agree current scripts names are a mess and should be changed too)

No problem! You're welcome.

Yes good, but first think for us, this is do the `checkkey()` working correctly. I hope for bug in C++ code. Will see.

doegox, 2020-04-21:
how to trigger the bug ? I tried to write a dump on a 1k card and it worked without bug

Windslab, 2020-04-21:
> how to trigger the bug ? I tried to write a dump on a 1k card and it worked without bug

It's can be trigger at this moment:
```
----------------------------------------------------------------
 Permanent lock UID ? (card can never change uid again)  [y/n] ?n
----------------------------------------------------------------
Block: 00 KeyB FFFFFFFFFFFF OK
Block: 01 KeyB FFFFFFFFFFFF OK
Block: 02 KeyB FFFFFFFFFFFF OK
Block: 03 KeyB FFFFFFFFFFFF OK
Block: 04 KeyB FFFFFFFFFFFF OK
Block: 05 KeyB FFFFFFFFFFFF OK
Block: 06 KeyB FFFFFFFFFFFF OK
Block: 07 KeyB FFFFFFFFFFFF OK
Block: 08 KeyB FFFFFFFFFFFF OK
Block: 09 KeyB FFFFFFFFFFFF OK
Block: 10 KeyB FFFFFFFFFFFF OK
Block: 11 KeyB FFFFFFFFFFFF OK
Block: 12 KeyB FFFFFFFFFFFF OK
Block: 13 KeyB FFFFFFFFFFFF OK
Block: 14 KeyB FFFFFFFFFFFF OK
Block: 15 KeyB FFFFFFFFFFFF OK
Block: 16 KeyB FFFFFFFFFFFF OK
Block: 17 KeyB FFFFFFFFFFFF OK
Block: 18 KeyB FFFFFFFFFFFF OK
Block: 19 KeyB FFFFFFFFFFFF OK
Block: 20 KeyB FFFFFFFFFFFF OK
Block: 21 KeyB FFFFFFFFFFFF OK
Block: 22 KeyB FFFFFFFFFFFF OK
Block: 23 KeyB FFFFFFFFFFFF OK
Block: 24 KeyB FFFFFFFFFFFF OK
Block: 25 KeyB FFFFFFFFFFFF OK
Block: 26 KeyB FFFFFFFFFFFF OK
Block: 27 KeyB FFFFFFFFFFFF OK
Block: 28 KeyB FFFFFFFFFFFF OK
Block: 29 KeyB FFFFFFFFFFFF OK
Block: 30 KeyB FFFFFFFFFFFF OK
Block: 31 KeyB FFFFFFFFFFFF OK
Block: 32 KeyB FFFFFFFFFFFF OK
Block: 33 KeyB FFFFFFFFFFFF OK
Block: 34 KeyB FFFFFFFFFFFF OK
Block: 35 KeyB FFFFFFFFFFFF OK
Block: 36 KeyB FFFFFFFFFFFF OK
Block: 37 KeyB FFFFFFFFFFFF OK
Block: 38 KeyB FFFFFFFFFFFF OK
Block: 39 KeyB FFFFFFFFFFFF OK
#db# Auth error
```

You can see that function can get only until 39 Blocks. Another 25 is unavailable - that's bug.

Windslab, 2020-04-21:
PS: Ofcourse a card must be blank

doegox, 2020-04-21:
Show me all the way, not just the dump of the bug.
Give me the means to reproduce the bug.

doegox, 2020-04-21:
```
[usb] pm3 --> script run mfc_gen3_writer
[+] executing lua /home/phil/devel/00smartcard/readers/proxmark/proxmark3-RRG.git/client/luascripts/mfc_gen3_writer.lua           
[+] args ''           
Waiting for card... press Enter to quit
----------------------------------------------------------------
 1 | ass/elit
 2 | ass/ciph
 3 | ass/elit
 4 | ass/ciph
 5 | crypto_p
 6 | test/cry
 7 | test/cry
 8 | crypto_p
 9 | arser/cl
 10 | arser/ar
 11 | arser/ar
 12 | arser/cl
 13 | and_form
 14 | and_form
 15 | s/iclass
 16 | ass/hash
 17 | ass/elit
 18 | ass/ciph
 19 | ass/hash
 20 | ass/elit
 21 | ass/ciph
 22 | test/cry
 23 | test/cry
 24 | crypto_p
 25 | and_form
 26 | and_form
 27 | ts/test_
 28 | ts/calc_
 29 | ts/dumpt
 30 | ts/mifar
 31 | ts/param
 32 | sson/obj
 33 | sson/obj
 34 | sson/lib
 35 | itool/li
 36 | lua/obj/
 37 | lua/obj/
 38 | lua/obj/
 39 | lua/obj/
 40 | lua/obj/
 41 | lua/obj/
 42 | lua/obj/
 43 | lua/obj/
 44 | lua/obj/
 45 | lua/obj/
 46 | lua/obj/
 47 | lua/obj/
 48 | lua/obj/
 49 | lua/obj/
 50 | lua/obj/
 51 | lua/obj/
 52 | lua/obj/
 53 | lua/obj/
 54 | ne/lf_pr
 55 | ne/lf_em
 56 | on-ask-m
 57 | on-manch
 58 | on-ask-m
 59 | on-ask-m
 60 | HIDemu-F
 61 | ypatch-m
 62 | applypat
 63 | remotes/
 64 | remotes/
 65 | heads/mf
 66 | heads/fp
 67 | brute/.g
 68 | c06_cons
 69 | ck/crack
 70 | /obj/buc
 71 | /obj/buc
 72 | cipher_i
find: ‘*dump.eml’: No such file or directory
 73 | tion_Ins
----------------------------------------------------------------
 Your card has UID 01020304

 Select which dump to write (1 until 73)
----------------------------------------------------------------
 -->     

```
Hmm that's a bug....

Windslab, 2020-04-21:
Script works only with Windows. It's looks like you started on *NIX.

The multiple operating system also my purpose.

Windslab, 2020-04-21:
> Show me all the way, not just the dump of the bug.
> Give me the means to reproduce the bug.

Just put a blank(keys must be FFFFFFFFFFFF) mifare 1k card to your Proxmark and choice a dump. You got it.

doegox, 2020-04-21:
ok, I don't run Windows I don't know Lua, I can't help

Windslab, 2020-04-21:
> ok, I don't run Windows I don't know Lua, I can't help

I have appreciate your honestly, trying and opening. Thank's.

mwalker33, 2020-04-22:
In the checkkey function, check the formatting options.
It looks like you are formatting the options as strings, not numbers.

current
```
cmd = Command:newNG{cmd = cmds.CMD_HF_MIFARE_READBL, data = ('%02s%02s%06s'):format((i-1), default_key_type, default_key)}
```
so 40 in hex = 64 dec which is past the end of the number of blocks.

try something like
```
cmd = Command:newNG{cmd = cmds.CMD_HF_MIFARE_READBL, data = ('%02x%02x%06s'):format((i-1), default_key_type, default_key)}
```

Windslab, 2020-04-22:
@mwalker33 Your code blocks is the same. What is the code you have prepared?

Windslab, 2020-04-22:
%02x%02x%06s - is working untill 63 sector. @mwalker33 thx

closed 2020-04-22 by Windslab