RfidResearchGroup/proxmark3 #711: [request]

Windslab, 2020-04-23:
Do any bodey can polism my code? Why it does not work? Just wont to check if card do any response or not. That's for our script for gen3. Many Thanks.

`local cmds = require('commands')
local lib14a = require('read14a')
local rawdata = '90f0cccc10'

function showdata(usbpacket)
    local cmd_response = Command.parse(usbpacket)
    local len = tonumber(cmd_response.arg1) *2
    local data = string.sub(tostring(cmd_response.data), 0, len);
    print("<< ",data)
end

function sendRaw(rawdata)
    print('>> ', rawdata)
    local flags = lib14a.ISO14A_COMMAND.ISO14A_NO_DISCONNECT + lib14a.ISO14A_COMMAND.ISO14A_RAW
    local command = Command:newMIX{cmd = cmds.CMD_HF_ISO14443A_READER, arg1 = flags, arg2 = string.len(rawdata)/2, data = rawdata}
    return command:sendMIX()
end

res = sendRaw(rawdata)
showdata(res)`

Windslab, 2020-04-23:
Chr, could you do? What wrong. Up.

Windslab, 2020-04-23:
Yes, you can do what you want. That's your Business not my. I am there for fun.

You trying to trolling me that's you don't know Lua and doing some of changers at C code. Haha, I have appreciated your answer. Difficult to work with you, you no fix it, you just move it to trash.

That's why I will stop at this moment. Wishing you the best.

doegox, 2020-04-23:
> You trying to trolling me that's you don't know Lua and doing some of changers at C code.

I'm not trolling at all, show me one single commit where I did a significant change in a Lua script in the whole repo history.
I know C and I don't know Lua, as simple as that.

Windslab, 2020-04-23:
> I'm not trolling at all, show me one single commit where I did a significant change in a Lua script in the whole repo history.
Yes seems that's my great ignorance. I can't get used that you can't drive a bike but every day using motorcycle. It's sounds like "Hey I am pro coder to Delphi" but I don't know Pascal, "The all questions about Pascal not to me". LMAO! That's differences between us about I always talking...

If you not trolling, please get back the script to his place and trying to find some of answer to my post. Then it will be done and ready to use. It's script use as well by Russians people. You better more write, less doing.

closed 2020-04-23 by iceman1001

Windslab, 2020-04-23:
Very fair deed @iceman1001 you even don't gave chanse to answer @doegox 

In this case, one team with the same peoples. That's not good.