RfidResearchGroup/proxmark3 #714: [Not a real bug, but not a feature request] Building Termux package with recent reworks/refactors

msoose, 2020-04-24:
**Describe the bug**
There were a few changes to dependencies and architecture detection made recently, I have had to re-work my patch for Termux, but am unable to compile the latest master, but I think this is related to recent changes. Any assistance would be appreciated. Maybe we can work on setting the dependencies and Makefiles to be more platform agnostic so no patches are required to prepare a package for Termux.

**To Reproduce**
Steps to reproduce the behavior:
1. Cross compile for Termux using **clang**
2. Get error while linking binary

`clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:289: proxmark3] Error 1
`

`obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'CLIParserInit'                                                                                                                          
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'arg_lit0'                                                                                                                               
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'arg_strn'              
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'arg_end'               
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'CLIParserParseString'                                                                                                                   
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'CLIParserFree'                                                                                                                          
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'CLIParamHexToBuf'          
obj/cmddata.o:cmddata.c:function CmdDataNDEF: error: undefined reference to 'CLIParserFree'             
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVExec: error: undefined reference to 'CLIParserInit'                                                                                                                         
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVExec: error: undefined reference to 'arg_lit0'                                                                                                                              
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVExec: error: undefined reference to 'arg_lit0'                                                                                                                              
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVExec: error: undefined reference to 'CLIParserParseString' 
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVExec: error: undefined reference to 'CLIParserFree'                                                                                                                         
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVPPSE: error: undefined reference to 'CLIParserInit'             
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVPPSE: error: undefined reference to 'arg_lit0'                  
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVPPSE: error: undefined reference to 'CLIParserParseString'      
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVPPSE: error: undefined reference to 'CLIParserFree'                                                                                                                         
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVSearch: error: undefined reference to 'CLIParserInit'                                                                                                                       
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVSelect: error: undefined reference to 'CLIParamHexToBuf'                                                                                                                    
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVGPO: error: undefined reference to 'CLIParamHexToBuf'                                                                                                                       
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVReadRecord: error: undefined reference to 'CLIParamHexToBuf'                                                                                                                
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVAC: error: undefined reference to 'arg_str0'                                                                                                                                
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVScan: error: undefined reference to 'arg_str1'                                                                                                                              
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVScan: error: undefined reference to 'CLIParserParseString'                                                                                                                  
obj/emv/cmdemv.o:cmdemv.c:function CmdEMVScan: error: undefined reference to 'CLIParamStrToBuf'                                                                                                                      
obj/emv/cmdemv.o:cmdemv.c:function OUTLINED_FUNCTION_23: error: undefined reference to 'arg_end'                                                                                                                     
obj/emv/cmdemv.o:cmdemv.c:function OUTLINED_FUNCTION_48: error: undefined reference to 'arg_rem'                                                                                                                     
obj/emv/cmdemv.o:cmdemv.c:function OUTLINED_FUNCTION_59: error: undefined reference to 'arg_strn'
obj/emv/cmdemv.o:cmdemv.c:function OUTLINED_FUNCTION_65: error: undefined reference to 'arg_strn'
obj/cmdhf.o:cmdhf.c:function CmdHFPlot: error: undefined reference to 'arg_end'                    
obj/cmdhf14a.o:cmdhf14a.c:function CmdHF14AAPDU: error: undefined reference to 'arg_str0'          
`

**Expected behavior**
Clean compile when using clag, with AVX512 disabled

**Screenshots**
N/A
**Desktop (please complete the following information):**
 - OS: Android / Termux
- Cross-compiled using Termux docker image which uses clang
**Additional context**
Can provide build-time patch if requested

doegox, 2020-04-24:
Hello
Yes we tried to restructure a few things in the historical Makefile at the same time a cmake was added for the client, trying to bridge the gap between both approaches and getting the same behavior on all platforms.
Proxmark3 is tested on a large number of platforms and it was quite challenging.
Now, you're bringing one more platform we never tried before so I'm not surprised things have been shoved a bit on your side.

I'm interested, please share details.
Probably the easiest would be to drive me to a step-by-step to the Termux docker setup.

Note that I tried to compile the client on my linux with clang and it worked fine.

doegox, 2020-04-24:
from your error message, the linker doesn  find the cliparser library. It has moved to client/deps/cliparer, are you sure your patch compiles it properly and provides the path to the linker ?

msoose, 2020-04-24:
> from your error message, the linker doesn't find the cliparser library. It has moved to client/deps/cliparer, are you sure your patch compiles it properly and provides the path to the linker ?

Ah, let me check this right now. I should have read the error closer.

msoose, 2020-04-24:
Ah, I figured it out. All dependencies inside `client/deps` need to be built and linked now. Thanks for your help there. I'll give a quick reply to your previous questions.

Termux provides a Docker setup for a quick cross compile environment for submitting binary packages. The following patch is applied and it's built for repositories, it just forces some variables, paths, links, etc...

Perhaps some of the required changes below can assist with any future restructures to help streamline things?

```
diff -uNr proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/deps/hardnested/Makefile proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/deps/hardnested/Makefile
--- proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/deps/hardnested/Makefile  2020-04-24 16:21:38.000000000 +0000
+++ proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/deps/hardnested/Makefile      2020-04-24 17:37:26.500726041 +0000
@@ -4,7 +4,7 @@
 MYDEFS =
 MYSRCS = hardnested_bruteforce.c
 
-cpu_arch = $(shell uname -m)
+cpu_arch = generic
 ifneq ($(findstring 86, $(cpu_arch)), )
     MULTIARCHSRCS = hardnested_bf_core.c hardnested_bitarray_core.c
 endif
@@ -24,7 +24,7 @@
             $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX.o) \
             $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)
 
-SUPPORTS_AVX512 :=  $(shell echo | gcc -E -mavx512f - > /dev/null 2>&1 && echo "True" )
+SUPPORTS_AVX512 := False
 
 HARD_SWITCH_NOSIMD = -mno-mmx -mno-sse2 -mno-avx -mno-avx2
 HARD_SWITCH_MMX = -mmmx -mno-sse2 -mno-avx -mno-avx2
diff -uNr proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/deps/reveng/Makefile proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/deps/reveng/Makefile
--- proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/deps/reveng/Makefile      2020-04-24 16:21:38.000000000 +0000
+++ proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/deps/reveng/Makefile  2020-04-24 17:42:33.854155450 +0000
@@ -30,6 +30,5 @@
        $(info [-] CC $<)
        $(Q)$(CC) $(CFLAGS) -DBMPTST -o $@ $<
        $(info [=] TEST $@)
-       $(Q)( ./$@ && $(TOUCH) $@ ) || ( $(RM) $@ && $(FALSE) )
 
 .PHONY: all clean
diff -uNr proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/Makefile proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/Makefile
--- proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/client/Makefile  2020-04-24 16:21:38.000000000 +0000
+++ proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/client/Makefile      2020-04-24 17:41:26.768980474 +0000
@@ -22,7 +22,7 @@
 OBJDIR = obj
 
 LDLIBS ?= -L/usr/local/lib
-LDLIBS += -lreadline -lpthread -lm
+LDLIBS += -lreadline -lm
 
 # RPi Zero gcc requires -latomic
 # but MacOSX /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld
@@ -54,7 +54,7 @@
 ZLIB = $(OBJDIR)/libz.a
 
 LIBS = -I$(LUALIBPATH) -I$(MBEDTLSLIBPATH) -I$(JANSSONLIBPATH) -I$(CBORLIBPATH) -I$(ZLIBPATH) -I$(REVENGPATH) -I$(AMIIBOLIBPATH) -I$(HARDNESTEDPATH) -I$(CLIPARSERPATH)
-INCLUDES_CLIENT = -I./src -I../include -I../common -I../common_fpga $(LIBS)
+INCLUDES_CLIENT = -I./src -I../include -I../common -I../common_fpga -I/data/data/com.termux/files/usr/include $(LIBS)
 CFLAGS ?= -Wall -Werror -O3
 # We cannot just use CFLAGS+=... because it has impact on sub-makes if CFLAGS is defined in env:
 PM3CFLAGS = $(CFLAGS) $(INCLUDES_CLIENT)
@@ -286,7 +286,7 @@
 proxmark3: LDLIBS+=$(LUALIB) $(JANSSONLIB) $(MBEDTLSLIB) $(CBORLIB) $(ZLIB) $(REVENGLIB) $(AMIIBOLIB) $(HARDNESTEDLIB) $(CLIPARSERLIB) $(QTLDLIBS)
 proxmark3: $(OBJDIR)/proxmark3.o $(COREOBJS) $(CMDOBJS) $(OBJCOBJS) $(QTGUIOBJS) $(LUALIB) $(JANSSONLIB) $(CBORLIB) $(REVENGLIB) $(MBEDTLSLIB) $(ZLIB) $(AMIIBOLIB) $(HARDNESTEDLIB) $(CLIPARSERLIB) lualibs/pm3_cmd.lua lualibs/mfc_default_keys.lua
        $(info [=] LD $@)
-       $(Q)$(LD) $(LDFLAGS) $(OBJDIR)/proxmark3.o $(COREOBJS) $(CMDOBJS) $(OBJCOBJS) $(QTGUIOBJS) $(LDLIBS)  -o $@
+       $(Q)$(LD) $(LDFLAGS) $(OBJDIR)/proxmark3.o $(COREOBJS) $(CMDOBJS) $(OBJCOBJS) $(QTGUIOBJS) $(LDLIBS)  -o $@ deps/hardnested/libhardnested.a deps/cliparser/libcliparser.a deps/tinycbor/tinycbor.a deps/reveng/libreveng.a deps/jansson/libjansson.a deps/liblua/liblua.a obj/libmbedtls.a obj/libz.a
 
 src/proxgui.cpp: src/ui/ui_overlays.h
 
diff -uNr proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/Makefile.defs proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/Makefile.defs
--- proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04/Makefile.defs    2020-04-24 16:21:38.000000000 +0000
+++ proxmark3-011f973e7ff145ee0f6ccf58c850765513c43d04.mod/Makefile.defs        2020-04-24 17:42:53.202493467 +0000
@@ -21,9 +21,6 @@
 TARFLAGS ?= -v --ignore-failed-read -r
 TARFLAGS += -C .. -f
 CROSS  ?= arm-none-eabi-
-CC = gcc
-CXX = g++
-LD = g++
 
 PATHSEP=/
 PREFIX ?=              /usr/local

```

doegox, 2020-04-24:
hmm some changes seem challenging to integrate, especially if we can't probe which arch we're compiling to. But maybe via some overwritable vars we can do it.
For the Makefile.defs for example, no need to change it, you can do
```
make client CC=clang CXX=clang++ LD=clang++
```

doegox, 2020-04-24:
```diff
-       $(Q)$(LD) $(LDFLAGS) $(OBJDIR)/proxmark3.o $(COREOBJS) $(CMDOBJS) $(OBJCOBJS) $(QTGUIOBJS) $(LDLIBS)  -o $@
+       $(Q)$(LD) $(LDFLAGS) $(OBJDIR)/proxmark3.o $(COREOBJS) $(CMDOBJS) $(OBJCOBJS) $(QTGUIOBJS) $(LDLIBS)  -o $@ deps/hardnested/libhardnested.a deps/cliparser/libcliparser.a deps/tinycbor/tinycbor.a deps/reveng/libreveng.a deps/jansson/libjansson.a deps/liblua/liblua.a obj/libmbedtls.a obj/libz.a
```
That shouldn't be needed, please try to find the root cause.
The libs are already part of `$(LDLIBS)` as per
```
LDLIBS+=$(LUALIB) $(JANSSONLIB) $(MBEDTLSLIB) $(CBORLIB) $(ZLIB) $(REVENGLIB) $(AMIIBOLIB) $(HARDNESTEDLIB) $(CLIPARSERLIB) $(QTLDLIBS)
```
with e.g.
```
CLIPARSERPATH = ./deps/cliparser
CLIPARSERLIB = $(CLIPARSERPATH)/libcliparser.a
```
Is it a problem of order of the arguments on the command line ?

msoose, 2020-04-24:
Just a quick fixup patch to get the package building again and in the repository.

doegox, 2020-04-24:
for the additional includes you can also solve it out of the patch by overwriting CFLAGS from the command line and adding the extra includes

doegox, 2020-04-24:
about reveng `$(Q)( ./$@ && $(TOUCH) $@ ) || ( $(RM) $@ && $(FALSE) )` it's there for a reason. If bmptst returned an error it means we've to add sth to config.h to take into account the Termux platform, else reveng won't work as expected.

doegox, 2020-04-24:
for `SUPPORTS_AVX512 :=  $(shell echo | gcc -E -mavx512f - > /dev/null 2>&1 && echo "True" )` there was a bug (forcing gcc instead of $(CC)).
I pushed a fix

doegox, 2020-04-25:
could you show how I can compile myself pm3 under termux ?

msoose, 2020-04-30:
https://github.com/termux/termux-packages/wiki/Building-packages
https://github.com/termux/termux-packages/tree/master/packages/proxmark3

You can grab the entire repo, Docker in, and my existing build script and patch will be used if you `./build-package proxmark3` 

Any simply and straight forward changes that can be integrated in order to reduce the patching would be appreciated.

doegox, 2020-04-30:
Thanks @msoose 
Here is a patch to use with `git am`.
Sorry I had to zip it to be accepted for upload.

[0001-proxmark3-update-to-last-master-and-simplify-build-s.zip](https://github.com/RfidResearchGroup/proxmark3/files/4561718/0001-proxmark3-update-to-last-master-and-simplify-build-s.zip)

Please test.
If any questions/remarks, don't hesitate.

doegox, 2020-04-30:
(the only patch that remains is for pthread, which seems a quite common patch for other termux packages as well)

iceman1001, 2020-05-08:
ping.  What is that status on this one? @msoose

msoose, 2020-05-10:
Thanks for the changes, should really simplify things. I'm hoping to get these in to the Termux package this coming week.

doegox, 2020-05-15:
I pushed a change in the Makefile. Now you can use `SKIPPTHREAD=1`.
So no need for the patch anymore, just do
```
termux_step_make() {
    LDLIBS="-L$TERMUX_PREFIX/lib" INCLUDES_CLIENT="-I/data/data/com.termux/files/usr/include" \
    make client -j $TERMUX_MAKE_PROCESSES CC=$CC CXX=$CXX LD=$CXX cpu_arch=generic SKIPREVENGTEST=1 SKIPQT=1 SKIPBT=1 SKIPPTHREAD=1
}
```

iceman1001, 2020-05-17:
Great adjustments @doegox , 
@msoose  , waiting for your tests then we can close this one.

msoose, 2020-05-17:
Passes all build tests, thanks for your work on this @doegox, will make maintaining the package a lot easier.

closed 2020-05-17 by msoose

doegox, 2020-05-17:
great :)

doegox, 2020-05-21:
poking @msoose 
we got a tagged rel, you might consider updating termux build

msoose, 2020-05-21:
@doegox thanks for that, will push an update that uses the releases, I'll probably stick to on-device compiling.