RfidResearchGroup/proxmark3 #715: macOS/Homebrew make failure

KaitoKid, 2020-04-24:
Fresh installation of Homebrew and tried to install proxmark, but running into this error on `make`
[image]

doegox, 2020-04-24:
The mentioned lines are not part of the Proxmark source code, it's hard to debug based on that info.

can you run 
```
make clean
make V=1 PLATFORM=PM3RDV4 2>&1 |tee catalina_log.txt
```
and provide the logfile ?
thanks!

doegox, 2020-04-24:
I can't reproduce the issue on Catalina 10.15
Which instructions did you follow from
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md ?

KaitoKid, 2020-04-24:
It's strange since I was able to build it manually, but if I ran the following I would get that error:

1. `brew tap RfidResearchGroup/proxmark3`
2. `brew install --HEAD proxmark3`

This is a fresh installation of Catalina and Homebrew as well, and only seems to come up when I try to install through homebrew.

doegox, 2020-04-24:
hmm indeed I can reproduce now.
somehow through the formula, it seems to see qt and fail linking it while manually it compiles without gui. I'll try to investigate

doegox, 2020-04-24:
My guess is that qt in brew is currently somehow broken:
```
brew install qt
pkg-config --libs Qt5Core
Package Qt5Core was not found in the pkg-config search path.
Perhaps you should add the directory containing `Qt5Core.pc'
to the PKG_CONFIG_PATH environment variable
No package 'Qt5Core' found
```
and still `/usr/local/opt/qt5/lib/pkgconfig/Qt5Core.pc` exists

doegox, 2020-04-24:
ok some progress : indeed brew qt changed, we've now to export explictely qt path because apparently they don't expose it systematically for some cmake issue.
So now with `export PKG_CONFIG_PATH="/usr/local/opt/qt/lib/pkgconfig"` I can reproduce the issue manually.

doegox, 2020-04-24:
Try again `brew install --HEAD proxmark3`
It should work now.

KaitoKid, 2020-04-25:
Looks like ad3267c fixed it! Closing this out

closed 2020-04-25 by KaitoKid