RfidResearchGroup/proxmark3 #716: Proxmark3 (RfidResearchGroup firmware does not work on Ubuntu 20.04)

2rsharipov, 2020-04-25:
Hello!

I reinstalled the operating system on Ubuntu 20.04
And I followed the instructions for installing and configuring my device.
But I got problems again.

1.Update
sudo apt-get update

2.Requirements
sudo apt-get install --no-install-recommends git ca-certificates build-essential pkg-config \
libreadline-dev gcc-arm-none-eabi libnewlib-dev qtbase5-dev

3.Clone Fork
git clone https://github.com/RfidResearchGroup/proxmark3.git

4.Change Directory
cd proxmark3

5.Get the latest commits
git pull

6.Install the blacklist rules and add user to dialout group.
sudo make udev

7.Restart
Restart Ubuntu

8.Enter proxmark folder
cd proxmark3

9.Create a file Makefile.platform in the root directory "proxmark3" of the repository:
```
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

PLATFORM=PM3OTHER
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
#STANDALONE=LF_SAMYRUN
#STANDALONE=LF_ICEHID

```
10.Clean and Compile
make clean && make all

11.Check Connection
Once this is complete run the following comands to make sure the proxmark is being picked up by your computer.
sudo dmesg | grep -i usb
```
...
[   35.299395] usb 3-4: new full-speed USB device number 3 using xhci_hcd
[   37.215038] usb 3-4: new full-speed USB device number 4 using xhci_hcd
[   37.365603] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[   37.365608] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   37.365611] usb 3-4: Product: proxmark3
[   37.365613] usb 3-4: Manufacturer: proxmark.org
[   37.365615] usb 3-4: SerialNumber: iceman
[   37.391689] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[   37.393631] usbcore: registered new interface driver cdc_acm
[   37.393633] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  107.440392] usb 3-4: USB disconnect, device number 4
[  150.000977] usb 3-4: new full-speed USB device number 5 using xhci_hcd
[  155.295229] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[  155.295234] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  155.295237] usb 3-4: Product: proxmark3
[  155.295239] usb 3-4: Manufacturer: proxmark.org
[  155.295241] usb 3-4: SerialNumber: iceman
[  155.296529] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[  252.417793] usb 3-4: USB disconnect, device number 5
[  323.289034] usb 3-4: new full-speed USB device number 7 using xhci_hcd
[  323.439763] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[  323.439769] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  323.439771] usb 3-4: Product: proxmark3
[  323.439774] usb 3-4: Manufacturer: proxmark.org
[  323.439775] usb 3-4: SerialNumber: iceman
[  323.441616] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[  329.844195] usb 3-4: USB disconnect, device number 7
[  378.848982] usb 3-4: new full-speed USB device number 9 using xhci_hcd
[  378.999687] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[  378.999692] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  378.999696] usb 3-4: Product: proxmark3
[  378.999698] usb 3-4: Manufacturer: proxmark.org
[  378.999701] usb 3-4: SerialNumber: iceman
[  379.000986] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[  511.453785] usb 3-4: USB disconnect, device number 9
[  516.036892] usb 3-4: new full-speed USB device number 10 using xhci_hcd
[  516.187787] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[  516.187792] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  516.187794] usb 3-4: Product: proxmark3
[  516.187797] usb 3-4: Manufacturer: proxmark.org
[  516.187799] usb 3-4: SerialNumber: iceman
[  516.189446] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
```

12.Remove Modem Manager
sudo apt remove modemmanager

13.Restart
Restart Ubuntu

14.Enter proxmark folder
cd proxmark3

15.Flash the BOOTROM & FULLIMAGE
./pm3-flash-bootrom
```
administrator@adminpc:~/proxmark3$ ./pm3-flash-bootrom
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/administrator/.proxmark3/log_20200425.txt
[+] About to use the following file:          
[+]     /home/administrator/proxmark3/client/../bootrom/obj/bootrom.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0          
 🕐 ............................................................          
[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0
          
[!!] 🚨 Could not find Proxmark3 on /dev/ttyACM0.
          
[!!] 🚨 Aborted on error.          

Have a nice day!          
```

./pm3-flash-fullimage
```
administrator@adminpc:~/proxmark3$ ./pm3-flash-fullimage
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/administrator/.proxmark3/log_20200425.txt
[+] About to use the following file:          
[+]     /home/administrator/proxmark3/client/../armsrc/obj/fullimage.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0          
 🕐 ............................................................          
[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0
          
[!!] 🚨 Could not find Proxmark3 on /dev/ttyACM0.
          
[!!] 🚨 Aborted on error.          

Have a nice day!          
```

15.Run the client
./pm3
```
administrator@adminpc:~/proxmark3$ ./pm3
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/administrator/.proxmark3/log_20200425.txt
[=] Using UART port /dev/ttyACM0  
```        
          
[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0

**I really need your help in solving the problem!**

iceman1001, 2020-04-25:
Try the one-button technique.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md#determine-if-the-bootloader-was-damaged-or-only-the-main-os-image

2rsharipov, 2020-04-25:
Dear **iceman1001**!

I thank you for your prompt reply.
I myself have empirically found a solution to my problem.
It was enough to raise privileges using before executing the command **"sudo"**
Maybe you consider it necessary to add this information to the problem solving section.

closed 2020-04-25 by 2rsharipov