RfidResearchGroup/proxmark3 #717: Proxmark3 (Clone Mifare Ultralight)

2rsharipov, 2020-04-25:
Hello!

1. I put tag on Proxmark3 **from** which I want to copy (**Tag1**)

2. I run command - **hf mfu info**:
```
[usb] pm3 --> hf mfu info
          
[=] --- Tag Information ---------          
[=] -------------------------------------------------------------          
[+]       TYPE: MIFARE Ultralight (MF0ICU1)            
[+]        UID: 34 A7 2E 21 B4 92 60           
[+]     UID[0]: 34, Mikron JSC Russia          
[+]       BCC0: 35 ( ok )          
[+]       BCC1: 67 ( ok )          
[+]   Internal: 0B ( not default )          
[+]       Lock: 78 80 - 1128          
[+] OneTimePad: 4F 45 05 6E - 765110
```

3. I put tag on Proxmark3 **to** which I want to copy (**Tag2**)

4. I run command - [b]hf mfu info[/b]:
```
[usb] pm3 --> hf mfu info
          
[=] --- Tag Information ---------          
[=] -------------------------------------------------------------          
[+]       TYPE: MIFARE Ultralight C (MF0ULC)            
[+]        UID: 04 DE 9D 82 F0 5C 80           
[+]     UID[0]: 04, NXP Semiconductors Germany          
[+]       BCC0: CF ( ok )          
[+]       BCC1: AE ( ok )          
[+]   Internal: 48 ( default )          
[+]       Lock: 78 80 - 1128          
[+] OneTimePad: 4F 45 05 6E - 765110
          

--- UL-C Configuration           
 Higher Lockbits [40/0x28]: 00 00 00 00 - 00          
         Counter [41/0x29]: 00 00 00 00 - 00          
           Auth0 [42/0x2A]: 30 00 00 00 page 48/0x30 and above need authentication          
           Auth1 [43/0x2B]: 00 00 00 00 read and write access restricted          
[=] Trying some default 3des keys          
#db# Cmd Error: 00          
#db# Authentication failed          
#db# Cmd Error: 00          
#db# Authentication failed          
#db# Cmd Error: 00          
#db# Authentication failed          
[+] Found default 3des key:           
         deskey1 [44/0x2C]: 42 52 45 41 [BREA]          
         deskey1 [45/0x2D]: 4B 4D 45 49 [KMEI]          
         deskey2 [46/0x2E]: 46 59 4F 55 [FYOU]          
         deskey2 [47/0x2F]: 43 41 4E 21 [CAN!]          

 3des key: 49 45 4D 4B 41 45 52 42 21 4E 41 43 55 4F 59 46          
[usb] pm3 --> hf mfu info
          
[=] --- Tag Information ---------          
[=] -------------------------------------------------------------          
[+]       TYPE: MIFARE Ultralight (MF0ICU1)            
[+]        UID: 34 A7 2E 21 B4 92 60           
[+]     UID[0]: 34, Mikron JSC Russia          
[+]       BCC0: 35 ( ok )          
[+]       BCC1: 67 ( ok )          
[+]   Internal: 0B ( not default )          
[+]       Lock: 78 80 - 1128          
[+] OneTimePad: 4F 45 05 6E - 765110
```

5. I put **Tag1** on Proxmark3

6. I run command - **hf mfu dump**:
```
[usb] pm3 --> hf mfu dump
[+] TYPE: MIFARE Ultralight (MF0ICU1)            
[+] Reading tag memory...          

*special* data
          

DataType  | Data                    | Ascii          
----------+-------------------------+---------          
Version   | 00 00 00 00 00 00 00 00| ........          
TBD       | 00 00                   | ..          
TBD       | 00                      | .          
Signature1| 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00| ................          
Signature2| 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00| ................          
Counter0  | 00 00 00                | ...          
Tearing0  | 00                      | .          
Counter1  | 00 00 00                | ...          
Tearing1  | 00                      | .          
Counter2  | 00 00 00                | ...          
Tearing2  | 00                      | .          
-------------------------------------------------------------          

Block#   | Data        |lck| Ascii          
---------+-------------+---+------          
  0/0x00 | 34 A7 2E 35|   | 4..5          
  1/0x01 | 21 B4 92 60|   | !..`          
  2/0x02 | 67 0B 78 80|   | g.x.          
  3/0x03 | 4F 45 05 6E| 1 | OE.n          
  4/0x04 | F8 DF 62 A1| 1 | ..b.          
  5/0x05 | 47 87 6A CC| 1 | G.j.          
  6/0x06 | 48 42 69 90| 1 | HBi.          
  7/0x07 | 00 00 00 00| 0 | ....          
  8/0x08 | 00 00 00 00| 0 | ....          
  9/0x09 | 00 00 00 00| 0 | ....          
 10/0x0A | 00 00 00 00| 0 | ....          
 11/0x0B | 00 00 00 00| 0 | ....          
 12/0x0C | 00 00 00 00| 0 | ....          
 13/0x0D | 00 00 00 00| 0 | ....          
 14/0x0E | 00 00 00 00| 0 | ....          
 15/0x0F | 00 00 00 00| 1 | ....          
---------------------------------          
[=] Using UID as filename          
[+] saved 120 bytes to binary file hf-mfu-34A72E21B49260-dump.bin           
[+] saved to json file hf-mfu-34A72E21B49260-dump.json           

```

7. I run command - **hf mfu eload u hf-mfu-34A72E21B49260-dump.eml**:
```
[usb] pm3 --> hf mfu eload u hf-mfu-34A72E21B49260-dump.eml
[+] loaded 120 bytes from text file hf-mfu-34A72E21B49260-dump.eml           
[=] Uploading to emulator memory          
..............................
          
[-]  ⛔ Warning, Ultralight/Ntag file content, Loaded 30 blocks into emulator memory   
```       

8. I put **Tag2** on Proxmark3

9. I run command - [b]hf mfu sim t 7 u hf-mfu-34A72E21B49260-dump.eml[/b]:
```
[usb] pm3 --> hf mfu sim t 7 u hf-mfu-34A72E21B49260-dump.eml

 Emulating ISO/IEC 14443 type A tag with 4,7 byte UID
          
Usage: hf 14a sim [h] t <type> u <uid> [x] [e] [v]          
Options:          
    h     : This help          
    t     : 1 = MIFARE Classic 1k          
            2 = MIFARE Ultralight          
            3 = MIFARE Desfire          
            4 = ISO/IEC 14443-4          
            5 = MIFARE Tnp3xxx          
            6 = MIFARE Mini          
            7 = AMIIBO (NTAG 215),  pack 0x8080          
            8 = MIFARE Classic 4k          
            9 = FM11RF005SH Shanghai Metro          
    u     : 4, 7 byte UID          
    x     : (Optional) Performs the 'reader attack', nr/ar attack against a reader          
    e     : (Optional) Fill simulator keys from found keys          
    v     : (Optional) Verbose          
Examples:          
          hf 14a sim t 1 u 11223344 x           
          hf 14a sim t 1 u 11223344           
          hf 14a sim t 1 u 11223344556677          
``` 

10. I run command - **hf mfu sim t 7 u 34A72E21B49260**:
```
[usb] pm3 --> hf mfu sim t 7 u 34A72E21B49260
[+] Emulating ISO/IEC 14443 type A tag with 7 byte UID (34 A7 2E 21 B4 92 60)           
[=] Press pm3-button to abort simulation          
```

I’ve been waiting for 10 minutes and nothing comes up.
One indicator are lit on the device - green (a) and nothing happens.

**What am I doing wrong, what other commands do I need to execute?**

iceman1001, 2020-04-25:
This is where you report issues / suggest new features for the source code.
What is the bug you are trying report? You ask questions on the forum, read wiki pages, blogposts, videos on how to operate your device.

2rsharipov, 2020-04-25:
Dear **iceman1001**!

I studied wiki, help and information on forums and google search engine.
I also asked my questions on your profile forum - http://www.proxmark.org/forum/viewtopic.php?id=7734 , but no one can tell me the solution to my question.
I apologize that maybe I asked my question not in a targeted way.
**I beg you to tell me how to solve my problem!**

iceman1001, 2020-04-25:
Closing.  Use forum.

closed 2020-04-25 by iceman1001