RfidResearchGroup/proxmark3 #718: Proxmark3 (No valid keys found)

2rsharipov, 2020-04-25:
Hello!

1. I run command - **hf search**:
```
[usb] pm3 --> hf search
 ? Searching for ISO14443-A tag...           
[+]  UID: 43 8B B7 75           
[+] ATQA: 00 04           
[+]  SAK: 08 [2]           
[+] POSSIBLE TYPE:    MIFARE Classic 1K / Classic 1K CL2           
[+] POSSIBLE TYPE:    MIFARE Plus 2K / Plus EV1 2K           
[+] POSSIBLE TYPE:    MIFARE Plus CL2 2K / Plus CL2 EV1 2K           
[=] proprietary non iso14443-4 card found, RATS not supported          
[+] Prng detection: weak           
          
[+] Valid ISO14443-A tag found
```

2. I run command - hf mf chk *1 ? d mfc_default_keys:
```
[usb] pm3 --> hf mf chk *1 ? d mfc_default_keys
[+] Loaded 920 keys from mfc_default_keys           
................................................................................................................................................................................................................................................................................................................................................................          
[=] Time in checkkeys: 217 seconds
          
[=] testing to read key B...          
          
[+] found keys:           
[+] |-----|----------------|---|----------------|---|          
[+] | Sec | key A          |res| key B          |res|          
[+] |-----|----------------|---|----------------|---|          
[+] | 000 | ------------   | 0 | ------------   | 0 |          
[+] | 001 | ------------   | 0 | ------------   | 0 |          
[+] | 002 | ------------   | 0 | ------------   | 0 |          
[+] | 003 | ------------   | 0 | ------------   | 0 |          
[+] | 004 | ------------   | 0 | ------------   | 0 |          
[+] | 005 | ------------   | 0 | ------------   | 0 |          
[+] | 006 | ------------   | 0 | ------------   | 0 |          
[+] | 007 | ------------   | 0 | ------------   | 0 |          
[+] | 008 | ------------   | 0 | ------------   | 0 |          
[+] | 009 | ------------   | 0 | ------------   | 0 |          
[+] | 010 | ------------   | 0 | ------------   | 0 |          
[+] | 011 | ------------   | 0 | ------------   | 0 |          
[+] | 012 | ------------   | 0 | ------------   | 0 |          
[+] | 013 | ------------   | 0 | ------------   | 0 |          
[+] | 014 | ------------   | 0 | ------------   | 0 |          
[+] | 015 | ------------   | 0 | ------------   | 0 |          
[+] |-----|----------------|---|----------------|---|          
[+] ( 0 :Failed / 1 :Success)          
[+] Generating binary key file          
[+] Found keys have been dumped to hf-mf-438BB775-key.bin --> 0xffffffffffff has been inserted for unknown keys.
```

3. I run command - **hf mf fchk 1 m**:
```
[usb] pm3 --> hf mf fchk 1 m
[+] No key specified, trying default keys          
[ 0] ffffffffffff          
[ 1] 000000000000          
[ 2] a0a1a2a3a4a5          
[ 3] b0b1b2b3b4b5          
[ 4] c0c1c2c3c4c5          
[ 5] d0d1d2d3d4d5          
[ 6] aabbccddeeff          
[ 7] 1a2b3c4d5e6f          
[ 8] 123456789abc          
[ 9] 010203040506          
[10] 123456abcdef          
[11] abcdef123456          
[12] 4d3a99c351dd          
[13] 1a982c7e459a          
[14] d3f7d3f7d3f7          
[15] 714c5c886e97          
[16] 587ee5f9350f          
[17] a0478cc39091          
[18] 533cb6c723f6          
[19] 8fd0a4f256e9          
[20] 0000014b5c31          
[21] b578f38a5c61          
[22] 96a301bce267          
[=] Running strategy 1          
          
[=] Chunk: 0.4s | found 0/32 keys (23)
          
[=] Running strategy 2          
.          
[=] Chunk: 3.9s | found 0/32 keys (23)
          
[=] Time in checkkeys (fast):  4.4s
          
[!]  ⚠️  No keys found 
```

4. I run command - **script run hf_bruteforce -s 0x438BB775 -e 0x438BB775 -t 1000 -x mfc**:
```
[usb] pm3 --> script run hf_bruteforce -s 0x438BB775 -e 0x438BB775 -t 1000 -x mfc
[+] executing lua /home/administrator/proxmark3/client/luascripts/hf_bruteforce.lua           
[+] args '-s 0x438BB775 -e 0x438BB775 -t 1000 -x mfc'           
----------------------------------------

Bruteforcing Mifare Classic card numbers

----------------------------------------
Running: "hf 14a sim t 1 u 000000438bb775"
[+] Emulating ISO/IEC 14443 type A tag with 7 byte UID (00 00 00 43 8B B7 75)           
[=] Press pm3-button to abort simulation
```

I’ve been waiting for 10 minutes and nothing comes up.
One indicator are lit on the device - green (a) and nothing happens.

5. I run command - **hf mf autopwn**:
```
[usb] pm3 --> hf mf autopwn
[!]  ⚠️  no known key was supplied, key recovery might fail          
[+] loaded 23 keys from hardcoded default array          
[=] running strategy 1          
          
[=] Chunk: 0.4s | found 0/32 keys (23)
          
[=] running strategy 2          
.          
[=] Chunk: 3.9s | found 0/32 keys (23)
          
[=] --------------------------------------------------------------------------------
          
[=] executing Darkside attack. Expected execution time: 25sec on average          
[=] press pm3-button on the Proxmark3 device to abort both Proxmark3 and client.          
[=] --------------------------------------------------------------------------------
          
.................................................................................................................................................................
```

I’ve been waiting for 10 minutes and nothing comes up, only the preloader with dots runs.
Two indicators are lit on the device - blue (d) and orange (c) and nothing happens.

**I tried different options for the commands. But the keys are still not found.
What am I doing wrong, what other commands do I need to execute?**

iceman1001, 2020-04-25:
This is where you report issues / suggest new features for the source code. 
What is the bug you are trying report?  You ask questions on the forum,  read wiki pages, blogposts, videos on how to operate your device.

2rsharipov, 2020-04-25:
Dear **iceman1001**!

I studied wiki, help and information on forums and google search engine.
I also asked my questions on your profile forum - http://www.proxmark.org/forum/viewtopic.php?id=7715 , but no one can tell me the solution to my question.
I apologize that maybe I asked my question not in a targeted way.
**I beg you to tell me how to solve my problem!**

iceman1001, 2020-04-25:
Closing. Use forum.

closed 2020-04-25 by iceman1001