RfidResearchGroup/proxmark3 #720: [idea] You can't write dump to another card

Windslab, 2020-04-26:
The general idea thats we needs to have opportunity to makes clones for each types of the cards. Dumping functions must be reviewed. Standard functionality not allow you to write a dump for a blank/other written card. At the moment impossible to make a clone with a few clicks. It should be in the task list.

iceman1001, 2020-04-27:
Feel free to contribute,  however we don't use "clicks" in this repo,  the proxmark3 client is CLI-based,  which makes me think you maybe posted this suggestion in wrong repo.

Windslab, 2020-04-27:
> Feel free to contribute

Thanks for the inviting, I already tried. Today only idea for your team. I contribute by my management skills.

> however we don't use "clicks" in this repo, the proxmark3 client is CLI-based, which makes me think you maybe posted this suggestion in wrong repo.

Some themes sad to read answer like this I really don't know do you kidding or serious. Yeah it's CLI based - console application. I was afraid to replace word `clicks` to `keys` because then for you these `AB` association. I don't know why you don't get.

iceman1001, 2020-06-02:
Since we are now enforcing accurate and detailed _idea request_  styled issues policy. 
This means that this issue will be close if you don't provide with a detailed solution to your idea.  

General terms like "clones of cards"  isn't specific enough.  Contradictionary terms like  "doesn't allow" when client does allow,  etc.  

Please revise your issue with the appropreate information needed.

closed 2020-06-02 by Windslab

iceman1001, 2020-06-02:
In order to make this policy clear for everyone,   we have created overview wiki page and we urge you to use the forum to discuss your idea aswell.

[Wiki Wishlist](https://github.com/RfidResearchGroup/proxmark3/wiki/Wishlist)

[Proxmark3 forum](http://www.proxmark.org/forum/index.php)

Windslab, 2020-06-02:
Many Thanks. Looking forward.

But if you give a discount for your last RDV device, make sure I will more
invest to the community. Please do not declare all to black and white.

The Idea to copy cards in my video:
https://www.youtube.com/watch?v=-uSDJGUt-5c

doegox, 2020-06-02:
> But if you give a discount for your last RDV device, make sure I will more invest to the community. Please do not declare all to black and white.

What @iceman1001 was telling is that it's not *our* RDV, it's made and sold by people in Asia. All contributors of the opensource firmware and software, including Iceman and myself, are their clients, buying their hardware products. We're not in position to offer discounts, that's something up to the sellers to decide. Lab401 is sometimes offering discounts. RRG is also sometimes offering discounts to those who contributed to the Kickstarter campain. And I repeat, we're in no way implied in their business.