RfidResearchGroup/proxmark3 #729: Stuck after hf search

ManChiuNgai, 2020-05-12:
I got stuck after using hf search and Led D keep lighted. All other function will get time out at this time. The button on proxmark3 is not working too. I need to re-plug the pm3 to get normal.

**To Reproduce**
Steps to reproduce the behavior:
1. hf search
2. get stuck after searching

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 1909 x64

<code> [usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with MinGW-w64 10.1.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/66fd6d70 2020-05-12 22:57:57
       os: RRG/Iceman/master/66fd6d70 2020-05-12 22:58:14
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 254050 bytes (48%) Free: 270238 bytes (52%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
</code>

 - hw status
<code> [usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz)
#db#   [b] bits per sample.....8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
#db#    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
#db#
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......103424
#db#   Transfer Speed PM3 -> Client = 206848 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............31852 Hz
#db# Installed StandAlone Mode
#db#   LF EM4100 read/sim/write/brute mode
</code>

**Additional context**
Add any other context about the problem here.

ManChiuNgai, 2020-05-12:
I tried some interesting test on this. I found it can be unfreeze by touching 13.56Mhz reader. Also, Felica function may not running operation normally. Here is a video.

[https://youtu.be/aBfcYWNDYVY](https://youtu.be/aBfcYWNDYVY)

doegox, 2020-05-12:
Hmm ok we can reproduce it with a PM3OTHER device.
The bug doesn't appear with a PM3RDV4.

doegox, 2020-05-12:
The bug appeared indeed when Felica was added to hf search, commit 9fdc29e91c02f8c0c1f86ea8ca11bb9955907583
Still digging...

iceman1001, 2020-05-12:
Looking at that video,  ```hf search````  iClass timeous out, leaving the device in a unstable state?

doegox, 2020-05-12:
it's not iclass, it's felica and whatever done after felica is wrong
and indeed bringing sth in the field "unlocks" the pm3

doegox, 2020-05-12:
FTR sample bytes from FPGA differ between rdv4 and non-rdv4 in this FeliCa demodulation when there is no card on the reader.

* rdv4 : they converve quickly to 00, e.g. FF61000000...
* non-rdv4: they converge quickly to FF, e.g. 00F8FFFFFF...

The result is that non-rdv4 is always in STATE_TRYING_SYNC while rdv4 is in STATE_UNSYNCD

When a card is brought (anything that takes some field energy, not specifically Felica), non-rdv4 gets 00... and stays in STATE_UNSYNCD, allowing the timeout to operate.

I don't know why FPGA is sending FFs on non-rdv4 in that situation.
My fix was to also be able to timeout from STATE_TRYING_SYNC.

As when a card is present, both rdv4 and non-rdv4 can read properly FeliCa cards.

closed 2020-05-12 by doegox