RfidResearchGroup/proxmark3 #730: Unable to compile client after 94133a8

swg0101, 2020-05-12:
It appears that as of the 94133a8 Merge branch 'strict' commit the compilation of the client has been failing when compiling for PM3OTHER.

See screenshot as attached.
[image]

doegox, 2020-05-12:
which env is it ? termux ?

doegox, 2020-05-12:
does it compile if you do `make client NOERROR=1` ? (on latest commit)

swg0101, 2020-05-12:
Ah yes, NOERROR=1 works. This is on Termux Clang.

doegox, 2020-05-12:
Hmm when trying on the termux-packages cross-compilation docker image, it works fine.
But the compiler is older: Android (5220042 based on r346389c) clang version 8.0.7
I need to dig further

doegox, 2020-05-12:
ok I could reproduce, I pushed a fix to ignore these ones, still I find it weird, should be properly addressed

doegox, 2020-05-12:
btw could you share how you compile pm3 on termux ? I could compile the client but didn't find arm-none-eabi-gcc

swg0101, 2020-05-12:
Two-step process - compile pm3 client in Clang, compile boot image in Ubuntu:
~/start-ubuntu.sh "cd ~/proxmark3 && git checkout master && git pull && make bootrom fullimage PLATFORM=PM3OTHER -j4" && make -C ~/proxmark3 client PLATFORM=PM3OTHER -j4 NOERROR=1

doegox, 2020-05-13:
Thanks!
I think this strict warning revealed an issue in the headers which by chance defaulted into little endian instead of alerting about unknown endianess.
I pushed another fix in a branch https://github.com/RfidResearchGroup/proxmark3/commit/2c4eda6ab9b83e3d5853408c354d344d0255e0f0 
Let's see if it works properly on CI. If yes I'll merge.

swg0101, 2020-05-13:
Yes, it no longer errors out when compiling after make clean and without NOERROR=1, only the following appears as warnings:

`
[-] CC src/cmdsmartcard.c
src/cmdtrace.c:79:39: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    uint16_t next_records_datalen = *((uint16_t *)(trace + tracepos + sizeof(uint32_t) + sizeof(uint16_t)));
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:95:37: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
        uint32_t next_timestamp = *((uint32_t *)(trace + *tracepos));
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:97:36: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
        uint16_t next_duration = *((uint16_t *)(trace + *tracepos));
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:99:36: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
        uint16_t next_data_len = *((uint16_t *)(trace + *tracepos)) & 0x7FFF;
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:129:19: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
    timestamp = *((uint32_t *)(trace + tracepos));
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:136:18: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    data_len = *((uint16_t *)(trace + tracepos));
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:219:25: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
    first_timestamp = *((uint32_t *)(trace));
                        ^~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:220:19: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
    timestamp = *((uint32_t *)(trace + tracepos));
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:223:18: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    duration = *((uint16_t *)(trace + tracepos));
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:226:18: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint16_t *' (aka 'unsigned short *') increases required alignment from 1 to 2 [-Wcast-align]
    data_len = *((uint16_t *)(trace + tracepos));
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdtrace.c:442:37: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
        uint32_t next_timestamp = *((uint32_t *)(trace + tracepos));
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
11 warnings generated.
`

doegox, 2020-05-13:
> ~/start-ubuntu.sh 

For that one I need to install a chroot Ubuntu ? I found https://wiki.termux.com/wiki/Ubuntu but the script doesn't find the image to download.

doegox, 2020-05-13:
> only the following appears as warnings

yes we're aware of these, we need to rework the handling of traces, thanks

doegox, 2020-05-13:
I merged the other fix, you can test it, you'll get the same result.

swg0101, 2020-05-13:
'''
wget https://raw.githubusercontent.com/Neo-Oli/termux-ubuntu/master/ubuntu.sh
sed -i 's/disco/focal/g' ubuntu.sh
~/start-ubuntu.sh "apt update && apt install -y nano git gcc-arm-none-eabi make gcc g++"
sed -i 's$#command+=" -b /data/data/com.termux/files/home:/root"$command+=" -b /data/data/com.termux/files/home:/root"$' start-ubuntu.sh
rm ubuntu.tar.gz
'''

doegox, 2020-05-13:
ha thanks! I figured the disco/focal myself and missed your msg.
the part I was missing was to get the same homedir, I cloned twice the repo :)

closed 2020-05-13 by doegox

swg0101, 2020-05-13:
Awesome - it's a bit inefficient though with that many dependencies, but since Termux itself doesn't really have a workable gcc in the repo for arm-none-eabi that's the only way I found for it to work properly.

swg0101, 2020-05-13:
But yeah, I am using the following macros in my fish shell (not the default -ash shell) and Proxmark3 works quite well with the old tsu package since the OnePlus 6T kernel already has ACM built-in ($PREFIX/etc/fish/config/fish):

```
function pmb
 if not set -q argv[1]
     ~/start-ubuntu.sh "cd ~/proxmark3 && git checkout master && git pull && make bootrom fullimage PLATFORM=PM3OTHER -j4" && make -C ~/proxmark3 client PLATFORM=PM3OTHER -j4
 else
     ~/start-ubuntu.sh "cd ~/proxmark3 && git fetch && git checkout $argv[1] && make bootrom fullimage PLATFORM=PM3OTHER -j4" && make -C ~/proxmark3 client PLATFORM=PM3OTHER -j4
 end
end
alias pmf="tsudo ~/proxmark3/client/proxmark3 --port /dev/ttyACM0 --flash --unlock-bootloader --image ~/proxmark3/bootrom/obj/bootrom.elf --image ~/proxmark3/armsrc/obj/fullimage.elf"
alias pmfb="tsudo ~/proxmark3/client/proxmark3 --port /dev/ttyACM0 --flash --unlock-bootloader --image ~/proxmark3/bootrom/obj/bootrom.elf"
alias pmff="tsudo ~/proxmark3/client/proxmark3 --port /dev/ttyACM0 --flash --image ~/proxmark3/armsrc/obj/fullimage.elf"
```

iceman1001, 2020-05-13:
Good finding and good fix for termux builds.  Its one of those env we don't test that much on, maybe now @doegox has it setup so we can make build tests easier on it.

How about using existing scripts instead of your alias pmf,pmfb, pmff?
test the following we have for it and see if they work
```
./pm3
./pm3-flash-fullimage
./pm3-flash-bootrom
```

swg0101, 2020-05-13:
Nope, they don't really work:

```
u0_a134@localhost ~> cd ~/proxmark3/
u0_a134@localhost ~/proxmark3 (master)> ./pm3
[=] Waiting for Proxmark3 to appear...
^C⏎                                                                                                                                                                                                                                                                                                                          u0_a134@localhost ~/proxmark3 (master) [SIGINT]> tsudo ./pm3
./pm3: line 34: udevadm: command not found
[=] Waiting for Proxmark3 to appear...
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found

u0_a134@localhost ~/proxmark3 (master)> tsudo ./pm3-flash-fullimage
./pm3: line 34: udevadm: command not found
[=] Waiting for Proxmark3 to appear...
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
^C⏎                                                                                                                                                                                                                                                                                                                          u0_a134@localhost ~/proxmark3 (master)> tsudo ./pm3-flash-bootrom
./pm3: line 34: udevadm: command not found
[=] Waiting for Proxmark3 to appear...
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
./pm3: line 34: udevadm: command not found
^C⏎                                                                  
```

iceman1001, 2020-05-13:
interesting,  no udevadm on termux.  Thanks for testing.

doegox, 2020-05-13:
I might try to fix the script to make it termux compatible too, but first I need to root my phone :D

swg0101, 2020-05-15:
@doegox Here's some reference from my OnePlus 6T phone:

```
root@localhost /sys# cat  /sys/class/tty/ttyACM0/../../uevent
DEVTYPE=usb_interface
DRIVER=cdc_acm
PRODUCT=9ac4/4b8f/100
TYPE=2/0/0
INTERFACE=2/2/1
MODALIAS=usb:v9AC4p4B8Fd0100dc02dsc00dp00ic02isc02ip01in00

root@localhost /sys [1]# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 4.09
S:  Manufacturer=Linux 4.9.179-Omega xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=02(comm.) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=9ac4 ProdID=4b8f Rev= 1.00
S:  Manufacturer=proxmark.org
S:  Product=proxmark3
S:  SerialNumber=iceman
C:* #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=01 Driver=cdc_acm
E:  Ad=83(I) Atr=03(Int.) MxPS=   8 Ivl=255ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=01(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 4.09
S:  Manufacturer=Linux 4.9.179-Omega xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
```

Perhaps looking for **PRODUCT=9ac4** would be good enough from the sysfs.

doegox, 2020-05-16:
thanks, please test latest pm3