RfidResearchGroup/proxmark3 #736: Build error (bootrom)

ikarus23, 2020-05-19:
Hi, when I try to compile the latest master (db416483421d168541ec34307929450f824bbfcc), I get an error:
```
[=] LD obj/bootrom.elf
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: obj/bootrom.o: in function `reply_old.constprop.0.isra.0':
bootrom.c:(.text.reply_old.constprop.0.isra.0+0x18): undefined reference to `memset'
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: obj/bootrom.o: in function `BootROM':
bootrom.c:(.text.BootROM+0xb4): undefined reference to `memset'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:55: obj/bootrom.elf] Error 1
make: *** [Makefile:95: bootrom/all] Error 
```
I'm using Arch linux with:
```
arm-none-eabi-gcc -v                                                                                          :(
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/10.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-10.1.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.1.0 (Arch Repository) 
```

evs-ch, 2020-05-19:
Can confirm, happens for me as well. Seems to be related to changes from gcc 10; when building with `arm-none-eabi-gcc-9.3.0-1` (in a clean chroot), the build is successful.

iceman1001, 2020-05-19:
https://stackoverflow.com/questions/48148077/arm-eabi-none-undefined-reference-to-memset-error-for-lpc2148

evs-ch, 2020-05-19:
If it helps: by either omitting `-nodefaultlibs` from `$LDFLAGS` in [bootrom/Makefile:55](https://github.com/RfidResearchGroup/proxmark3/blob/f7534f412081631faf0ab1a174373674d7e1122e/bootrom/Makefile#L55) or appending `-lc` to the end of the same line / `$LIBS`, things also can be coerced into building with gcc 10 again.

No idea if that's the way to go, though ^^

doegox, 2020-05-19:
Hmm adding the default libs brings the bootrom from 3888 to 4224.
Still investigating why such difference with gcc 10...

doegox, 2020-05-19:
Well bootrom (and armsrc) should not need default libs.
The weird thing is that compiling with `-Os` seems to create a call to memset out of the manual zeroing code https://github.com/RfidResearchGroup/proxmark3/blob/master/bootrom/bootrom.c#L320-L322
Commenting these lines or removing -Os make the error disappearing.
But none of these are actual solutions.

doegox, 2020-05-19:
Ok I get progress with adding `-fno-builtin` but now new errors are popping.
At least we're on the good track.

doegox, 2020-05-19:
So technically this issue is solved but there are still other errors linked to arm-none-eabi-gcc 10 so I'll leave this issue open till it compiles fine.
Now:
- [ ] warnings in legicrf.c
- [ ] multiple defs of blocknr
- [ ] multiple defs of FpgaSendQueueDelay

doegox, 2020-05-19:
we should be good now, it revealed some mess in variables exported without real reason

ikarus23, 2020-05-19:
Thank you! Great job! Everything works and I've got no error.

closed 2020-05-19 by ikarus23

mercertom, 2020-12-27:
I'm still suffering this problem.

ikarus23, 2020-12-27:
What system are you on? What version of the repo do you use? master? What version of gcc for ARM?
(I don't have any issues.)

mercertom, 2020-12-27:
What system are you on? Manjaro
What version of the repo do you use? I have tried all 3: old iceman, rrg, regular proxmark3. Two fail at this step, the other in a different way.
master? dunno what this means, so probably not.
What version of gcc for ARM? The up-to-date for Manjaro. 10.2.something.

ikarus23, 2020-12-27:
That is really strange. Could you `make clean` and then `make` and post the error?