RfidResearchGroup/proxmark3 #737: Empty HF 15 tracelog from the device

FlUxIuS, 2020-05-19:
Hi, 

The tracelog for ISO15693 tags is empty when downloading:

```
[usb] pm3 --> hf 15 reader

[+]  UID: E0 07 A2 00 00 00 61 AD
[+] TYPE: Texas Instrument France
[usb] pm3 --> trace list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 0 bytes)
[usb] pm3 --> trace list 15
[=] downloading tracelog from device
[+] Recorded activity (trace len = 0 bytes)
```
But when it seems to work, it displayes weird traces as follows:

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
          0 |          0 | Rdr |0a!                                                                      |     | 
```

iceman1001, 2020-05-19:
Now thats a bug :)

doegox, 2020-05-19:
ok I see, iso15 is apparently going a bit wild on the timestamps and my rewrite of LogTrace refuses to create entries with corrupted durations.
Let's see...

doegox, 2020-05-19:
should be fixed now, still, globally, some LogTrace usages are weird imho

FlUxIuS, 2020-05-20:
Hi again,

I still have the issue even with few ski passes.

Here is the trace when I issue the read command: 

```
[usb] pm3 --> hf 15 reader

[+]  UID: E0 07 80 98 3E 79 60 83
[+] TYPE: Texas Instrument; Tag-it HF-I Plus (RF-HDT-DVBB tag or Third Party Products)
[usb] pm3 --> trace list 15
[=] downloading tracelog from device
[+] Recorded activity (trace len = 0 bytes)
[usb] pm3 --> trace list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 0 bytes)
```
And there when I use `hf search`:

> [usb] pm3 --> hf search
>  🕒 Searching for ISO15693 tag...             
> [+]  UID: E0 07 80 98 3E 79 60 83
> [+] TYPE: Texas Instrument; Tag-it HF-I Plus (RF-HDT-DVBB tag or Third Party Products)
> 
> [+] Valid ISO15693 tag found
> 
>  🕑                                           
> [usb] pm3 --> trace list
> [=] downloading tracelog from device
> [+] Recorded activity (trace len = 110 bytes)
> [=] Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
> [=] ISO14443A - All times are in carrier periods (1/13.56MHz)
> 
>       Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
> ------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     | 
>           0 |          0 | Rdr |0a!                                                                      |     |

doegox, 2020-05-20:
chan you try again with hw dbg 3 ?
stupid question but did you reflash the pm3?

FlUxIuS, 2020-05-20:
> can you try again with hw dbg 3 ?
> stupid question but did you reflash the pm3?

Wait forget it! My fault this time I forgot to flash firmware -_-:

```
[usb] pm3 --> hf 15 reader

[+]  UID: E0 07 80 98 3E 79 60 83
[+] TYPE: Texas Instrument; Tag-it HF-I Plus (RF-HDT-DVBB tag or Third Party Products)
[usb] pm3 --> trace list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 36 bytes)
[=] Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
[=] ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       3440 | Rdr |26  01  00! f6! 0a!                                                      | !crc| REQA
       3456 |       3456 | Tag |00! 01  83  60! 79  3e  98  80  07  e0  d4! 33!                          | !crc| 

```

closed 2020-05-20 by FlUxIuS

doegox, 2020-05-20:
hehe I was suspecting sth like that ;)

FlUxIuS, 2020-05-20:
> hehe I was suspecting sth like that ;)

Yeah, I just thought about that after sending this issue... -_-'