RfidResearchGroup/proxmark3 #742: Error in Compile proxmark3

MohammadSecure, 2020-05-20:
Hi 
when i compile proxmark3 [https://github.com/RfidResearchGroup/proxmark3.git](url)
with this syntax `./proxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image ../bootrom/obj/bootrom.elf --image ../armsrc/obj/fullimage.elf`
This error in last seen : 

> Error: PHDR is not contained in Flash

please help me

iceman1001, 2020-05-20:
- Did you get some errors when you compiled the project?
- are you running the client from inside the client folder?
 - why not use the ./pm3-flash-all script?

iceman1001, 2020-05-24:
Ping @MohammadSecure  ,

MohammadSecure, 2020-05-31:
I run this script `./pm3-flash-all` but i this is error out put:
`oading usable ELF segments:`
`[+]    0: V 0x00102000 P 0x00102000 (0x000434f0->0x000434f0) [R X] @0x94`
`[!!] 🚨 Error: PHDR is not contained in Flash`
`[+] All done`

and when run this syntax : `./proxmark3 /dev/ttyACM0 `
output is:
`[=] Session log /home/ali/.proxmark3/logs/log_20200531.txt`
`[=] Loading Preferences...`
`[+] loaded from JSON file /home/mrg/.proxmark3/preferences.json`
`[=] Using UART port /dev/ttyACM0`
`[!!] 🚨 ERROR: cannot communicate with the Proxmark`

iceman1001, 2020-05-31:
You didn't answer my questions.

iceman1001, 2020-06-02:
Closing since OP doesn't answer questions or provide accurate debug information needed.

closed 2020-06-02 by iceman1001