RfidResearchGroup/proxmark3 #746: You cannot use 32-bit time_t (_USE_32BIT_TIME_T) with _WIN64

wuyasec, 2020-05-21:
error：
[-] CC mf_nonce_brute.c
In file included from D:/ProxSpace/msys2/mingw64/x86_64-w64-mingw32/include/crtdefs.h:10,
                 from D:/ProxSpace/msys2/mingw64/x86_64-w64-mingw32/include/inttypes.h:11,
                 from mf_nonce_brute.c:7:
D:/ProxSpace/msys2/mingw64/x86_64-w64-mingw32/include/corecrt.h:133:2: error: #error You cannot use 32-bit time_t (_USE_32BIT_TIME_T) with _WIN64
  133 | #error You cannot use 32-bit time_t (_USE_32BIT_TIME_T) with _WIN64
      |  ^~~~~
make[1]: *** [../../Makefile.host:75: obj/mf_nonce_brute.o] Error 1
make: *** [Makefile:89: mf_nonce_brute/all] Error 2

iceman1001, 2020-05-21:
Please provide details about OS,  your build env,   and you are using the latest source?

wuyasec, 2020-05-21:
win10pro 1909 x64 ；；Same operation as official video；；Resources are just downloaded

iceman1001, 2020-05-21:
Good,  And your build env?   are you using WSL or Proxspace?   If proxspace with version and is it 32 or 64b?

wuyasec, 2020-05-21:
Proxspace；run runme64.bat

iceman1001, 2020-05-21:
If you modify  
https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/mf_nonce_brute/mf_nonce_brute.c#L3
```
-#if defined(_WIN32)
+#if defined(_WIN32) || defined(__WIN32__)
 # define _USE_32BIT_TIME_T 1
 #endif
```

https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/mf_nonce_brute/mf_nonce_brute.c#L353
```
@@ -350,7 +350,7 @@ int main(int argc, char *argv[]) {
     //calc (parity XOR corresponding nonce bit encoded with the same keystream bit)
     uint16_t xored = xored_bits(nt_par, nt_enc, ar_par, ar_enc, at_par, at_enc);

-#ifndef __WIN32
+#if !defined(_WIN32) || !defined(__WIN32__)
     thread_count = sysconf(_SC_NPROCESSORS_CONF);
     if (thread_count < 2)
         thread_count = 2;
```

make changes and try compiling

wuyasec, 2020-05-21:
Same error；corecrt.h:121-145
`#ifndef _TIME32_T_DEFINED
#define _TIME32_T_DEFINED
typedef long __time32_t;
#endif

#ifndef _TIME64_T_DEFINED
#define _TIME64_T_DEFINED
__MINGW_EXTENSION typedef __int64 __time64_t;
#endif /* _TIME64_T_DEFINED */

#ifdef _USE_32BIT_TIME_T
#ifdef _WIN64
#error You cannot use 32-bit time_t (_USE_32BIT_TIME_T) with _WIN64
#undef _USE_32BIT_TIME_T
#endif
#endif /* _USE_32BIT_TIME_T */

#ifndef _TIME_T_DEFINED
#define _TIME_T_DEFINED
#ifdef _USE_32BIT_TIME_T
typedef __time32_t time_t;
#else
typedef __time64_t time_t;
#endif
#endif /* _TIME_T_DEFINED */`

iceman1001, 2020-05-21:
Try this one,

```
#if !defined(_WIN64)
 #if defined(_WIN32) || defined(__WIN32__)
 # define _USE_32BIT_TIME_T 1
  #endif
#endif
```

wuyasec, 2020-05-21:
[*] MAKE mf_nonce_brute/all
[-] CC mf_nonce_brute.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/bucketsort.c
[-] CC iso14443crc.c
[-] CC sleep.c
[=] LD mf_nonce_brute
pm3 ~$

At last：[=] LD mf_nonce_brute，Whether the completion of the compilation task

commit a19ba257 (2020-05-21): fix: #746 -  ProxSpace 64b seem to report both _WIN32 and _WIN64
closed 2020-05-21 by iceman1001