RfidResearchGroup/proxmark3 #747: Unable to simulate raw binary mfu/ntag dumps

doyler, 2020-05-24:
**Describe the bug**
Using the current version of this repository, old NTAG125 raw binary dumps are no longer being properly simulated.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the EML file - hf mf eload u /Users/doyler/tools/ntag125/test
2. Attempt to simulate an NTAG125 - hf mf eload u /Users/doyler/tools/ntag125/test
3. Verify that the simulation did not work as expected:
```
[usb] pm3 --> hf mf eload u /Users/doyler/tools/ntag125/test
[+] loaded 1020 bytes from text file /Users/doyler/tools/ntag125/test.eml
[+] old mfu dump format, was converted on load to 243 pages
[=] Uploading to emulator memory
...............................................

[-] ⛔ Warning, Ultralight/Ntag file content, Loaded 257 blocks into emulator memory
```

**Expected behavior**
Using the older repository (https://github.com/iceman1001/proxmark3), this data is properly simulated.
```
pm3 --> hf mf eload u /Users/doyler/tools/ntag125/test
............................................
Loaded 255 blocks from file: /Users/raymond.doyle/tools/ntag125/test.eml          

pm3 --> hf 14a sim t 7 u <UID>
Emulating ISO/IEC 14443 type A tag with 7 byte UID (<UID>)          
Press pm3-button to abort simulation          
#db# Received unknown command (len=1):          
#db# 00          
#db# Emulator stopped. Tracing: 1  trace length: 2328
```

doyler, 2020-05-24:
Submitted issue per Twitter conversation and confirmation - https://twitter.com/herrmann1001/status/1264337070186782720

iceman1001, 2020-05-24:
yeah,   this happend when we introduced the new mfu dump format.
In short,   we have  plain raw dumps,  like pm3 did in the beginning or what other apps/devices might create.  Then we had the need for more extra information when it came to MFU/NTAG family of tags.
Hence the now "old binary format" was created.   This was all fine, until some year ago when we realiased we are missing  stuff like pack etc.   Hence the new binary mfu format we use today.
We also went to create json format to overcome the bad effects of the different binary formats.
Now,  this problem most likely arise from @doyler is using plain binary dumps.   Which isn't supported well any longer.    

I am working on a suitable fix,  where the pm3 client will support all three binary formats.

For reference,  the different binary formats below.

```
// New Ultralight/NTAG dump file format
// Length must be aligned to 4 bytes (UL/NTAG page)
#define MFU_DUMP_PREFIX_LENGTH 56

typedef struct {
    uint8_t version[8];
    uint8_t tbo[2];
    uint8_t tbo1[1];
    uint8_t pages;                  // max page number in dump
    uint8_t signature[32];
    uint8_t counter_tearing[3][4];  // 3 bytes counter, 1 byte tearing flag
    uint8_t data[1024];
} PACKED mfu_dump_t;

// Old Ultralight/NTAG dump file format
#define OLD_MFU_DUMP_PREFIX_LENGTH 48

typedef struct {
    uint8_t version[8];
    uint8_t tbo[2];
    uint8_t tearing[3];
    uint8_t pack[2];
    uint8_t tbo1[1];
    uint8_t signature[32];
    //uint8_t counter[3];
    uint8_t data[1024];
} old_mfu_dump_t;

// plain format
    uint8_t data[1024];
```

commit e7585b59 (2020-05-24): fix #747, 'hf mfu eload' - now detects and converts between plain/old/new mfu binary format
closed 2020-05-24 by iceman1001