RfidResearchGroup/proxmark3 #751: The Err

Windslab, 2020-05-31:
[image]

Analyze

doegox, 2020-05-31:
What is the command line you used to compile?
A quick fix is to use `make SKIPPYTHON=1`

Windslab, 2020-05-31:
I had used autobuild-sh from this video https://youtu.be/EfSgw40w4pc

https://yev.ooo/autobuild-sh

Windslab, 2020-05-31:
Even if I have put the libruary into the client folder the error is some about "0X000000" bla bla bla...

Windslab, 2020-05-31:
PS: After update the ProxSpace, `pacman -Syu` new sources does not compile at all.

doegox, 2020-05-31:
Please contact the maintainer of the script.
Best is to use the release version, it's more stable than the git repo.

Windslab, 2020-05-31:
@doegox your `make SKIPPYTHON=1` has working as well

Better then null, but that's not deal with grace product making, that's like an Mercedes but wait a minute, just one missing screw to wheel.

Could you please fix in the root?

doegox, 2020-05-31:
No, git master is not a product, it's our development tree.
https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.9237 is the "product".
You're using an external script we don't provide ourselves and I don't know what went wrong.
This is why I tell you to contact the script maintainer.
If the maintainer finds an issue in our repo, we'll be glad to fix it.

closed 2020-05-31 by iceman1001

Windslab, 2020-05-31:
Yeah thanks, I have had fixed them all at few clicks to my local repo.