RfidResearchGroup/proxmark3 #752: [Bug] Compiling Err

Windslab, 2020-06-01:
Yeah I had pumped my `ProxSpace` to the last version `ps-upgrade`, even `gcc-arm-none-eabi`. That's err will be in each case who now will try to compile. Analyze. Where is my gifts for the founded bugs? I am first. Where is discount for RDV4? 😄 

[image]

Regards,
Yev
Product Manager

doegox, 2020-06-01:
You're not the first, sorry :)
Our CI AppVeyor alerted us already 4 days ago
https://ci.appveyor.com/project/RfidResearchGroup/proxmark3/builds/33189794
but we didn't have the occasion to find a proper fix yet.
Don't hesitate to contribute with a proper fix.

Windslab, 2020-06-01:
Ohhh, are you working in UK style, just see 4 days and then fix?:D

Okay, great that's you know this bug. Do you have discount code for RDV4 at the Lab401?

closed 2020-06-02 by iceman1001

iceman1001, 2020-06-02:
Try pulling latest,  it compiles on my PS34.  

About discount codes,
LAB401 is a webshop which happens to sell RRG products.  Not much we, the maintainers, can do,  so I am curious on who you think should get paid?  I really would love to get paid for my efforts.

commit c3ab0b65 (2020-06-02): fix: #752  MINGW env needs to be using gnu printf