RfidResearchGroup/proxmark3 #755: Sniff are not working

Windslab, 2020-06-03:
What'up

1. `hf 14a sniff`
2. `hf list`

Got an crush of my Proxmark 3 Easy

> [usb] pm3 --> hf 14a sniff
> #db# Starting to sniff
> [usb] pm3 --> hf list
> [=] downloading tracelog from device
> [=] Waiting for a response from the Proxmark3...
> [=] You can cancel this operation by pressing the pm3 button
> [-] Timed out while trying to download data from device
> [!] timeout while waiting for reply.
> [usb] pm3 -->

I seen these long time a go but always had think that's problem with my device, but on original firmware it's working as well.

iceman1001, 2020-06-04:
It works on my RDV4. There have been reports the cheap clones of pm3 easy having issues with the different firmwares. Aswell as reports on bad antennas. Usually we have no clue what / how the quality process at those manufactures.   

If you figure out whats going on in the hardware, feel free to submit fixes to handle it.  Untill then I am closing this.   And do please use the suggested template when creating an issue. They are there for a reason.

closed 2020-06-04 by iceman1001

Windslab, 2020-06-04:
But do it's works on your Proxmark 3 Easy?

Can you confirm that's you have an original Easy and it's works?

iceman1001, 2020-06-04:
I don't have a original pm3 easy.  You can test the offical repo to see if it works there.

Windslab, 2020-06-04:
It's works about 3-4 month a go at your firmware. I exactly played with `hf list` on my replica

doegox, 2020-06-04:
if you could search for the commit where things got broken, with `git bisect`, it would help us a lot.

Windslab, 2020-06-04:
Okay I will check, but how I can revert repo to 4 month a go?

doegox, 2020-06-04:
checkout an old commit, e.g.
`git checkout f32d94b99cf288ba9cebceb42020c285f038714f` for a commit from 12 February
(make sure to always `make clean` before changing commit)
From there, if that commit works fine, you can start git bisect.
See this tutorial https://www.youtube.com/watch?v=u9-toShXbBE
In short first `git bisect start` then to that this commit if it's working: `git bisect good`, then `git checkout master` to get latest and as this one is not working for you, do `git bisect bad`.
From that moment, automatically, it will fetch a commit somewhere between these two points.
You compile, flash, test. Then you `make clean` and then tag it good or bad depending on the result (`git bisect good/bad`) and it automatically checks out another commit, etc, till there is one single commit. If by chance a commit cannot be compiled, you can `git bisect skip`.

mwalker33, 2020-06-04:
Did you press the button to stop the sniff prior the hf list ?
My test was NOT on the very latest (but I can do a full test later with the current/latest if needed.

My test
I placed the PM3 Easy between a mifare classic card and my RDV4
I placed the easy into sniff mode, then run some commands via the rdv4

PM3 Easy Clone.
[ ARM ]
  bootrom: RRG/Iceman/preferred-paths/f4a68878-dirty-unclean 2020-05-08 11:07:55
       os: RRG/Iceman/preferred-paths/f4a68878-dirty-unclean 2020-05-09 16:55:55
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

If I did not press the button prior to the hf list, I got the same response as the OP, but if I run the sniff, execute a command then press the button then hf list, I get the expected output on the easy.

I expect that would be the expected behaviour ?

Windslab, 2020-06-04:
[usb] pm3 --> hf sniff
#db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

[usb] pm3 -->

[usb] pm3 --> hf sniff
#db# Trigger kicked! Value: 125, Dumping Samples Hispeed now.
#db# HF Sniffing end
#db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

Not working at all.

mwalker33, 2020-06-04:
I download, complied and flashed the lastest onto my PM3 Easy clone.
all seems to be working (this was from windows 10)
```
[usb] pm3 --> hf 14a sniff
#db# Starting to sniff
   ****** Here i read the mifare with  the RDV4  and then pressed the button on the PM3 Easy ****
#db# maxDataLen=4, Uart.state=0, Uart.len=0
#db# traceLen=314, Uart.output[0]=0000009e
[usb] pm3 --> hf 14a list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 314 bytes)
[=] Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
[=] ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52                                                                       |     | WUPA
       2244 |       4612 | Tag |02  00                                                                   |     |
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
```

next i tried to sniff the mifare classic while doing an hf mf auto, this would then fill up the buffer and exit without the button press.

```
[usb] pm3 --> hf 14a sniff
#db# Starting to sniff
#db# maxDataLen=4, Uart.state=0, Uart.len=0
#db# traceLen=39162, Uart.output[0]=00000052
[usb] pm3 --> hf 14a list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 39162 bytes)
[=] Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
[=] ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       4672 | Tag |73  7e  2c  72                                                           |     |
      13628 |      22940 | Rdr |42! 3d! 03! e7  93! b0  ec! a9                                           | !crc|
      24192 |      28928 | Tag |a3  15  ef  ab                                                           |     |
```

And again, seems to be working for my on windows 10 PM3 Easy clone.

As a guess, I am wondering if your issue is linked to what you are trying to sniff and how it generates the traffic.

Windslab, 2020-06-05:
Thanks for your response.
Again, the sniffer even doesn't start at my Proxmark 3 Easy, at all. I
already told you, have you see my logs?

Windslab, 2020-06-05:
WOW WOW Stop playing games.

Comon! I was not talking about `hf 14a sniff`, I was talking about `hf 14b
sniff` and `hf sniff` - these don't work!

The command `hf 14a sniff` working as well:

[usb] pm3 --> hf 14a sniff
#db# Starting to sniff
#db# maxDataLen=6, Uart.state=0, Uart.len=0
#db# traceLen=126, Uart.output[0]=00000026
[usb] pm3 --> hf list
[=] downloading tracelog from device
[+] Recorded activity (trace len = 126 bytes)
[=] Start = Start of Start Bit, End = End of last modulation. Src = Source
of Transfer
[=] ISO14443A - All times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)
                              | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |          0 | Rdr |f0  25  d4  00  52  f5  5f  34  05  77  28
 5c  4d  a2  00  00  00  32   |     |
            |            |     |46  66  6d  01  01  12  02  02  07  ff  03
 02  00  13  04  01  64  07   |     |
            |            |     |01  03  7c  96
                              |  ok |
     222704 |     223760 | Rdr |26
                              |     | REQA
    8194176 |    8194176 | Rdr |f0  25  d4  00  1e  91  f7  a1  65  8b  cd
 a4  45  43  00  00  00  32   |     |
            |            |     |46  66  6d  01  01  12  02  02  07  ff  03
 02  00  13  04  01  64  07   |     |
            |            |     |01  03  5e  24
                              |  ok |
    8416864 |    8417920 | Rdr |26
                              |     | REQA
[usb] pm3 --> hf sniff
[usb] pm3 --> #db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

[usb] pm3 -->

NOT WORKING:

[usb] pm3 --> hf sniff
[usb] pm3 --> #db# Trigger kicked! Value: 125, Dumping Samples Hispeed now.
#db# HF Sniffing end
#db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

[usb] pm3 -->[usb] pm3 --> hf sniff
[usb] pm3 --> #db# Trigger kicked! Value: 125, Dumping Samples Hispeed now.
#db# HF Sniffing end
#db# Buffer cleared (40000 bytes)
#db# Skipping first 0 sample pairs, Skipping 0 triggers.

[usb] pm3 -->

AND

[usb] pm3 --> hf 14b sniff
#db# Trigger kicked! Value: 125, Dumping Samples Hispeed now.
#db# HF Sniffing end
[usb] pm3 -->

Do the same check please!

пт, 5 июн. 2020 г. в 03:31, Yevgeny Shevchenko <ulrseu@gmail.com>:

mwalker33, 2020-06-05:
Sorry, I thought it was the 14a as that is what was in your first post.

```
What'up
hf 14a sniff
hf list
Got an crush of my Proxmark 3 Easy
[usb] pm3 --> hf 14a sniff
#db# Starting to sniff
...
```

Windslab, 2020-06-05:
Yeah I messed up. Sorry!

doegox, 2020-06-05:
If you're talking about 14b, 14b is not well behaving since very long, we've a one-year-old internal issue on that:

Example reading a Navigo with libnfc versus pm3: for all long responses, pm3 is losing the last bytes.
```
TX:(-4) 00 a4 04 00 08 31 54 49 43 2e 49 43 41 [2f 00]
RX:(-4)    6a 82 [fe 00]
[usb] pm3 --> hf 14b raw -s -p -c 0200a4040008315449432e494341
[LEN 5] 02 6A 82 [4B 4C] OK          

TX:(-4) 00 a4 00 00 02 3f 00 [06 00]
RX:(-4)     85 17 00 01 00 00 00 12 12 00 00 01 03 01 01 00 15 15 15 00 00 00 00 00 00 90 00 [54 00]
[usb] pm3 --> hf 14b raw -p -c 0300a40000023f00
[LEN 21] 03 85 17 00 01 00 00 00 12 12 00 00 01 03 01 01 00 15 15 [15 00] FAIL          

TX:(-4) 00 a4 00 00 02 00 02 [43 00]
RX:(-4)     85 17 02 04 02 1d 01 1f 00 00 00 01 01 01 01 00 00 00 00 00 00 00 00 00 00 90 00 [75 00]
[usb] pm3 --> hf 14b raw -p -c 0200a40000020002
[LEN 20] 02 85 17 02 04 02 1D 01 1F 00 00 00 01 01 01 01 00 00 [00 00] FAIL          

TX:(-4) 00 b2 01 04 1d [17 00]
RX:(-4)     00 00 00 00 00 00 00 04 00 71 00 42 00 00 00 00 c1 7a 16 45 02 50 14 00 01 29 98 43 00 90 00 [a2 00]
[usb] pm3 --> hf 14b raw -p -c 0300b201041d
[LEN 22] 03 00 00 00 00 00 00 00 04 00 71 00 42 00 00 00 00 C1 7A 16 [45 02] FAIL          
```

I didn't try yet on latest code but it's probably similar.