RfidResearchGroup/proxmark3 #756: [idea] Support for compiling Android platform executables.

xianglin1998, 2020-06-05:
When completing the branch of libpm3, I thought about the compilation process of NDK. By adding NDK path and some ABI parameters to the environment, we can directly compile the executable files corresponding to ABI, and then send them to Android devices for execution, without building any compilation environment (like termux) on the mobile side.

See: https://developer.android.com/ndk/guides/cmake#command-line

[image]

As mentioned in the above document, we need to specify some parameters, which can be obtained from a properties file or the system environment. When this variable exists, we can compile PM3 into the specified ABI executable file.

But we may need to solve the problems of other libraries' platforms.

If all the construction processes are solved by cmake, then multi platform support will be very simple, let's look forward to it.

xianglin1998, 2020-06-05:
@doegox

doegox, 2020-06-05:
It's interesting indeed.
Could you try that process on your client/android/CMakeLists.txt ?
So making your CMakeLists.txt more similar to the one of the libpm3 branch + a script to call cmake with these parameters.
Then, once the libpm3 branch is mature and merged, we can bridge the gap with your client/android/CMakeLists.txt and have one common approach.

xianglin1998, 2020-06-05:
I am now trying to compile with NDK directly under the /client. If it succeeds, it is possible to run PM3 client directly on Android platform in Linux executable mode in the future. No JNI is needed, which also helps me integrate termux view.

xianglin1998, 2020-06-05:
If you don't need to customize Java API in android, then client/android/CMakeLists.txt Will no longer be used.

iceman1001, 2020-06-05:
Interesting,  looking forward to hear of your results

xianglin1998, 2020-06-05:
[image]

xianglin1998, 2020-06-05:
Now I am compiling /android/CmakeLists.txt. It's strange that readLine is not referenced in the cmake list, but I still get this error.

doegox, 2020-06-05:
There are stuff like 
src/cmdhflegic.c
16:#include <readline/readline.h>

doegox, 2020-06-05:
Hmm that one is guarded with #ifndef ANDROID  but maybe there are new ones...

xianglin1998, 2020-06-05:
It is like a complie err？we can know is a lib can't found, -lreadline

doegox, 2020-06-05:
I think it's because of amiibo, just remove `pm3rrg_rdv4_amiibo` from target_link_libraries
It's not used at the moment anyway

doegox, 2020-06-05:
yes it's in client/deps/amiibo.cmake :
```
target_link_libraries(pm3rrg_rdv4_amiibo PRIVATE
        readline
```

xianglin1998, 2020-06-05:
Wow, yes ...

xianglin1998, 2020-06-05:
[image]

xianglin1998, 2020-06-05:
I'm build successfully.

xianglin1998, 2020-06-05:
Next i will test cross compilation. Thanks. :)

doegox, 2020-06-05:
great !

doegox, 2020-06-05:
I pushed https://github.com/RfidResearchGroup/proxmark3/commit/8852e1f9cdfc6d8e42087899cd9e73ee35262bea

iceman1001, 2020-06-05:
Great!

xianglin1998, 2020-06-08:
@doegox I always get a err on compile lua: 
[image]

xianglin1998, 2020-06-08:
I tried to replace it with the gnu99 standard and it didn't work.

iceman1001, 2020-06-08:
If you look in the offending file,  you see that define is guarded by  LUA_USE_POSIX. 
I guess adapting to android?   Did you have this problem before?
```
#if defined(LUA_USE_POSIX)

#define l_fseek(f,o,w) fseeko(f,o,w)
#define l_ftell(f) ftello(f)
#define l_seeknum off_t
```

xianglin1998, 2020-06-08:
[image]

xianglin1998, 2020-06-08:
I found this problem, but I still reported an error when I tried to set the compilation standard on cmake.

xianglin1998, 2020-06-08:
I didn't find this problem before, maybe because my cmake didn't add the -Werror flag.

iceman1001, 2020-06-08:
As @doegox  mentioned before,  your cmakelists.txt is different from the one we use in client folder.
The idea was to make your android specific stuff in to it.

Anyway,  Where did you add the `-std=gnu99`  flag?

iceman1001, 2020-06-08:
Inside this?
`target_compile_options(pm3rrg_rdv4_lua PRIVATE -Wall -Werror -O3)`

xianglin1998, 2020-06-08:
[image]

Look :)

xianglin1998, 2020-06-08:
Use function add_compile_options()

iceman1001, 2020-06-08:
Don't know if it makes any difference,  but setting it like you did,  doesn't that set it for the rest of the whole project?  look at the line second from the bottom,  that is how we set it using  module and PRIVATE

doegox, 2020-06-08:
If you can tell us how to reproduce your current cross-compiling environment, we can try on our side

xianglin1998, 2020-06-08:
1. Donwload NDK toolchain, see: https://dl.google.com/android/repository/android-ndk-r21-linux-x86_64.zip

2. Extract and remember the directory of NDK

3. I believe you already have cmake environment, so next run cmake
see: 
cmake -DCMAKE_TOOLCHAIN_FILE=**(Your NDK DIR)**/build/cmake/android.toolchain.cmake -DANDROID_ABI=armeabi-v7a -DANDROID_NATIVE_API_LEVEL=$android-19 CmakeLists.txt

4. Because we want to compile the Linux executable, it's better not to add the Android flag. Therefore, we need to manually modify the llexcc under liblua
see:
#if !defined(getlocaledecpoint)
#define getlocaledecpoint() (localeconv()->decimal_point[0])
#endif

5. All operations are performed under the client. You will get the same error as me without any accident.

error: implicit declaration of function 'fseeko' is invalid in C99

doegox, 2020-06-08:
thanks, I'll try

xianglin1998, 2020-06-08:
aaaaaaaaaaa， I'm realy sad，i always get error error: implicit declaration of function 'fseeko' is invalid in C99 is NDK error

see: https://android.googlesource.com/platform/bionic/+/master/docs/32-bit-abi.md
fu*k, i delete that define, it can work.

[image]

xianglin1998, 2020-06-08:
@doegox Now we need to deal with the macro definition. In liolib.c

doegox, 2020-06-08:
I pushed https://github.com/RfidResearchGroup/proxmark3/commit/526407ed5c219e263b4bdd31bea53f072109ee19 which fixed Lua under Android.
Now if you want to really compile the proxmark3 client and not a libpm3, we need readline.

xianglin1998, 2020-06-09:
OK，how can we solve the dependency problem of readLine? Do you want to bind a version of readLine or download it through shell and build it?

doegox, 2020-06-09:
I pushed more stuff to disable readline. There are still a few issues with commands calling kbd_enter_pressed() but the client is usable with commands less interactive or with option "-c".
My current way:
```
rm -rf build && mkdir build && cd build && \
cmake -DCMAKE_TOOLCHAIN_FILE=~/devel/00android/android-ndk-r21/build/cmake/android.toolchain.cmake -DANDROID_ABI=armeabi-v7a -DANDROID_NATIVE_API_LEVEL=android-19 -DSKIPREADLINE=1 -DSKIPBT=1 .. && \
make -j
```
Which now fails at finding bzip2.
For that one, it would be nice to be able to bring bzip2 somehow.
It's here: https://android.googlesource.com/platform/external/bzip2/

xianglin1998, 2020-06-09:
@doegox I found a cmake function 
[image]

It is can work to download and build readline.

xianglin1998, 2020-06-09:
[image]

This is build successfully, we need to include this path as a head file dir.

doegox, 2020-06-09:
interesting !
we'll need sth similar for bzip2

xianglin1998, 2020-06-09:
If has source, we can xxx.cmake again, download and cross compile.

doegox, 2020-06-09:
As I said, src is here https://android.googlesource.com/platform/external/bzip2/

doegox, 2020-06-09:
BTW I didn't merge the code with SKIPREADLINE in master yet, so if you do changes related to readline, please do it on the opt_readline branch.
I will merge it as soon as it's working fine on OSX.

xianglin1998, 2020-06-09:
OK

iceman1001, 2020-06-09:
@doegox  merged the readline changes.
It still the bzlib.h  issue.

xianglin1998, 2020-06-10:
[image]

Make successfully, we need reddline & ncurses & bzip2.
Next i will push to phone to test. if no problem, i will push changed to repo.

doegox, 2020-06-10:
great news !

xianglin1998, 2020-06-10:
[image]

It is working, bug have some problem:

1. backspace key delete char but console print space

2. detele key print char "~", yes,  under normal circumstances, the terminal will not respond to delete key.

3.  up key and down key can't  show history.

6. The user dir error.

It is worth mentioning that the problem of input exception also appears when Windows compiles proxmark3.exe. Their common feature is static compilation.

doegox, 2020-06-10:
Input exception ?
Can you push your current tests in a WIP branch?

xianglin1998, 2020-06-10:
Yes i can. do your have the android emulator or android phone(have root)?

we do test need:

1. make proxmark3 bin
2. Use adb tools push bin to /data
3. Use "adb shell" cmd to exe.

xianglin1998, 2020-06-11:
@doegox I have a RP in my repo, you can check and test.

iceman1001, 2020-06-11:
Great!

I can't find any PR in your repo but I looked at the cmake changes and we should use original bzip2 repo for download.  I doubt you want to start merging fixes in your fork from that repo, it just adds extra work for you and uncertainty for this repo not knowing if its uptodate or not.

xianglin1998, 2020-06-11:
Yes, I don't have PR, because there is no branch to store in the rrg repo, i can't push to master or any any any.

1. I use my repo for bzip2 because I'm in China, so I can't download it from Google's git during testing. The release version for RRG matser branch can use the origin, hahahaha, gfw bothers all china dev.    ;)

@iceman1001

iceman1001, 2020-06-11:
yeah,  I am sorry for that inconvenience with gfw you guys have to work with.
You are doing fine with your Github,  not that many Chinese devs use Github. 
I understand why you choose that way for you repo now.  Totally forgot about gfw.

xianglin1998, 2020-06-11:
It doesn't matter, hahaha. Now we need to test the usability of cross compilation and solve the problem of input exception. :)

doegox, 2020-06-11:
I'm sorry but I still didn't understand the "input exception". Can you explain it again ?

xianglin1998, 2020-06-11:
If you can compile the client on windows, you have a good chance to reproduce this problem when compiling all static.

1. The backspace key can delete characters, but print a space every time a character is deleted

For example, when you input "AAA" and click backspace, you can find that AAA is still on the terminal and print space: "AAA   "
See exists SO thread: https://stackoverflow.com/questions/47792556/strange-behavior-of-backspace-in-a-running-readline-code

2. The up and down keys cannot call history.

doegox, 2020-06-11:
You were missing the HAVE_READLINE define I think.

I merged your work in the original CMakeLists.txt in a branch:

Please test https://github.com/RfidResearchGroup/proxmark3/tree/cmake_android
with options as in https://github.com/RfidResearchGroup/proxmark3/blob/cmake_android/client/cmake-android.sh

I still want to change some stuff because I don't want the readline/ncurses/bzip2 compilation to happen with BUILD_IN_SOURCE=ON but it should be done somewhere in the build directory

xianglin1998, 2020-06-11:
OK, thanks.

[6 more replies]

closed 2020-06-12 by doegox