RfidResearchGroup/proxmark3 #758: [idea] Decide on Stack size/Use the whole RAM for BigBuf

slurdge, 2020-06-06:
Right now, we are not deciding on a specific stack size, giving all the space after the .bss section to stack. On current firmware, this totals a large 15KiB of space only for stack. With 64KiB of RAM, it may be a little bit too generous of us.
I propose we do the contrary: Stack Size if configurable at compile steps (so people can accommodate different scenarios), and then BigBuf will take all the remaining space.
As a test, I did a 
```c
static uint32_t * BigBuf = &__bss_end__;
```
and also a
```
MEMORY
{
    bootphase1 : ORIGIN = 0x00100000, LENGTH = 0x200             /* Phase 1 bootloader: Copies real bootloader to RAM */
    bootphase2 : ORIGIN = 0x00100200, LENGTH = 0x2000 - 0x200    /* Main bootloader code, stored in Flash, executed from RAM */
    osimage    : ORIGIN = 0x00102000, LENGTH = 512K - 0x2000     /* Place where the main OS will end up */
    ram        : ORIGIN = 0x00200000, LENGTH = 64K - 0x20        /* RAM, minus small common area */
    stack      : ORIGIN = 0x00200000 + 64K - 4K - 0x20, LENGTH = 4K /* Stack */
    commonarea : ORIGIN = 0x00200000 + 64K - 0x20, LENGTH = 0x20 /* Communication between bootloader and main OS */
}

/* Export some information that can be used from within the firmware */
_bootphase1_version_pointer = ORIGIN(bootphase1) + LENGTH(bootphase1) - 0x4;
_osimage_entry = ORIGIN(osimage);
_bootrom_start = ORIGIN(bootphase1);
_bootrom_end = ORIGIN(bootphase2) + LENGTH(bootphase2);
_flash_start = ORIGIN(bootphase1);
_flash_end = ORIGIN(osimage) + LENGTH(osimage);
_stack_start = ORIGIN(stack);
_stack_end = ORIGIN(ram) + LENGTH(ram) - 8;
```
And everything works fine.

If we want to be smart about it we need to:

1. Find a way to specify the stack size somewhere, e.g. in the link scripts;
2. Remove the hardcoded BIGBUF_LEN and make it a call (possibly also an API call for client);
3. Enjoy larger buffers for capture (IMHO 4KiB of stack may be enough, putting the BigBuf to 50KiB).

Bonus points: We use the stack size to estimate a BigBuf size and change compression window of LZ4 accordingly...

iceman1001, 2020-06-06:
now you touched something I haven't looked into.
Not my comfort zone at all.
The BIGBUF size has been a strange limitation and how we use BIGBUF is another mixed up place.
With both trace log data and signal collecting.   Hard to keep track of which while not breaking anything else.
The diff between 40 vs 50 kb in capture.. well.. with the decimate function it makes some difference.
long sniffs.   yeah,  those would be positive.  To get the max out of the device.

slurdge, 2020-06-07:
Yeah basically it doesn't change much from current situation, it just removes the somewhat arbitrary "40KiB and nothing else". Suppose you want a standalone with only 2k of stack , now you could have that without problems. Some people could also on the contrary have big nested functions and request a reduction in bigbuf.

The only difficult point is to know the size of bigbuf for the fpga_compress so that if someone wants a smaller bigbuf, its possible. Right now  I would suggest as a first step to say that bigbuf is _at minimum_ 40KiB, potentially more.

iceman1001, 2020-06-07:
sounds like a plan.   
go go go

iceman1001, 2020-06-09:
@slurdge ?

slurdge, 2020-06-10:
Yeah looking at it, but it touches quite a few areas, esp. the internal API with BigBuf. I need to get rid of #define first.

slurdge, 2020-06-11:
We can close now.

closed 2020-06-11 by slurdge