RfidResearchGroup/proxmark3 #777: pm3.sh -  more than 10 comports.

iceman1001, 2020-06-11:
**Describe the bug**
when finding a comport larger than 10,   com10 etc,  it reports it as  com1.

**To Reproduce**
1. assign a bluetooth comport above 10
2. ./pm3 --list

**Expected behavior**
More than 10comports 
and the comports above 10 should have correct numbering.

doegox, 2020-06-11:
Can you show me a sample of
```
powershell.exe -command "Get-CimInstance -ClassName Win32_PnPEntity | Where-Object Caption -like 'Standard Serial over Bluetooth link (COM*' | Select Name" 2> /dev/null 
```

We need to fix the `| awk '$0 ~ /COM/{print substr($6,2,4)}'`

iceman1001, 2020-06-11:
```
PS C:\Users\User> powershell.exe -command "Get-CimInstance -ClassName Win32_PnPEntity | Where-Object Caption -like 'Standard Serial over Bluetooth link (COM*' | Select Name"

Name
----
Standard Serial over Bluetooth link (COM10)

```

commit 47262ce1 (2020-06-11): fix #777  thanks @doegox
commit d229364a (2020-06-12): Variant for fix #777 independent of sentence and nr of com port digits
closed 2020-06-11 by iceman1001