RfidResearchGroup/proxmark3 #779: Big big ass

Windslab, 2020-06-11:
I am sory, but your tamplate for students. See what's going on:

1. No smiles: :snowflake:/:coffee: to Windows client. Yes CMD not supporting graphical as I know. Maybe better to remove it at all? It looks like graphity at the wall.
2. No colors to Windows client. Maybe better to remove it from *nix to? Or make some of preset for Windows Terminal, not make shure that's right way.
3. `hw tune` GUI for me is notworking, no reaction for changing lifts.
4. `hf plot` GUI is the same.
[image]
Some themes the picture is hiding(missing)
[image]
5. `hf 14b sniff` and `hf sniff` are not working at all, looks like a commands going to /dev/null! The `hf 14a sniff` working.
6. `hf mf pref` if I choice any command it always return me a it list again. Maybe I not get how its works, but in any case it return list again.

Stop play games. **A lot of work**.

doegox, 2020-06-11:
this issue doesn't help in any way and is too rude to be tolerated

closed 2020-06-11 by doegox

Windslab, 2020-06-11:
Hah, Are you kidding?

I have appreciate your humor, get back all. You a missing. You always all
delete. You can't play admin games.

I am an Volunteer, I haven't any money from code supporting, like you.

I am not rude, that's Russia, and I talking wery soft and regarding, you
didn't seen a rude. Bit irony.

Open again if you have a conscience.

iceman1001, 2020-06-12:
I never met such a rude person in a open source project before as you. 
Learn some manners and a understand how open source project works.

Windslab, 2020-06-12:
We all do everything directly and speak clearly and on business. Thay all Russians who tryung talking kindly like you is simmulating. Make shure for me all your answering in any times is very strange, like my questions to you.

Don't worry at business mails is the same like my in all peoples. We not teaching manners at schools. Only Vodka/Balalayka xD

My position to brake your templates about manners. And I already told not for nothing to you all that's we from different countries.

You better think about money from sales of RDV4.01 :) People will read they all bugs and you will losing your money. Not worry I know more about business more than you think.

Hope that's will be better.

Best Regards

doegox, 2020-06-12:
Let's make this clear again:
**We earn 0$, we're unrelated to the sellers of Proxmark, we're not an after-sales service, we don't owe you anything, we're spending our free time trying to make the proxmark firmware better**

And stop bringing your nationality to justify your acts. Over years we interacted with many many nationalities and cultural differences. It's clearly not about Russians, it's about **you**, sir.

merlokk, 2020-06-12:
About business...
I'm can do what you want in the proxmark for like $65/hour. All you want) All-all you want....
And I think all top 10% in the contributors page can do the same not more as $100/hour.
Business.
As for our free time - we do what we want. Some of us maintain this repository regularly. Some not.
But its our time.
Fill free to add your time here. Develop some features, etc.
If you want us to do something - just look at first part of message. It very clear.
BTW. I live near you) so know many about exUSSR....

Windslab, 2020-06-12:
> About business...
> I can do what you want in the proxmark for like $65/hour. All you want) All-all you want....
> And I think all top 10% from the contributors page can do the same not more than $100/hour.
> Business.
> As for our free time - we do what we want. Some of us maintain this repository regularly. Some not.
> But its our time.
> Feel free to add your time here. Develop some features, etc.
> If you want us to do something - just look at first part of message. It very clear.
> BTW. I live near you) so know many about exUSSR....

Hah, If you have free time, you have an easy low paid job. Especially in Kharkiv city, $65/day maximum.
$65 per hour, really? You are hungry and seems really have an low salary :) Or looks liake a youthful maximalizm :)

I have intouch with my PhD friends in UK, even my IT teacher from Oxford has less then you have asked. For a moment, did you know how much Research Manager/Programmist get per month at the typical LTD?  It's about £2000. Top seniors's with his since works is about £4000.

No I honestly Russian, just was born at Ukraine. Tampluary ride on around.

The same an Easy funcional, bigger price, the same bugs. You don't fixed bugs that's has posted for you. That's why you have $0.

Hardworkers for free. Wishing you the best. Haha.

merlokk, 2020-06-12:
You cant imagine that someone can have some more money? we have junior salaries (after university) from $2000 here.... $65/day is a salary for cleaners in the supermarket. (i'm not sure...  maybe more)
And i'm not a junior....  Some time ago I had USSR red passport if it sounds something to you.

But here not a place to check who is better. My work you can see at github.

If you have friends from somewhere that can write you code for $1/hour - just go there.
Then in like 10 years you will realize why i cost $65/h. 
And i'm not want you as client. My client's queue like 3 months. Its just to a sample how to do things in the real world. 

so. if you cant pay - just dont pay. do not bla-bla-bla.

Windslab, 2020-06-12:
> You cant imagine that someone can have some more money? we have junior salaries (after university) from $2000 here.... $65/day is a salary for cleaners in the supermarket. (i'm not sure... maybe more)
> And i'm not a junior.... Some time ago I had USSR red passport if it sounds something to you.
> 
> But here not a place to check who is better. My work you can see at github.
> 
> If you have friends from somewhere that can write you code for $1/hour - just go there.
> Then in like 10 years you will realize why i cost $65/h.
> And i'm not want you as client. My client's queue like 3 months. Its just to have a sample how to do things in the real world.
> 
> so. if you cant pay - just dont pay. do not bla-bla-bla.

Sad to read this, I touched you @merlokk 

You talking like a demaged person who voices around the worlds like "Hey we are living betetter then UK or Europian". That's wery funny for peoples from outside, who working for example in Paris for the 3000 Euro per month after taxes, or Yahoo programists in USA about $5000.

But do you know, that's good that's you are trying.

Have are you asked blues, what they think about your country and city? Why Slovenian put filial in Kharkiv? And how in your country who got a lot of credits, salarys bigger then in EU/UK up to twice?

That's because for low paying to you, that's why in Slovenia this will be really 2500 Euro per month. For you will be okay 1000 and that's your real salary.

In your rich Slovenian Company even can't use something another than Bootstrap. Yeah nice specialists hah. The companys meting by it websites.

I no need your services at all, and you better to change to red passport again. Zemlyak :)

Good Luck

FFY00, 2020-06-12:
Hey, can someone stop all this BS and lock the topic? They will keep going on as long at there is engagement.

Windslab, 2020-06-12:
I not inviting you to these topic. My Franch friend :)