RfidResearchGroup/proxmark3 #780: [Bug] Founded bugs

Windslab, 2020-06-12:
1. No smiles: :snowflake:/:coffee: to Windows client. Yes CMD not supporting graphical as I know. Maybe better to remove it at all? It looks like graphity at the wall.
2. No colors to Windows client. Maybe better to remove it from *nix to? Or make some of preset for Windows Terminal, not make shure that's right way.
3. `hw tune` GUI for me is notworking, no reaction for changing lifts.
4. `hf plot` GUI is the same.
[image]
Some themes the picture is hiding(missing)
[image]
5. `hf 14b sniff` and `hf sniff` are not working at all, looks like a commands going to /dev/null! The `hf 14a sniff` working.
6. `hf mf pref` if I choice any command it always return me a it list again. Maybe I not get how its works, but in any case it return list again.

iceman1001, 2020-06-12:
Use the provided template as seen in the screen clip below to fill in one bug at the time with all details.  This issue doesn't fullfill the requirments and will be closed.

[image]

closed 2020-06-12 by iceman1001

Windslab, 2020-06-12:
Thank you.

I will do that next time, but also wishing to improve your template.

But could you plase take a look at the bugs?

doegox, 2020-06-12:
Please read carefully https://instabug.com/blog/how-to-write-a-bug-report-the-ideal-bug-report/
* title => you do it wrong
* description => you do it wrong
* no more than one bug in your report. => you do it wrong
etc

Windslab, 2020-06-12:
> Please read carefully https://instabug.com/blog/how-to-write-a-bug-report-the-ideal-bug-report/
> * title => you do it wrong
> * description => you do it wrong
> * no more than one bug in your report. => you do it wrong
> etc
> 

Okay. Thanks for tips.