RfidResearchGroup/proxmark3 #781: command history not save when ctrl-c exiting client

iceman1001, 2020-06-12:
**Describe the bug**
When you press ctrl-c to exit / interupt the client (it happens)  the command history isn't saved.
So you loose the commands you tried. 

**To Reproduce**
1. enter client.,
2. run a command you didn't have in your history.  
3. press ctrl-c 
4. enter client again
5. up arrow,  doesn't show the command you ran.

**Expected behavior**
I expect the client to save the history after each run?

**Desktop (please complete the following information):**
latest source from repo.

doegox, 2020-06-12:
we should be able to trap the ctrl-c and save before exit
I would avoid writing the file at each cmd

doegox, 2020-06-12:
```c
#include <signal.h>
#include <stdlib.h>

void intHandler(int) {
    savehist...;
}

int main(int argc, char *argv[]) {
    struct sigaction act;
    act.sa_handler = intHandler;
    sigaction(SIGINT, &act, NULL);
...
}
```

doegox, 2020-06-25:
this might help https://stackoverflow.com/a/34675886

iceman1001, 2020-06-25:
if you do,  you loose  ctrl-r ,  ctrl-u   etc...

iceman1001, 2021-10-14:
yeah,  Dunno if I remember it correct,  but I looked into this too a year ago,
and I think I needed to guard the calls for signal handling and for the log history which is readline related.

doegox, 2021-10-16:
What is the status exactly today ?
On Linux it seems to always save the history properly on ctrl-c. I seldom encountered history files saved with corrupted filenames in the current workdir (but never managed to reproduce it when I wanted), that's why I pushed now a history_path=NULL to avoid to be used after free.
On Windows (PS) the signal handling code is commented.

closed 2024-04-09 by iceman1001