RfidResearchGroup/proxmark3 #782: Android cross-compiled client over adb shell: readline backspace issues

doegox, 2020-06-12:
**Describe the bug**
Android client accessed over adb has readline issues.
Backspace deletion is not effective, messing up the screen when resizing term window, trying to delete a char or using history arrows.

**To Reproduce**
Steps to reproduce the behavior:
1. Cross-compile the client for Android (cf client cmake)
2. Run client on Android via adb shell
3. Do any action requiring readline to redraw the cli (arrows, backspace,...)
4. Prompt messed up.

See https://github.com/RfidResearchGroup/proxmark3/issues/756#issuecomment-643172352 for an example

xianglin1998, 2020-06-15:
Similar questions: https://stackoverflow.com/questions/57857568/cross-compiled-python-curses-problem-over-ssh-but-not-on-serial-port

doegox, 2020-06-15:
very promising findings !

xianglin1998, 2020-06-15:
Yes, the console problem origin is terminfo map file lost.

I found a env arg: TERMINFO 
from: https://stackoverflow.com/questions/7158602/how-to-set-custom-search-paths-for-the-terminfo-database-when-building-ncurses-f/40269919#40269919

export TERMINFO = (The path to terminfo files dir)

The proxmark3 run on termux is working.

We have a problem that we need to pay attention to, which will affect the static compilation of proxmark3, why does proxmark3 compile statically on windows have this problem? yes, they are also missing.

This is a disgusting dependency loss, I wasted four days on this issue.

Next, let's optimize the execution of libpm3 and CLI to avoid the problem of lack of dependency.

Thank you for your contribution.
Thanks .

@doegox 
@iceman1001

doegox, 2020-06-15:
Well done !
Do you have changes to commit to the repo to handle this issue nicely ?

xianglin1998, 2020-06-16:
Finally, I chose to follow the terminfo specification and hard code the path according to the environment of the target machine. In this case, there are two solutions:

1. Path hard coding at compile time, which is the most basic implementation, takes effect when the environment variable is not set.

2. Before running, set the relevant environment variable of terminfo, so that ncurses can find the customized path.

I added some resource cleanup at compilation, avoided using the old cache.

xianglin1998, 2020-06-16:
Your cmake build bzip2 have some problem, i am fixed.

iceman1001, 2020-06-18:
Is this one solved now with latest merges?

doegox, 2020-06-18:
> Finally, I chose to follow the terminfo specification and hard code the path according to the environment of the target machine. In this case, there are two solutions:
> 1. Path hard coding at compile [...]

Can you show a detailed example ?

xianglin1998, 2020-06-18:
Yes i can.

1. https://github.com/RfidResearchGroup/proxmark3/blob/da00cc040518dd8fabaf5f2dc36e87ff172c3c4d/client/CMakeLists.txt#L82

After --host=arm, add --datadir=(your terminfo path) to bring ncurses find the terminfo files.
Maybe we should provide a cmake variable to control? For example: cmake -DTERMINFO=xxxdir

2. At runtime, we can set a terminal env. TERMINFO or TERMINFO_DIRS
set TERMINFO, the ncurses only find in that path.
set TERMINFO_DIRS, the ncurses will find in Default and TERMINFO_DIRS

Such as Termux(Android terminal):

a. export TERMINFO=/data/data/com.termux/files/usr/share/terminfo
b. export TERMINFO_DIRS=/data/data/com.termux/files/usr/share/terminfo:/usr/share/terminfo

finally:

There are two ways to solve this problem. The first is to specify the terminfo path in the target device at compile time. The second is to guide ncurses to actively find the specified path by setting environment variables at runtime.

@doegox 
@iceman1001 

:)

doegox, 2020-06-18:
I pushed sth for you... Statically embedded term info, no need for external db anymore, so proxmark3 works fine without need for termux.
https://github.com/RfidResearchGroup/proxmark3/commit/94edc41c3c794e7a1b9b2f5e1cde3fdef6c69835
and a few compilation flags to get compilation faster as we don't care about other artefacts besides the library itself.

xianglin1998, 2020-06-23:
I'm finally free. I'll test it.

xianglin1998, 2020-06-23:
Wow, it's a great implementation, and you've freed me from a complex environment configuration.

@doegox

doegox, 2020-06-23:
I guess we can close this one

closed 2020-06-23 by doegox