RfidResearchGroup/proxmark3 #783: [idea] Low hanging fruit

Danyc0, 2020-06-12:
**Is your feature request related to a problem? Please describe.**
I've been a Proxmark user for ~5 years now and have spent my fair share of time trawling through the code, and often failing to get to the root of my issues.
I was thinking that maybe something as simple as a 'low hanging fruit' issues tag (and maybe something similar on the public roadmap) could help people like me get to know the code base better and hopefully be more able to properly contribute to the project.

**Describe the solution you'd like**
A 'Low hanging fruit' tag on GitHub issues and the public roadmap.

**Additional context**
I'd just like finish by thanking all the contributers past, present and future for all their hard work on the Proxmark and all who have made it the amazing tool it is today.

doegox, 2020-06-12:
Hello, thank you for the suggestion!
It's interesting but a bit challenging. Indeed when we see something easy to fix we tend to immediately solve it. Which is not really the best to attract people like you to contribute with easier bugfixes, I must admit.
Currently the opened issues are quite complex to start with imho, same with the ideas on the roadmap.
Meanwhile, one suggestion, maybe not the most sexy one but allowing to poke a bit around the whole code and learn along the path:
Compile with `make GCCEXTRA=1` and try to fix a few warnings.

doegox, 2020-06-12:
I tried to tag a few of the issues I think they're not too hard: https://github.com/RfidResearchGroup/proxmark3/labels/low-hanging%20fruit

iceman1001, 2020-06-13:
The `make GCCEXTRA=1`  .... will give a bunch of low hanging fruits.  :)

iceman1001, 2020-06-26:
Since we have the label and have labeled some issues and with more issues on the way I feel ok to close this one.

Nice suggestion. I hope it will lead to more peoples participating

closed 2020-06-26 by iceman1001

Danyc0, 2020-06-26:
Amazing, thank you both! I'll see what I can do!