RfidResearchGroup/proxmark3 #788: Compiling on Mac OS Catalina Fails

infosecconsultant, 2020-06-13:
**Describe the bug**
When compiling with: make clean && make install the compilation fails with the following error:

```[-] CC start.c
In file included from start.c:18:0:
BigBuf.h:42:1: error: target attribute is not supported on this machine [-Werror=attributes]
 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen, uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);
 ^
cc1: all warnings being treated as errors
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2
```
**To Reproduce**
Steps to reproduce the behavior:
1. Clone current repo
2. make clean && make all
3. Fails. 

**Expected behavior**
Compilation to be successful. 

**Screenshots**
```
===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all

===================================================================
Client platform:   Darwin
GUI support:       QT5 found, enabled
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.7 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
compiler version:  Apple clang version 11.0.3 (clang-1103.0.32.59)
===================================================================
[*] MAKE amiibo
make[2]: Nothing to be done for `all'.
[*] MAKE cliparser
make[2]: Nothing to be done for `all'.
[*] MAKE jansson
make[2]: Nothing to be done for `all'.
[*] MAKE hardnested
make[2]: Nothing to be done for `all'.
[*] MAKE lua for macosx
make[3]: Nothing to be done for `all'.
[*] MAKE mbedtls
make[2]: Nothing to be done for `all'.
[*] MAKE reveng
make[2]: Nothing to be done for `all'.
[*] MAKE tinycbor
make[2]: Nothing to be done for `all'.
[*] MAKE whereami
make[2]: Nothing to be done for `all'.
[=] LD proxmark3
[*] MAKE bootrom/all
compiler version:  arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
make[1]: Nothing to be done for `all'.
[*] MAKE fpga_compress/all
make[1]: Nothing to be done for `all'.
[*] MAKE armsrc/all
compiler version:  arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
[-] CC start.c
In file included from start.c:18:0:
BigBuf.h:42:1: error: target attribute is not supported on this machine [-Werror=attributes]
 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen, uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);
 ^
cc1: all warnings being treated as errors
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2
```

**Desktop (please complete the following information):**
 - OS: Mac OS Catalina 

**Additional context**
Have an older repo present that was compiled on this system so it should still be able to compile I'd have thought? 

```
  ██████╗ ███╗   ███╗ ████╗            
  ██╔══██╗████╗ ████║   ══█║           
  ██████╔╝██╔████╔██║ ████╔╝           
  ██╔═══╝ ██║╚██╔╝██║   ══█║     ❄️  iceman@icesql.net ☕          
  ██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
  ╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          
          

 [ Proxmark3 RFID instrument ]           

 [ CLIENT ]           
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 11.0.3 (clang-1103.0.32.59) OS:OSX ARCH:x86_64          

 [ PROXMARK3 RDV4 ]           
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]           
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/69b6a8f7-dirty-unclean 2020-04-20 23:57:41
       os: RRG/Iceman/master/69b6a8f7-dirty-unclean 2020-04-20 23:57:48
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16          

```

infosecconsultant, 2020-06-13:
I have a second spare device with a fresh install. I thought perhaps there was something on the other device that was causing issues. 
The compilation steps followed above also fail, although at a different point.

```
PC:proxmark3 user$ make all

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
===================================================================
Client platform:   Darwin
GUI support:       QT not found, disabled
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
compiler version:  Apple clang version 11.0.3 (clang-1103.0.32.62)
===================================================================
[-] CC src/proxmark3.c
src/proxmark3.c:302:21: error: use of undeclared identifier 'rl_event_hook'
                    rl_event_hook = check_comm;
                    ^
1 error generated.
make[1]: *** [obj/proxmark3.o] Error 1
make: *** [client/all] Error 2
PC:proxmark3 user$ 
```

iceman1001, 2020-06-13:
You will need to update your ARM-EABI-NONE tool chain to a later version.
On homebrew,   if you reinstall,  it will download a later working tool chain.

infosecconsultant, 2020-06-13:
Ah okay. I was not aware that was required to install.

I have run the following commands (as per: https://github.com/armmbed/homebrew-formulae):
```
brew update
brew tap ArmMbed/homebrew-formulae
brew install arm-none-eabi-gcc
```
Followed by
```
make clean && make all
```
I realised when I tried on the other machine that there was a version of arm-none-eabi-gcc in the RfidResearchGroup/proxmark3 homebrew repo.

```
brew tap RfidResearchGroup/proxmark3
brew install RfidResearchGroup/proxmark3/arm-none-eabi-gcc
```
Followed by
```
make clean && make all
```
Both of these files resulted in the same error.
Cheers

iceman1001, 2020-06-13:
Ok,  so this issue is about homebrew.  Then it should be in this repo
https://github.com/RfidResearchGroup/homebrew-proxmark3

infosecconsultant, 2020-06-13:
My issue isn't so much with getting a functional version, but rather specifically compiling it myself.
As you suggest, installing the client from homebrew works just fine. 

Is there anything else I could have missed?

doegox, 2020-06-13:
> Both of these files resulted in the same error.

Which error are we talking about now ? 
This one ?
```
src/proxmark3.c:302:21: error: use of undeclared identifier 'rl_event_hook'
                    rl_event_hook = check_comm;
                    ^
```
This one is because you're missing the readline.
Maybe you don't want to use the proxmark recipe, still you'll need the same dependencies, see https://github.com/RfidResearchGroup/homebrew-proxmark3/blob/master/proxmark3.rb#L14-L17

doegox, 2020-06-13:
These are explained here btw
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md#homebrew-mac-os-x-developer-installation

infosecconsultant, 2020-06-14:
Ah! 
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md#homebrew-mac-os-x-developer-installation

This sorted it out nicely. The key thing that was missing was to Install makefile dependencies
```
brew install recode
brew install astyle
```

Thanks for your help folks!

closed 2020-06-14 by infosecconsultant

dantrauner, 2020-06-30:
Before I open a new issue, I figured I'd try my luck here – I seem to be failing to compile on macOS Catalina with:

```
[-] CC src/proxmark3.c
src/proxmark3.c:261:9: error: use of undeclared identifier 'rl_catch_signals'
        rl_catch_signals = 1;
        ^
src/proxmark3.c:262:9: error: implicit declaration of function 'rl_set_signals' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        rl_set_signals();
        ^
src/proxmark3.c:343:21: error: use of undeclared identifier 'rl_event_hook'
                    rl_event_hook = check_comm;
                    ^
3 errors generated.
make[1]: *** [obj/proxmark3.o] Error 1
make: *** [client/all] Error 2
```

Except I have `readline` installed (and installed `recode` and `astyle` manually as well, all via Homebrew). Any ideas? I also tried force-linking `readline` with brew, as well as setting `LDFLAGS`/`CPPFLAGS`.

iceman1001, 2020-06-30:
Strange, I just pulled and compiled on 10.15 without any issue.

```
trusteduser@TrustedUsers-Mac proxmark3 % gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.29)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

doegox, 2020-06-30:
Try compiling with verbose (make VERBOSE=1 with main makefile or make V=1 with cmake) and check if injected paths for readline seem correct

dantrauner, 2020-06-30:
@doegox that'll do it! The path was wrong because I don't have Homebrew at `/usr/local` (it's at `~/homebrew`). This is probably something I should change anyways, but to confirm, manually changing the `/usr/local` prefix in `LDLIBS` and `INCLUDES` on L261/262 in the client Makefile to point at where my `readline` was installed fixed this.

doegox, 2020-06-30:
great!