RfidResearchGroup/proxmark3 #789: [idea] Use pkg-config to detect dependencies

RedL0tus, 2020-06-13:
**Is your feature request related to a problem? Please describe.**
There may be certain reason that the user want to install certain dependency in an irregular location, and it is sometimes annoying to configure CFLAGS/LDFLAGS for them. For example:

 - The [`Nix`](https://nixos.org) package manager: A brief description is that It downloads all of the packages into `/nix` and then create symbolic links to the user environment, which means it does not care about [FHS](https://refspecs.linuxfoundation.org/fhs.shtml) or [LSB](https://refspecs.linuxfoundation.org/lsb.shtml) at all. It can also be used in OSs other than NixOS as a secondary package manager. I kind of like it and I think it performs better than Homebrew on most things, but it makes me feel cursed if I need to write all of the CFLAGS/LDFLAGS by hand.
   - For example, the absolute path to `libreadline` looks like this on my machine: `/nix/store/jz489ki9hsblmk0qkv5i1xxmdz24x7q6-readline-6.3p08/lib`
 - Homebrew now refuses to link macOS provided/shadowed softwares, and libreadline is one of them.
 - The package provided by the official repository is outdated/buggy/too-new.

**Describe the solution you'd like**
Luckily [`pkg-config`](https://www.freedesktop.org/wiki/Software/pkg-config/) is a standard solution in unix-like environments, created by fd.o. With the help of `pkg-config`, it would be painless to deal with the previous described scenarios. There are only one environment variable needed at most for `pkg-config` to do its job (`PKG_CONFIG_PATH`). And in the Nix case, the package manager itself will be taking care of it.

Also, `pkg-config` is [kind-of in use](https://github.com/RfidResearchGroup/proxmark3/blob/master/client/Makefile#L109) right now, which means extending its usage will not introduce any new dependency at all, and can be combined with the traditional CFLAGS/LDFLAGS method as well.

**Describe alternatives you've considered**
Or use other building systems like [`CMake`](https://cmake.org), [`Meson`](https://mesonbuild.com), or the more traditional combination of [`autotools`](https://www.gnu.org/software/autoconf/) and [`automake`](https://www.gnu.org/software/automake/).

Switching from plain Makefile to them will introduce some extra dependencies but they will handle dependency management better and easier than bare hands. They also offer extra features like a uniform way to customize the options from the command line.

Switching to them will also make the life of packagers for Linux distros easier since most of the packaging tools has integration with those building systems.

**Additional context**
Nothing more but a ridiculous usage of CMake: It can be used to write a path tracing engine:
https://zhuanlan.zhihu.com/p/123419161 (This passage is in Chinese, but the source code is attached directly).

iceman1001, 2020-06-14:
For the client,  we already have a `cmake` option inplace.  Have you tried it?

doegox, 2020-06-14:
If you have suggestions for more usage of pkg-config, go ahead and propose PRs.
For the Makefile and/or the CMakeLists.
But beware some libs don't provide always the .pc glue, and some libs have a .pc but only on some OSes, not all.
And we need to have a solution that compiles on all the OSes currently supported (Windows PS3.4, Windows WSL, OSX Homebrew, various Linux, RPi, RPIZ,...).

RedL0tus, 2020-06-14:
@iceman1001

> For the client, we already have a `cmake` option inplace. Have you tried it?

That looks really great. But I recommend to use CMake modules like find_xxx rather than handle them manually.

@doegox
> If you have suggestions for more usage of pkg-config, go ahead and propose PRs.
> For the Makefile and/or the CMakeLists.
> But beware some libs don't provide always the .pc glue, and some libs have a [...]

Pretty much all of the proper Linux distribution utilizes pkg-config and respect the FHS, it will not be a problem if we don't use any magic. The real problem may be Windows and that's partially why I recommend a full transition to `CMake`, because it does work on all of those platforms and helps while dealing with those platforms. And because of the Android cross compilation support is needed and present, CMake is needed anyway. Maintaining a proper set of CMake scripts may be somewhat easier to maintain than two sets of build scripts at the same time.

But there is one thing that I think we need to discuss. CMake can find libraries automatically but only in standard locations, which means if we need to keep the same `make all` one-liner style (just don't know how to name it correctly), a ton of workarounds are needed just as what's in the current `CMakeLists.txt`. I think is makes it overly complicated and still cannot cover all of the possible cases. So I'm wondering why not only do the bare minimum and leave it to the users to do the custom configurations. It will mostly be the same on most Linux distributions (including RPi, of course), and on Windows/macOS/Android, there are custom configurations to do anyway. I just forked the repo and preparing to propose a PR about that :)

RedL0tus, 2020-06-14:
Another proposal is whether to separate the firmware part from the client... It certainly needs some discussions.

iceman1001, 2020-06-15:
> Another proposal is whether to separate the firmware part from the client... It certainly needs some discussions.
Old rebunding question that pops up every now and then.  Short answer: the client and fw is tightly integrated,  where we recommed strongly using the same build of fw as well as client.   I am afraid that will not change.

As said this project compiles on several different hardware and OS's,   we spend quite much time making it that stable and easy for ppl to use.   We get quite less "it doesn't compile" questions nowdays than before.  So we have taken some pride in maintain it that way. 
Right now we are focusing of bringing Android compilation of the client nicely into the repo for easy usage on that platform.  Not to mention the libpm3 idea which does makes the maintainers quite busy as it is. 

With that said,  there is nothing stoping you from making your own fork and implement all different kinds of ideas.  I encurage that.  The Proxmark3 software was made to enable easy swapping of fw and client.

RedL0tus, 2020-06-15:
> > Another proposal is whether to separate the firmware part from the client... It certainly needs some discussions.
> 
> Old resurfacing question that pops up every now and then.  Short answer: the client and fw is tightly integrated,  where we recommed strongly using the same build of fw as well as client.   I am afraid that will not change.
> 
> As said this project compiles on several different hardware and OS's,   we spend quite much time making it that stable and easy for ppl to use.   We get quite less "it doesn't compile" questions nowdays than before.  So we have taken some pride in maintain it that way. 
> Right now we are focusing of bringing Android compilation of the client nicely into the repo for easy usage on that platform.  Not to mention the libpm3 idea which does makes the maintainers quite busy as it is. 
> 
> With that said,  there is nothing stoping you from making your own fork and implement all different kinds of ideas.  I encurage that.  The Proxmark3 software was made to enable easy swapping of fw and client.  

Sorry I didn't think that far, but what I mean is just to separate them into different codebases and that does not necessarily mean to allow using different versions of the client and the fw. Because on the code itself, there are no strong relationship between the two parts. They just target different things and require different dependencies/toolchains. Separate them won't bring too many changes and it can be a starting point to really make them version-agnostic.

It's just a proposal anyway :)

iceman1001, 2020-06-15:
Thanks, I am quite aware of the codebase and what the different parts does and how they integrate.  

I usually encurage ppl to add functions instead of refactor stuff.  Trust me when I say that this codebase was created by more than hundred contributors over the atleast tvelve years of existence.   It never had a overall vision and plan.  Nor strategy,  everyone added something that solved their specific problem. For instance, many has said "rewrite the client in XX language" ,   many has tried.  All has failed.  So the complexity is a bit more than meet the eye.    The state of this repo is miles away from the offical repo and has become so much better.  This has been an ongoing slow continous process but don't make the mistake and think its easy or simple.

Do enjoy the rabbithole of proxmarking,   realise the potentials and enhance it in order to overcome some rfid cards.   If you want some low hanging fruit to contribute with,  look for that label here on the issue or read the public road map,  or read the wishlist.   Plenty to improve.  Get excited and get started!

doegox, 2020-06-15:
> So I'm wondering why not only do the bare minimum and leave it to the users to do the custom configurations

I'm afraid that's not an option. As @iceman1001 said, we currently support many OSes out-of-the boxes and we want to keep it like that.

Still, we welcome improvements, including small refactorings that help making things easier to handle (e.g. the recent move of deps to /client/deps) but at one strong condition: what used to work needs to work, including support of all these OSes. And trust me that's quite of a challenge.

I made a draft that explains where we are and what still needs improvement in https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Development/Makefile-vs-CMake.md

RedL0tus, 2020-06-15:
> I'm afraid that's not an option. As @iceman1001 said, we currently support many OSes out-of-the boxes and we want to keep it like that.
>
> Still, we welcome improvements, including small refactorings [...]

I think maybe I can start by improving the current CMakeLists. Thank you for the information.

closed 2020-06-15 by RedL0tus