RfidResearchGroup/proxmark3 #800: [Bug] hf sniff

Windslab, 2020-06-19:
**Describe the bug**
When I write `hf sniff` to console with no/card I got nothing at all

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'pm3 folder'
2. Click on 'proxmark.exe'
3. Scroll down to 'end'
4. See error

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Win 10 x64]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
 - hw status OK
 - data tune -89 dB

**Additional context**
Add any other context about the problem here.

iceman1001, 2020-06-19:
This is no bug,  if you want to ask how the proxmark3 client works you should use the forum to ask your questions.

closed 2020-06-19 by iceman1001

iceman1001, 2020-06-22:
remade this command to work with NG and new dynmic stack.

Windslab, 2020-06-22:
Now it's much more better! But... But... I not see data. I think It's need's to show me any way Sinus>01010>HEX. What you think?

[image]

iceman1001, 2020-06-22:
You seem not to understand what the command does.
It captures raw signal data.  Nothing else.  You can look at the captured signal using the plot window. 

For decoded sniffs, you need to use the other sniff commands.

Windslab, 2020-06-22:
Oh, okay I got that. I was used it wrong. Yeah I see the sinus. Thank you.

Please hold on I have also one tip for you.

When I 'stop' sniff by Enter the `hf 14a list` exception any way and looks like this:

[image]

If I use pm3 button, seems it's works:

[image]

As my opinion key `Enter` must stop a stiff, like a PM3 button. At the moment it's actions looks like it stop sniff but going to exception. Do you support this idea?

iceman1001, 2020-06-22:
This issue is about `hf sniff`  and closed.