RfidResearchGroup/proxmark3 #801: [Bug] hf 14b sniff

Windslab, 2020-06-19:
**Describe the bug**
When I excute `hf 14b sniff` after that got only \n

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'pm3 folder'
2. Click on 'proxmark3.exe'
3. Scroll down to 'end'
4. See error

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Win 10 x64]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
 - hw status OK
 - data tune -89 dB

**Additional context**
After sniff the `hf list` does't work

[image]

I wishing to get an my data, right?

iceman1001, 2020-06-19:
`Hf 14b sniff`  would need some more love.   The 14B commands are kind of unstable. Feel free to fix them.

Windslab, 2020-06-19:
> `Hf 14b sniff` would need some more love. The 14B commands are kind of unstable. Feel free to fix them.

Not all people professional programmist, like you @iceman1001

Could you please give any example how it's works? For example for ID...

iceman1001, 2020-06-19:
You should ask questions about usage on the proxmark forum.  That is why it exists. 
It seems like you don't know about the built-in help function  `hf 14b sniff h`

Windslab, 2020-06-19:
Please make sure I did all, and nothing. I am writing here because all not
working at all. Okay?

Windslab, 2020-06-19:
See what I got:

[image]

merlokk, 2020-06-19:
what reader 14b do you have and what card?
and what command did you send to card?

Windslab, 2020-06-19:
> what reader 14b do you have and what card?
> and what command did you send to card?

An Easy reader, but it's okay. I trying to understand how being data exchange between my Red ID and CL0WN did.
Do it possible to for example save finger, photo?

Windslab, 2020-06-20:
I need make sure that's all Proxmark junkies(it this is rude?) has understood my not directly talking. Yes? Happy. Do you happy? :)

merlokk, 2020-06-20:
ok, lets try... the last attempt...

There are many systems that better than proxmark in usability angle. They have customer support and solid concept.
you can just buy it and thats all.
but...  it costs like $10k-$20k at start plus $3k-$7k yearly.
and one thing....  i have one of this system (it costs much more than i mentioned) and i was need to buy proxmark like 15 years ago because it not works as i need. and i was need to learn like 10 years... the easiest way to learn - read forum that Iceman told you....

the 2nd way (surprise...)
is buy some special tools for hacking cards...  it costs not less than $10k...
and...   you will not believe.... most of them have proxmark inside or something like....
and i think you can search it at russian forums...
here noone will teach you how to hack... because.

again so strange way to try to learn something by behave rudely with the creators of the device...

iceman1001, 2020-06-20:
Keep the issue clean and all posts releated to the issue.
Use the proxmark3 forum instead.

Windslab, 2020-06-20:
@merlokk Oleg, you all about money. Who always talking about it, usualy missing it :) I understood that you have a especial **Israel** machines...

@iceman1001 For me `hf sniff 14b` not working at all, I did all combinations. Have no any data after sniff operation. How I can invest?

iceman1001, 2020-06-21:
See my first post.

doegox, 2020-08-16:
New 14b code works now, try latest master.
```
[usb] pm3 --> hf 14b sniff
[usb] pm3 --> 
[#] Starting to sniff. Press PM3 Button to stop.
[#] Sniff stopped
[#] 
[#] Sniff statistics
[#] =================================
[#]   DecodeTag State........0
[#]   DecodeTag byteCnt......0
[#]   DecodeTag posCount.....0
[#]   DecodeReader State.....0
[#]   DecodeReader byteCnt...0
[#]   DecodeReader posCount..2
[#]   Trace length...........2709
[#] 
hf 14b list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 2709 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443B - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       3092 | Rdr |05  00  00  71  ff                                                       |  ok | REQB
 4294935316 |      27412 | Tag |50  c1  2c  8b  1b  00  00  00  00  91  71  71  50  94                   |  ok | 
      55656 |      58772 | Rdr |1d  c1  2c  8b  1b  00  08  01  00  25  ba                               |  ok | ATTRIB
      58772 |      73108 | Tag |00  78  f0                                                               |  ok | 
     162772 |     165908 | Rdr |02  00  a4  04  00  08  31  54  49  43  2e  49  43  41  1f  fb           |  ok | ?
     166036 |     188564 | Tag |02  6a  82  4b  4c                                                       |  ok | 
     251756 |     254868 | Rdr |03  00  a4  00  00  02  3f  00  1f  19                                   |  ok | ?
     178836 |     178836 | Tag |03  85  17  00  01  00  00  00  12  12  00  00  01  03  01  01  00  41   |     | 
            |            |     |41  41  00  00  00  00  00  00  90  00  9a  b6                           |  ok | 

```

closed 2020-08-16 by doegox