RfidResearchGroup/proxmark3 #803: [Bug] hw tune

Windslab, 2020-06-19:
**Describe the bug**
As my opinion the `hw tune` is doesen't work and the graphic is dead. I not see any changers when mooving resizes

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'pm3 folder'
2. Click on 'proxmark.exe'
3. Scroll down to 'end'
4. See error

[image]
[image]
[image]
[image]
[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: [Win 10 x64]
- inside proxmark3 client run the following commands and paste the output here.
- hw HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
- hw status OK
- data tune -89 dB

**Additional context**
I would like to get any response from graph when doing resize of the signal windows, yes?
Also I think the sinus line must bee more bold and digit's a bit more size.

And I not wishing to see line `[-] no repeating pattern found, try increasing window size` at the console when I do adjustments.

iceman1001, 2020-06-19:
This is no bug.   If you want to ask questions about how the proxmark3 client works, you should use the proxmark forum.

closed 2020-06-19 by iceman1001

Windslab, 2020-06-19:
> This is no bug. If you want to ask questions about how the proxmark3 client works, you should use the proxmark forum.

Yes this is not bug just useless thing, it's does't work for me, yes? :)
The forum will be after your SSL certificate :)

iceman1001, 2020-06-19:
Good luck with finding someone to help you.