RfidResearchGroup/proxmark3 #804: [Bug] Emulate VISA

Windslab, 2020-06-19:
**Describe the bug**
I think something went wrong when I trying to emulate an my VISA card in Stand-alone mode. Or I didn't understend how it's works.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'pm3 folder'
2. Click on 'proxmark3.exe'
3. Scroll down to 'end'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: [Win 10 x64]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
 - hw status OK
 - data tune -89 dB

**Additional context**
Something other must happen, yes?

iceman1001, 2020-06-19:
I tried the standalone mode and it works.    

If you have questions on the standalone mode you should ask the author of the standalone mode.  We don't give any support to usage, functions etc when it comes to standalone modes.  You can read more here: https://github.com/RfidResearchGroup/proxmark3/wiki/Standalone-mode  We offer just a easy way here in this repo to compile and add it.
As you can read in this document https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/Standalone/readme.md

If you have questions about how the proxmark3 client works,  you should use the proxmark3 forum to ask your questions.

Windslab, 2020-06-19:
Yes thanks for your check!

But could you please pick a screens how it's works for you?

Many Thanks

merlokk, 2020-06-19:
Here you need a valid POS terminal to check.
Just read your bank card, and put proxmark to valid POS. it just works in several cases...
depends on POS terminal...
BTW MSD transactions is obsolete.

Windslab, 2020-06-19:
@merlokk Yes thanks for the way!

But in my mine I have press pm3 button, put a VISA and then? Then I have saved card to the memory of PM3? Seems not :)

merlokk, 2020-06-19:
needs to have POS to check

mwalker33, 2020-06-19:
sniff = record what data the proxmark can see.
in sniff mode, the proxmark does not attempt to send commands etc. simply record what it sees.
If the function that retrieves the raw waveform data, knows how to decode it, it will and then display the data. 
lots of code that is doing a very good job of this, some code that still needs improving, and other that has not yet been developed... 
i.e. a work in progress and the community will contribute with what they find.

So, in this sniff example, you would need 3 things.
1. A card
2. A reader that will talk to the card in its native exchange (i.e. POS reader).
3. A proxmark to sniff/record the data and try to decode what it sees.

Windslab, 2020-06-19:
@mwalker33 do you talking about `hf 14a sniff` or `HF_MSDSAL` mode?

I talking about mode, and  truing to understend mechanics, how it's work. In case with sniffer it's clearly.

mwalker33, 2020-06-19:
I was trying to be generic (linked to the issue)
ie. SNIFF will just attempt to "record" what the PM3 sees and NOT cause any data to/from the card.
as @merlokk said, you need the POS reader to cause the card to exchange data.

I could be wrong, but I got the impression you may have miss understood what the "sniff" is trying to do.

Windslab, 2020-06-19:
@mwalker33 I talking about behavior in `HF_MSDSAL` mode. Not around `hf sniff`. Good. Nice.

iceman1001, 2020-06-19:
Since there is no confirmed bug,  I am closing.

closed 2020-06-19 by iceman1001

Windslab, 2020-06-19:
Yes you are right and decide. But please do not divide all to the bug an not. For me it's has still mistake and not understandable, if something another guy will be read, he do not get why you had close a topic.

Better if you/we will have a big data bank and find something, or will not and an empty. But anyway you need a dry topic moderation. Thank you.

iceman1001, 2020-06-20:
Good luck with finding help.

cabiamdos, 2021-12-24:
You click the proxmark3.exe file, when I do so they require me a lot of dll files. I execute from the command mingw prompt.

I also have the same issue.

Here is my [output](https://vimeo.com/659974338)

https://vimeo.com/659974338

Please contact me at telegram @cabiam or at email cabiam@criptext.com.
Why I am receiving this output?

`[#] +Found ISO 14443 Type A!
[#] [ Proxmark command ]
[#] 00 a4 04 00 0e 32 50 41
[#] 59 2e 53 59 53 2e 44 44
[#] 46 30 31 00
[#] [ Card answer ]
[#] 6f 2e 84 0e 32 50 41 59
[#] 2e 53 59 53 2e 44 44 46
[#] 30 31 a5 1c bf 0c 19 61
[#] 17 4f 07 a0 00 00 00 04
[#] 10 10 87 01 01 9f 0a 08
[#] 00 01 05 01 00 00 00 00
[#] 90 00
[#] ----
[#] [ Proxmark command ]
[#] 00 a4 04 00 07 a0 00 00
[#] 00 03 10 10 00
[#] [ Card answer ]
[#] 6a 82
[#] ----
[#] [ Challenge generated ]
[#] 80 a8 00 00 02 83 00 00
[#] [ Proxmark command ]
[#] 80 a8 00 00 02 83 00 00
[#] [ Card answer ]
[#] 6d 00
[#] ----
[#] [ Proxmark command ]
[#] 00 b2 01 0c 00
[#] [ Card answer ]
[#] 6d 00`