RfidResearchGroup/proxmark3 #809: [idea] More platform CROSS COMPILATION support.

xianglin1998, 2020-06-23:
Maybe we should implement multi cross compilation as platform independent? At present, we only to cross compile Android, but in some cases, we can cross compile other arm platforms or more more more.

In this implementation:

-DCMAKE_ TOOLCHAIN_ File=Toolchain description file implemented by the developer \

-DCROSS_COMPILATION=1

-D<Other Options> \

-DSKIPBT=1 -DSKIPPYTHON=1 -DSKIPPTHREAD=1 ..

-------------------------------------------------------------------------------------------------

We just need to control whether the current cross compilation is available
In client/CMakeLists.txt
Such as：
++ if(CROSS_COMPILATION) XXXXXXXXXXXX
 --  if(ANDROID) XXXXXXXXXXXX

@doegox

doegox, 2020-06-23:
ok we can probably assume we've to embed bzip2 and readline from sources eery time we're cross-compiling...
Testing if(CMAKE_TOOLCHAIN_FILE) should be enough as this is only present when cross-compiling: https://cmake.org/cmake/help/latest/variable/CMAKE_TOOLCHAIN_FILE.html

xianglin1998, 2020-06-23:
Good idea, that should be enough. :)

iceman1001, 2020-06-23:
Interesting concept but it really not a priority.

xianglin1998, 2020-06-23:
Yes, this is a small extension, and I have tested that it is available. ;)

iceman1001, 2020-06-23:
Maybe small in terms of code changes,  but big in terms of maintaining.  When ppl start having issues on their cross compiled environment,   who is going to help them?  you?
We spend already much effort on making it compile on the different OS/HW platforms as is.

xianglin1998, 2020-06-23:
Well, maybe this idea can be temporarily saved, waiting for other major things to be finished first. :)

closed 2020-06-23 by xianglin1998

doegox, 2020-06-23:
done

anyuser007, 2020-06-23:
../armsrc/Standalone/lf_em4100rwc.c: In function 'RunMod':
../armsrc/Standalone/lf_em4100rwc.c:164:21: error: implicit declaration of function 'CmdEM410xdemod' [-Werror=implicit-function-declaration]
                     CmdEM410xdemod(1, &high[selected], &low[selected]);
                     ^~~~~~~~~~~~~~
../armsrc/Standalone/lf_em4100rwc.c:164:21: error: nested extern declaration of 'CmdEM410xdemod' [-Werror=nested-externs]
../armsrc/Standalone/lf_em4100rwc.c:198:21: error: implicit declaration of function 'WriteEM410x' [-Werror=implicit-function-declaration]
                     WriteEM410x(0, (uint32_t)(low[selected] >> 32), (uint32_t)(low[selected] & 0xffffffff));
                     ^~~~~~~~~~~
../armsrc/Standalone/lf_em4100rwc.c:198:21: error: nested extern declaration of 'WriteEM410x' [-Werror=nested-externs]
cc1: all warnings being treated as errors
../common_arm/Makefile.common:109: recipe for target 'obj/lf_em4100rwc.o' failed
make[1]: *** [obj/lf_em4100rwc.o] Error 1
Makefile:133: recipe for target 'armsrc/all' failed
make: *** [armsrc/all] Error 2

doegox, 2020-06-23:
this is unrelated to this issue or commit, please open another issue or mention it on https://github.com/RfidResearchGroup/proxmark3/pull/810

xianglin1998, 2020-06-24:
Cross Build to Raspberry pi successfully, Leave footprints:

CMD: 

cmake \
-DCMAKE_TOOLCHAIN_FILE=/home/dxl/arm-cortexa9-linux-gnueabihf-4.9.3-20160512/toolchain.cmake \
-DSKIPBT=1 -DSKIPPYTHON=1  ..

Progress:

[image]

[image]

Success:

[image]

--------------------------------------------------------------------------------------------------------------------------------------------

# The toolchain.cmake demo(Work.)

```
cmake_minimum_required(VERSION 3.10)

set(CMAKE_SYSTEM_VERSION 1)

# once include.
if(ARM_TOOLCHAIN_INCLUDED)
  return()
endif(ARM_TOOLCHAIN_INCLUDED)
set(ARM_TOOLCHAIN_INCLUDED true)

# default cross opts, don't need change.
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_C_COMPILER_TARGET arm)
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

# Get root path of toolchain
get_filename_component(TOOLCHAIN_ROOT ${CMAKE_CURRENT_LIST_DIR}/4.9.3 ABSOLUTE)

# Set to find of root path.
set(CMAKE_FIND_ROOT_PATH ${TOOLCHAIN_ROOT})
# We need to set systoot to bring gcc find the includes or librarys.
SET(CMAKE_SYSROOT "${TOOLCHAIN_ROOT}/arm-cortexa9-linux-gnueabihf/sys-root")
# Set c flag to compile some source.
set(CMAKE_C_FLAGS "-std=gnu11 -w ${CMAKE_C_FLAGS}" CACHE FILEPATH "CFlags")
# Set the library path.
list(APPEND CMAKE_SYSTEM_LIBRARY_PATH "${TOOLCHAIN_ROOT}/lib/gcc/arm-cortexa9-linux-gnueabihf")
# Set ar
set(CMAKE_AR ${TOOLCHAIN_ROOT}/bin/arm-none-linux-gnueabihf-ar CACHE FILEPATH "Archiver")
# Set gcc
set(CMAKE_C_COMPILER ${TOOLCHAIN_ROOT}/bin/arm-none-linux-gnueabihf-gcc CACHE FILEPATH "GCC")
# Set g++
set(CMAKE_CXX_COMPILER ${TOOLCHAIN_ROOT}/bin/arm-none-linux-gnueabihf-g++ CACHE FILEPATH "G++")
# Set ranlib
set(CMAKE_RANLIB ${TOOLCHAIN_ROOT}/bin/arm-none-linux-gnueabihf-ranlib CACHE FILEPATH "Ranlib")
```

--------------------------------------------------------------------------------------------------------------------------------------------

Thanks again @doegox