RfidResearchGroup/proxmark3 #811: [Bug] Enter as hot key in Sniff mode

Windslab, 2020-06-23:
As my opinion key `Enter` must stop an any sniff, the same like a pressing PM3 button.
At the moment it's actions looks like it stop sniff but going to unshown exception.

doegox, 2020-06-24:
More a wish than a bug...
We're aware of the discussion and we have an internal ticket on that matter.
It's not that easy because to see if Enter is pressed, the pm3 has to check USB, which takes time during time critical parts such as sniffing.
So there is no quick & easy solution, unfortunately.
Not talking about wishes scenarios such as launching sniff, disconnecting the pm3, sniffing whatever, coming back and re-plugging pm3 etc.

Windslab, 2020-06-24:
> More a wish than a bug...
> We're aware of the discussion and we have an internal ticket on that matter.
> It's not that easy because to see if Enter is pressed, the pm3 has to check USB, which takes [...]

By the way if I press an Enter, client software understand that it need to exit from sniffing mode, but proxmark not. Mb also send to it stop command?

doegox, 2020-06-24:
> client software understand that it need to exit from sniffing mode

No it's just that the client is not blocking on a wait for pressed button, to enable the scenario I described earlier.
(but the unplug/replug scenario is not fully working yet, no need to open yet another bugreport on that)

Windslab, 2020-06-24:
Do you support my point that's Proxmark junkies needs to have an fix?

About tickets, that's like plane after landing can't stop it's turbines by button, pilots duing it by manual valve closing. Business class loosing its time and money, waiting when operation will be done...

Hope I have had explained to you main syntax :)

About welcome to contribute, I am not programist, hate when all calling me like that. But when I need then I coding... that's very strange...

doegox, 2020-06-24:
I'm aware there are still many things to improve on the proxmark, but there is no big dev team being paid for that, so progress is slow, based entirely on free time of a few enthusiastic people. And we've already a log list of TODOs.
So do you see opening so many tickets to remind us that there are indeed so many areas of improvements is not really helping this project at all, just keeping us busy trying to explain things to you and in the end, this means less time to do the real work.

Windslab, 2020-06-25:
@doegox You very bad working with your Time Management and with Proxmark.

Your carier is end at this project! :)

mwalker33, 2020-06-25:
Please keep this about the bug/issue and not about the people.

Windslab, 2020-06-26:
@mwalker33 @doegox That's was a joke :D

I hope there many ways to fix this.
By the way you can send some symbol to brak `while` loop inside the Proxmark, I understend that's speed of bus is to low(as I know about 50 ms sleep betwen commands in COM port), but you can do this operation for example each 250 ms.

Windslab, 2020-06-26:
Or another way! We bloking Enter key pressing in client, only after PM3 button press you'll get cmd line. Happy.

What you think.

closed 2020-09-02 by Windslab