RfidResearchGroup/proxmark3 #814: I can't read iclass card after pulling latest release but before pulling everything is OK

yilutongzai, 2020-06-24:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

yilutongzai, 2020-06-24:
authentication is wrong

iceman1001, 2020-06-24:
This issue has too few details for us to be able to understand what you want to report. 
I tried `hf iclass rdbl` last night and it worked.

closed 2020-06-24 by iceman1001