RfidResearchGroup/proxmark3 #815: Error in "make clean; make all"

yilutongzai, 2020-06-24:
**Describe the bug**
In file included from start.c:18:0:
BigBuf.h:42:1: error: target attribute is not supported on this machine [-Werror=attributes]
 bool RAMFUNC LogTrace(const uint8_t *btBytes, uint16_t iLen, uint32_t timestamp_start, uint32_t timestamp_end, uint8_t *parity, bool readerToTag);
 ^
cc1: all warnings being treated as errors
make[1]: *** [obj/start.o] Error 1
make: *** [armsrc/all] Error 2

yilutongzai, 2020-06-24:
It's on Mac OS10.13.6

iceman1001, 2020-06-24:
Did you follow the documentation?
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md

monster1025, 2020-06-24:
UPdate your compiler from homebrew. It will help.

anyuser007, 2020-06-28:
Ubuntu 18.04 AND KDE 
src/cmdlfio.c: In function ‘CmdIOProxDemod’:
src/cmdlfio.c:177:44: error: ‘%02X’ directive output may be truncated writing between 2 and 8 bytes into a region of size 4 [-Werror=format-truncation=]
         snprintf(crc_str, sizeof(crc_str), "(" _RED_("fail") ") 0x%02X != 0x%02X", crc, calccrc);
                                            ^~~
src/cmdlfio.c:177:77: note: format string is defined here
         snprintf(crc_str, sizeof(crc_str), "(" _RED_("fail") ") 0x%02X != 0x%02X", crc, calccrc);
                                                                             ^~~~
src/cmdlfio.c:177:44: note: directive argument in the range [0, 2147483647]
         snprintf(crc_str, sizeof(crc_str), "(" _RED_("fail") ") 0x%02X != 0x%02X", crc, calccrc);
                                            ^~~
In file included from /usr/include/stdio.h:862:0,
                 from src/cmdlfio.c:13:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 29 and 35 bytes into a destination of size 30
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:704: recipe for target 'obj/cmdlfio.o' failed
make[1]: *** [obj/cmdlfio.o] Error 1
Makefile:136: recipe for target 'client/all' failed
make: *** [client/all] Error 2

closed 2020-06-29 by iceman1001