RfidResearchGroup/proxmark3 #816: Unbrick proxmark3 with bait st-link v2

Crankle437, 2020-06-24:
Hello!

Has anyone managed to unbrick the proxmark3 easy with the bait st-link v2 (supports JTAG)? 

Openocd command:

```
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f at91sam7s512.cfg 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
srst_only srst_pulls_trst srst_gates_jtag srst_open_drain connect_deassert_srst
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
embedded:startup.tcl:162: Error: session transport is "hla_swd" but your config requires JTAG
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'jtag' called at file "at91sam7s512.cfg", line 25
in procedure 'default_to_jtag' called at file "embedded:startup.tcl", line 167
at file "embedded:startup.tcl", line 162
```

Config file (at91sam7s512.cfg):

```
# Ports
telnet_port 4444
gdb_port 3333

# Interface
#adapter_khz 1000
#source [find interface/raspberrypi2-native.cfg]

# use combined on interfaces or targets that can't set TRST/SRST separately
reset_config srst_only srst_pulls_trst

jtag newtap sam7x cpu -irlen 4 -ircapture 0x1 -irmask 0xf

target create sam7x.cpu arm7tdmi -endian little -chain-position sam7x.cpu 

sam7x.cpu configure -event reset-init { 
    soft_reset_halt
    mww 0xfffffd00 0xa5000004   # RSTC_CR: Reset peripherals
    mww 0xfffffd44 0x00008000   # WDT_MR: disable watchdog
    mww 0xfffffd08 0xa5000001   # RSTC_MR enable user reset
    mww 0xfffffc20 0x00005001   # CKGR_MOR : enable the main oscillator
    sleep 10
    mww 0xfffffc2c 0x000b1c02   # CKGR_PLLR: 16MHz * 12/2 = 96MHz
    sleep 10
    mww 0xfffffc30 0x00000007   # PMC_MCKR : MCK = PLL / 2 = 48 MHz
    sleep 10
    mww 0xffffff60 0x00480100   # MC_FMR: flash mode (FWS=1,FMCN=72)
    sleep 100

}

gdb_memory_map enable
#gdb_breakpoint_override hard
#armv4_5 core_state arm

sam7x.cpu configure -work-area-virt 0 -work-area-phys 0x00200000 -work-area-size 0x10000 -work-area-backup 0
flash bank sam7x512.flash.0 at91sam7 0 0 0 0 sam7x.cpu 0 0 0 0 0 0 0 18432
flash bank sam7x512.flash.1 at91sam7 0 0 0 0 sam7x.cpu 1 0 0 0 0 0 0 18432
```

I also tried a raspberry pi to unbrick the pm3, maybe the pm3 has some issues?

Open ocdcommand (at91sam7s512.cfg, uncomment the line under interface, `adapter_khz 1000` and `source [find interface/raspberrypi2-native.cfg]`)
```
openocd -f at91sam7s512.cfg 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
srst_only srst_pulls_trst srst_gates_jtag srst_push_pull connect_deassert_srst
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 1001 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: sam7x.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Embedded ICE version 0
Error: unknown EmbeddedICE version (comms ctrl: 0x00000000)
Info : sam7x.cpu: hardware has 2 breakpoint/watchpoint units
Info : accepting 'telnet' connection on tcp/4444
Error: timed out while waiting for target halted

Info : Halt timed out, wake up GDB.
```

Thank you very much.

iceman1001, 2020-06-24:
There is a note which is a good starting point,  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md

There some configurations under the tools folder.
https://github.com/RfidResearchGroup/proxmark3/tree/master/tools/jtag_openocd

Crankle437, 2020-06-24:
Thank you for your answer. I've already read this notes and a lot of other blog posts.

I've tried the config files from this folder again (raspberry pi):
```
openocd -f ra.cfg -f at91.cfg 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO config: tck = 11, tms = 25, tdi = 10, tdo = 9
BCM2835 GPIO config: srst = 18
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 1000 kHz
srst_only srst_pulls_trst srst_gates_jtag srst_push_pull connect_deassert_srst
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : JTAG only mode enabled (specify swclk and swdio gpio to add SWD mode)
Info : clock speed 1001 kHz
Info : JTAG tap: sam7x.cpu tap/device found: 0x3f0f0f0f (mfg: 0x787 (<unknown>), part: 0xf0f0, ver: 0x3)
Info : TAP auto0.tap does not have IDCODE
Info : TAP auto1.tap does not have IDCODE
Info : TAP auto2.tap does not have IDCODE
Info : TAP auto3.tap does not have IDCODE
Info : TAP auto4.tap does not have IDCODE
Info : TAP auto5.tap does not have IDCODE
Info : TAP auto6.tap does not have IDCODE
Info : TAP auto7.tap does not have IDCODE
Info : TAP auto8.tap does not have IDCODE
Info : TAP auto9.tap does not have IDCODE
Info : TAP auto10.tap does not have IDCODE
Info : TAP auto11.tap does not have IDCODE
Info : TAP auto12.tap does not have IDCODE
Info : TAP auto13.tap does not have IDCODE
Info : TAP auto14.tap does not have IDCODE
Info : TAP auto15.tap does not have IDCODE
Info : TAP auto16.tap does not have IDCODE
Info : TAP auto17.tap does not have IDCODE
Info : TAP auto18.tap does not have IDCODE
Info : TAP auto19.tap does not have IDCODE
Warn : Unexpected idcode after end of chain: 52 0x00000000
Warn : Unexpected idcode after end of chain: 84 0x00000000
Warn : Unexpected idcode after end of chain: 116 0x00000000
Warn : Unexpected idcode after end of chain: 148 0x00000000
Warn : Unexpected idcode after end of chain: 180 0x00000000
Warn : Unexpected idcode after end of chain: 212 0x00000000
Warn : Unexpected idcode after end of chain: 244 0x00000000
Warn : Unexpected idcode after end of chain: 276 0x00000000
Warn : Unexpected idcode after end of chain: 308 0x00000000
Warn : Unexpected idcode after end of chain: 340 0x00000000
Warn : Unexpected idcode after end of chain: 372 0x00000000
Warn : Unexpected idcode after end of chain: 404 0x00000000
Warn : Unexpected idcode after end of chain: 436 0x00000000
Warn : Unexpected idcode after end of chain: 468 0x00000000
Warn : Unexpected idcode after end of chain: 500 0x00000000
Warn : Unexpected idcode after end of chain: 532 0x00000000
Warn : Unexpected idcode after end of chain: 564 0x00000000
Warn : Unexpected idcode after end of chain: 596 0x00000000
Warn : Unexpected idcode after end of chain: 628 0x00000000
Error: double-check your JTAG setup (interface, speed, ...)
Error: Trying to use configured scan chain anyway...
Warn : AUTO auto0.tap - use "jtag newtap auto0 tap -irlen 2 -expected-id 0x00000000"
Error: auto0.tap: IR capture error; saw 0x0000 not 0x0001
Warn : Bypassing JTAG setup events due to errors
Info : Embedded ICE version 0
Error: unknown EmbeddedICE version (comms ctrl: 0x00000000)
Info : sam7x.cpu: hardware has 2 breakpoint/watchpoint units
Info : accepting 'telnet' connection on tcp/4444
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted
```
ra.cfg
[... 67 lines cut ...]
gdb_memory_map enable
#gdb_breakpoint_override hard
#armv4_5 core_state arm

sam7x.cpu configure -work-area-virt 0 -work-area-phys 0x00200000 -work-area-size 0x10000 -work-area-backup 0
flash bank sam7x.flash.0 at91sam7 0 0 0 0 sam7x.cpu 0 0 0 0 0 0 0 18432
flash bank sam7x.flash.1 at91sam7 0 0 0 0 sam7x.cpu 1 0 0 0 0 0 0 18432
```

St-link v2:
```
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f at91.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
srst_only srst_pulls_trst srst_gates_jtag srst_open_drain connect_deassert_srst
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
embedded:startup.tcl:162: Error: session transport is "hla_swd" but your config requires JTAG
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'jtag' called at file "at91.cfg", line 5
in procedure 'default_to_jtag' called at file "embedded:startup.tcl", line 167
at file "embedded:startup.tcl", line 162
```

iceman1001, 2020-06-24:
Trying cfg - files is one part,  did you connect the pins correct?   Any its not an issue for this repo. I suggest you use the forum to ask questions.

Crankle437, 2020-06-25:
Thank you, you are right, I'll move to a forum, can you please delete the issue?

closed 2020-06-25 by iceman1001