RfidResearchGroup/proxmark3 #819: LF antenna going hairwire in recent master revision

sh7d, 2020-06-25:
**Describe the bug**
In recent software revision (branch: master, commit: 1c275103948ca33580f621c7b3fc3266290de2d6) hw antenna tune going just hairwire and gives wired results:
```
[usb] pm3 --> hw tune
[=] Measuring antenna characteristics, please wait...
 🕕  12
[+] LF optimal: 542862,64 V -   0,00 kHz
[+] LF antenna is OK

[!] ⚠️  HF antenna is UNUSABLE

[-] ⛔ Not showing LF tuning graph since all values is zero.
```
Sometimes HF antena are working:
```
[usb] pm3 --> data tune
[=] Measuring antenna characteristics, please wait...
 🕕  12
[+] LF antenna:  1,33 V - 125,00 kHz
[+] LF antenna:  1,30 V - 134,83 kHz
[+] LF optimal:  1,46 V -  46,88 kHz
[!] ⚠️  LF antenna is UNUSABLE

[+] HF antenna: 46,80 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.
```
In any case lf tags are not found, so its not only the problem with the command:
lf search with tag on proxmark:
```
[usb] pm3 --> lf search
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[-] ⛔ No data found!
[=] Signal looks like noise. Maybe not an LF tag?
[usb] pm3 -->
```

When i checkout v4.9237 its working like a charm:
```
[usb] pm3 --> hw tune
[=] Measuring antenna characteristics, please wait...
 🕚  10
[+] LF antenna: 70,81 V - 125,00 kHz
[+] LF antenna: 34,79 V - 134,83 kHz
[+] LF optimal: 71,36 V - 126,32 kHz
[+] LF antenna is OK

[+] HF antenna: 47,44 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.

[usb] pm3 -->
```
And yes, i reflashed firmware between the runs

**To Reproduce**
Steps to reproduce the behavior:
1. Check out master
2. Compile and reflash
3. hw tune
4. Check out v4.9237
5. hw tune
6. Compare

**Expected behavior**
Hw tune should work same regardless of the software revision

**Desktop (please complete the following information):**
 - OS: Arch linux

Using the newest software from master branch:
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-483-g1c275103 2020-06-25 15:39:28
  compiled with GCC 10.1.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-483-g1c275103 2020-06-25 15:38:59
       os: RRG/Iceman/master/v4.9237-483-g1c275103 2020-06-25 15:39:24
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 263920 bytes (50%) Free: 260368 bytes (50%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............39764
[#]   Available memory........39764
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#] Current FPGA image
[#]   mode....................
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD567A882A7887B26
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0
[#] LF Sampling Stack
[#]   Max stack usage.........3936 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......309760
[#]   Transfer Speed PM3 -> Client = 619520 bytes/s
[#] Various
[#]   Max stack usage.........4080 / 8480 bytes
[#]   DBGLEVEL................1
[#]   ToSendMax...............-1
[#]   ToSendBit...............0
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31694 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................917 keys
[#]   T55x7...................110 keys
[#]   iClass..................7 keys
[usb] pm3 --> data tune
[=] Measuring antenna characteristics, please wait...
 🕕  12
[+] LF antenna:  1,33 V - 125,00 kHz
[+] LF antenna:  1,30 V - 134,83 kHz
[+] LF optimal:  1,46 V -  46,88 kHz
[!] ⚠️  LF antenna is UNUSABLE

[+] HF antenna: 46,80 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.
```

iceman1001, 2020-06-25:
maybe a `git bisect` to  find the triggering point.
It seems to me it doesn't always trigger.  We would need to find a way to trigger it reliable. 
Most likely the  sumadc/avgadc changes to get a better fine tuning in the voltage steps that overflows the u32.

iceman1001, 2020-06-25:
However we are not using 9K stack,  we are using 8488 bytes now.
the tune command doesn't use much variables.   Could be from swapping LF -> HF....

https://github.com/RfidResearchGroup/proxmark3/commit/b5345eb0bd83d53a0f4241ce6c726c238dec32a1

sh7d, 2020-06-25:
> However we are not using 9K stack, we are using 8488 bytes now.
Yes, and i reverted change on the newest master and it magically working, so its not that i reporting it working on daa39df
Just what i did it was:
```
git checkout master
git revert  a53c4a8a5bbc2c034d3f4341a5821f222b0d0e2
git mergetool
git add common_arm/ldscript.common
git commit
$ git show HEAD
commit 751288ace18e5343ef3e5e3d8376c7e264ee589c (HEAD -> master)
Author: sh7d <47714037+sh7d@users.noreply.github.com>
Date:   Thu Jun 25 16:57:30 2020 +0200

    Revert "chg: adapt stack ,  run "lf search" on a blank t55xx now ok"

    This reverts commit a53c4a8a5bbc2c034d3f4341a5821f222b0d0e2e.

diff --git a/common_arm/ldscript.common b/common_arm/ldscript.common
index 610e5c0b..c1fb5e84 100644
--- a/common_arm/ldscript.common
+++ b/common_arm/ldscript.common
@@ -9,7 +9,7 @@ ms of the GNU GPL, version 2 or,
 -----------------------------------------------------------------------------
 */

-stacksize = DEFINED(stacksize) ? stacksize : 8488;
+stacksize = DEFINED(stacksize) ? stacksize : 6K;
 commonareasize = 0x20;

 /* AT91SAM7S256 has 256k Flash and 64k RAM */
```
> Could be from swapping LF -> HF....
If you mean of swapping of physical connection i assure you that is impossible: i use proxmark 3 rdv4 and you cant close the casing when antenna is swapped like that (beside the wrong pin configuration), also tested it on the other (old type with no switches) antenna

iceman1001, 2020-06-25:
Try increasing 6k->7k,-8k  and see where the tune messes up

sh7d, 2020-06-25:
```
$ git diff
diff --git a/common_arm/ldscript.common b/common_arm/ldscript.common
index 610e5c0b..0472b8fd 100644
--- a/common_arm/ldscript.common
+++ b/common_arm/ldscript.common
@@ -9,7 +9,7 @@ ms of the GNU GPL, version 2 or,
 -----------------------------------------------------------------------------
 */

-stacksize = DEFINED(stacksize) ? stacksize : 8488;
+stacksize = DEFINED(stacksize) ? stacksize : 8K;
 commonareasize = 0x20;

 /* AT91SAM7S256 has 256k Flash and 64k RAM */
```
Works, double checked

aczid, 2020-06-25:
Wow, hitag2 works again. This is awesome!
Could somebody please enable the help output again? Looks like a 'default' case is missing in `cmdlfhitag.c`.

iceman1001, 2020-08-31:
I am closing this issue.   See  https://github.com/RfidResearchGroup/proxmark3/issues/921

I think many of those bad swaps of fpga image has been resolved in latest master, but if you still have this issue, looking at the output from `hw status`  shows a telling indication whats wrong.   See that linked issue for actions.

```
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#] Current FPGA image
[#]   mode....................                                    --- telling tale that device is unstable.
[#] Flash memory
```

closed 2020-08-31 by iceman1001