RfidResearchGroup/proxmark3 #824: lz4 decompress - coverity warning

iceman1001, 2020-06-29:
**Describe the bug**
Coverity warning about lz4 decompress in fpgaload.  A memcpy uses 4bytes while we feed it with 1byte.
@slurdge  :)

**Expected behavior**
No warning in Coverity

[image]

slurdge, 2020-06-29:
Actually @doegox had to put a memcpy here because some compilers (I believe yours included) complained about the manual copy that was done instead (a pointer problem IIRC).
->next_in has more than one byte in it, I don't know why it complains...

iceman1001, 2020-06-29:
Seems like a start to understand what is going on

iceman1001, 2020-09-30:
I updated the lz4 to the latest source from their repo.  Coverity didn't becpme happier...

slurdge, 2020-09-30:
Yeah probably there are some patches to put again ?

doegox, 2020-09-30:
I ported our current patches to that last version: removing the INLINE on the decompression fct and adding fct prototype

iceman1001, 2022-02-05:
closing since I don't see anything will change for the future.

closed 2022-02-05 by iceman1001