RfidResearchGroup/proxmark3 #827: hf lto info

iceman1001, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
LTO tags.
Extracting more detailed information from a dump file can be done with this shellscript. 
[lto analyzer script](https://github.com/Kevin-Nakamoto/LTO-CM-Analyzer/blob/master/lto_analyzer.sh)

**Describe the solution you'd like**
extract detailed information from a dump file,  or when _hf lto info_ command executes.

Kevin-Nakamoto, 2020-07-29:
I am wondering if you could give me an advice on this. First of all, the script converts dump data to human readable information. However, dumping data from LTO-CM takes roughly 30 seconds. I would assume that this is too long for 'hf lto info'. Maybe we should read data block partially rather than dumping whole blocks. If this sounds reasonable, I would like to suggest the following information for "hf lto info" command:

[lto_cm_mfg_info]
- LTO CM Serial Number
- CM Size
- Type

[cart_mfg_info]
- Cartridge Manufacturer
- Serial Number
- Date of Manufacture

iceman1001, 2020-07-29:
I am thinking in the lines of,

1 hf lto reader
2 hf lto info 
3 hf lto view f <filename>

For 1, is what "hf lto info" is today.  Which should be the "reader" command instead.
For 2,  if it takes a bit longer, no problem,  1-3sec,  with your suggestions above.
For 3,  read a dump file,  view decoded data, hex representation of raw data. etc.

Kevin-Nakamoto, 2020-07-29:
Thanks for your suggestions. As a first step, I will implement 2) as a preliminary enhancement so that I can measure the exact time. Depending on its results, I want to discuss if 1) is essential. For example, if 2) does not take that long time, I personally do not think that 1) would be required. I will keep you posted as the situation develops.

iceman1001, 2020-07-29:
indeed,  It comes down to how much data you need to read in order to decode according to your shell script

Kevin-Nakamoto, 2020-07-29:
Okay. Just for clarification, 3) is something that I would love to implement, too.

iceman1001, 2020-07-29:
Since you got your shell script already, 3)  will be very easy to implement ;)

iceman1001, 2021-06-15:
added more info to  `hf lto info`  ,   closing

closed 2021-06-15 by iceman1001