RfidResearchGroup/proxmark3 #830: EM4x50 Values

neijpass, 2020-07-04:
Testing the commands to EM4x50 (info) the values are not correct. Compared with a reader EMDB409 official EM Electronics.
[image]

Command emdb409 sent:

09:59:51 Sent: |02 08 33 00 00 00 21 01 1B 03|
09:59:52 Received: <02 09 33 00 01 00 00 00 00 3B 03 02 09 33 00 02 00 00 00 00 38 03 02 09 33 00 03 AA 55 00 10 D6 03 02 09 33 00 04 10 02 28 00 04 03 02 09 33 00 05 96 00 00 00 A9 03 02 09 33 00 06 02 28 20 01 37 03 02 09 33 00 07 AA 55 00 10 D2 03 02 09 33 00 08 10 02 28 00 08 03 02 09 33 00 09 96 00 00 00 A5 03 02 09 33 00 0A 02 28 20 01 3B 03 02 09 33 00 0B 10 02 28 00 0B 03 02 09 33 00 0C 53 CD AA 04 06 03 02 09 33 00 0D 00 EE 02 00 DB 03 02 09 33 00 0E 10 02 28 00 0E 03 02 09 33 00 0F 00 00 00 00 35 03 02 09 33 00 10 00 00 00 00 2A 03 02 09 33 00 11 00 00 00 00 2B 03 02 09 33 00 12 00 00 00 00 28 03 02 09 33 00 13 00 00 00 00 29 03 02 09 33 00 14 00 00 00 00 2E 03 02 09 33 00 15 00 00 00 00 2F 03 02 09 33 00 16 00 00 00 00 2C 03 02 09 33 00 17 00 00 00 00 2D 03 02 09 33 00 18 00 00 00 00 22 03 02 09 33 00 19 00 00 00 00 23 03 02 09 33 00 1A 00 00 00 00 20 03 02 09 33 00 1B 00 00 00 00 21 03 02 09 33 00 1C 00 00 00 00 26 03 02 09 33 00 1D 00 00 00 00 27 03 02 09 33 00 1E 00 00 00 00 24 03 02 09 33 00 1F 00 00 00 00 25 03 02 09 33 00 20 4D 0D 2F 02 77 03 02 09 33 00 21 32 10 00 58 61 03 >
09:59:52 OK

neijpass, 2020-07-04:
SERIAL NUMBER TAG:  022F0D4D

iceman1001, 2020-07-04:
@tharexde    seems to be something you would be able to investigate

neijpass, 2020-07-04:
If you want I can send you a card programmed with this series of values so that they can analyze it

tharexde, 2020-07-04:
I'll have a look at it first and if I need a card I'll give you a feedback. But it will take a few days.

tharexde, 2020-07-04:
Possibly I don't need a couple of days: I did a quick check with word no. 11 (= no. 14 in Proxmark 4x50_info output). 
They are identical but differ in "reading direction" i.e. LSB vs. MSB.  
Write down a words bit sequence and read it the other way round, then you'll see that they are equal. You know what I mean? 

@iceman1001: That's why I introduced MSB and LSB output, because I had the same issue with another commercial reader.

iceman1001, 2020-07-05:
Hm..   let me think about a output format for dump that will fit this msb/lsb scenario...

closed 2020-07-06 by iceman1001

iceman1001, 2020-07-06:
I pushed some changes.  @neijpass  test and see if  msb / lsb values matches your reader output.

Also added `lf em 4x50_dump`  command to save to file.

commit 9d204897 (2020-07-06): fix #830 - 'lf em 4x50_info/read' - now print msb and lsb again.