RfidResearchGroup/proxmark3 #831: Different implementations of EM410x emulation in LF_EM4100RWC and LF_EM4100RSWB

x25, 2020-07-05:
**Describe the bug**

Implementations of constructing the EM410x emulation buffer in the stand-alone modes LF_EM4100RSWB and LF_EM4100RWC are different. In my case, the LF_EM4100RWC produces invalid emulation, while the LF_EM4100RSWB implementation emulates correctly.

**To Reproduce**

I wrote a test program to check the contents of the buffer in `LF_EM4100RSWB` and` LF_EM4100RWC`  standalone modes. The code is taken from the lf_em4100rswb.c and lf_em4100rwc.c files.

<details><summary>CODE</summary>
<p>

#### You can test it onIine here: https://ideone.com/OGFyLu

```
#include <stdio.h>

static int buflen;

static void FillBuff(unsigned int bit) {
    printf("%d.", bit);
}

//Code From LF_EM4100RSWB  ( lf_em4100rswb.c  )
static unsigned long ReversQuads(unsigned long bits) {
    unsigned long result = 0;
    for (int i = 0; i < 16; i++) {
        result += ((bits >> (60 - 4 * i)) & 0xf) << (4 * i);
    }
    return result >> 24;
}

static void ConstructEM410xEmulBuf(unsigned long id) {
    //bba = BigBuf_get_addr();

    printf("LF_EM4100RSWB\n");
    int i, j, binary[4], parity[4];
    buflen = 0;
    for (i = 0; i < 9; i++)
        FillBuff(1);
    parity[0] = parity[1] = parity[2] = parity[3] = 0;
    for (i = 0; i < 10; i++) {
        for (j = 3; j >= 0; j--, id /= 2)
            binary[j] = id % 2;
        for (j = 0; j < 4; j++)
            FillBuff(binary[j]);
        FillBuff(binary[0] ^ binary[1] ^ binary[2] ^ binary[3]);
        for (j = 0; j < 4; j++)
            parity[j] ^= binary[j];
    }
    for (j = 0; j < 4; j++)
        FillBuff(parity[j]);
    FillBuff(0);
    printf("\n");
}

//Code From LF_EM4100RWC  ( lf_em4100rwc.c  )
static unsigned long rev_quads(unsigned long bits) {
    unsigned long result = 0;
    for (int i = 0; i < 16; i++) {
        result += ((bits >> (60 - 4 * i)) & 0xf) << (4 * i);
    }
    return result >> 24;
}

static void construct_EM410x_emul(unsigned long id) {

    printf("LF_EM4100RWC\n");
    int binary[4] = {0};
    int parity[4] = {0};
    buflen = 0;
    
    for (unsigned int i = 0; i < 9; i++)
        FillBuff(1);

    for (unsigned int i = 0; i < 10; i++) {
        for (unsigned int j = 3; j > 0; j--, id /= 2)
            binary[j] = id % 2;

        for (unsigned int j = 0; j < 4; j++)
            FillBuff(binary[j]);

        FillBuff(binary[0] ^ binary[1] ^ binary[2] ^ binary[3]);
        for (unsigned int j = 0; j < 4; j++)
            parity[j] ^= binary[j];
    }

    for (unsigned int j = 0; j < 4; j++)
        FillBuff(parity[j]);

    FillBuff(0);
    printf("\n");
}

int main()
{
    ConstructEM410xEmulBuf(ReversQuads(0x0000D0423C));
    ConstructEM410xEmulBuf(rev_quads(0x0000D0423C));
    construct_EM410x_emul(ReversQuads(0x0000D0423C));
    construct_EM410x_emul(rev_quads(0x0000D0423C));
    return 0;
}
```

</p>
</details>

As you can see, the contents of the buffer are different for the same key.

```
LF_EM4100RSWB
1.1.1.1.1.1.1.1.1.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.1.1.0.1.1.0.0.0.0.0.0.1.0.0.1.0.0.1.0.1.0.0.1.1.0.1.1.0.0.0.0.1.0.0.0
LF_EM4100RWC
1.1.1.1.1.1.1.1.1.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.1.0.1.0.0.1.1.0.0.0.0.0.0.0.1.0.0.1.0.1.0.0.1.0.0.0.1.0
```

**Expected behavior**

When testing on a real USB RFID reader, I see that emulating a key with `ConstructEM410xEmulBuf` from `LF_EM4100RSWB` gives results similar (**correct**) to my HoneyWell IdentKey card, and I see the **wrong key** when using `construct_EM410x_emul` from `LF_EM4100RWC`.

iceman1001, 2021-01-27:
I dunno,  there is an odd usage of bigbuffer static pointer reference when comparing those two construct bit stream.
Ask the creator of the standalone mode and see if you can get some help.
@monster1025   and @temskiy

monster1025, 2021-01-28:
I don't remember well where I copied this code, but if it works in LF_EM4100RSWB well - we can replace LF_EM4100RWC implementation to correct one =)

temskiy, 2021-01-28:
In  LF_EM4100RSWB and LF_EM4100RWC  function rev_quads identically. Functions construct_EM410x_emul  and ConstructEM410xEmulBuf  identically too. Because LF_EM4100RSWB  based on LF_EM4100RWC. What are you want to  replace? :)

monster1025, 2021-01-28:
construct_EM410x_emul and ConstructEM410xEmulBuf are **NOT** identical.

check diff:
RWC:
```
        for (unsigned int j = 3; j > 0; j--, id /= 2)
            binary[j] = id % 2;
```

RWSB:
```
        for (j = 3; j >= 0; j--, id /= 2)
            binary[j] = id % 2;
```

Fixed version for RWC is:
```
        for (int j = 3; j >= 0; j--, id /= 2)
            binary[j] = id % 2;
```

iceman1001, 2021-01-28:
@monster1025   are you using the latest source?

monster1025, 2021-01-28:
>@monster1025 are you using the latest source?
"Latest" source from topic head example with unsigned int and "j >= 0": https://pastebin.com/k2tf2b9A but it works fine with signed int (I think because id /= 2 drops bellow zero? =)).

```
% gcc test.c -o test && ./test 
LF_EM4100RSWB
1.1.1.1.1.1.1.1.1.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.1.1.0.1.1.0.0.0.0.0.0.1.0.0.1.0.0.1.0.1.0.0.1.1.0.1.1.0.0.0.0.1.0.0.0.
LF_EM4100RSWB
1.1.1.1.1.1.1.1.1.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.1.1.0.1.1.0.0.0.0.0.0.1.0.0.1.0.0.1.0.1.0.0.1.1.0.1.1.0.0.0.0.1.0.0.0.
LF_EM4100RWC
zsh: segmentation fault  ./test
```

temskiy, 2021-01-28:
At today in this repo..... construct_EM410x_emul and ConstructEM410xEmulBuf are IDENTICAL.

iceman1001, 2021-01-28:
yeah,  so the question now is the problem still exist in the current repo?

temskiy, 2021-01-28:
But latest version is not identical with function in start post and not identical with my first version merged in master  https://github.com/RfidResearchGroup/proxmark3/blob/311f43172f1056fab458e3c35b1bc119b0679477/armsrc/Standalone/lf_em4100rwc.c   
And i don`t knew where topicstarter get version from start post.

temskiy, 2021-01-28:
some time ago i pull latest version and successfully compile all with both standalone mode

x25, 2021-01-28:
hey guys! i just pulled master on 5 Jul 2020, commit e6c046e17c7be546bbca76804552a513155c4fa0

https://github.com/RfidResearchGroup/proxmark3/blob/e6c046e17c7be546bbca76804552a513155c4fa0/armsrc/Standalone/lf_em4100rwc.c

https://github.com/RfidResearchGroup/proxmark3/blob/e6c046e17c7be546bbca76804552a513155c4fa0/armsrc/Standalone/lf_em4100rswb.c

temskiy, 2021-01-28:
At [this ](https://github.com/RfidResearchGroup/proxmark3/blob/311f43172f1056fab458e3c35b1bc119b0679477/armsrc/Standalone/lf_em4100rwc.c#L60 ) commit,  function works right. You can see it [here](https://ideone.com/k1M2ZG). But after some commits function was broken.

iceman1001, 2021-01-28:
I am closing this issue since it is fixed in latest master.

closed 2021-01-28 by iceman1001